teemuatlut/TMCStepper #2: Cannot build with recent changes

alexxy, 2018-04-11:
Hi!

Seems recent addition of tmc2208 br0ke this lib

```
Indexing .pioenvs/LPC1768/libebf/libTMC2130Stepper_ID1493.a
In file included from .piolibdeps/TMCStepper/src/source/CHOPCONF.cpp:1:0:
.piolibdeps/TMCStepper/src/TMCStepper.h:501:25: error: expected ')' before '*' token
TMC2208Stepper(Stream * SerialPort, float RS, bool has_rx=true);
^
In file included from .piolibdeps/TMCStepper/src/source/COOLCONF.cpp:1:0:
.piolibdeps/TMCStepper/src/TMCStepper.h:501:25: error: expected ')' before '*' token
TMC2208Stepper(Stream * SerialPort, float RS, bool has_rx=true);
^
*** [.pioenvs/LPC1768/lib892/TMCStepper/source/CHOPCONF.o] Error 1
Compiling .pioenvs/LPC1768/lib892/TMCStepper/source/DRVCONF.o
*** [.pioenvs/LPC1768/lib892/TMCStepper/source/COOLCONF.o] Error 1
In file included from .piolibdeps/TMCStepper/src/source/DRVCONF.cpp:1:0:
.piolibdeps/TMCStepper/src/TMCStepper.h:501:25: error: expected ')' before '*' token
TMC2208Stepper(Stream * SerialPort, float RS, bool has_rx=true);
^
*** [.pioenvs/LPC1768/lib892/TMCStepper/source/DRVCONF.o] Error 1

```

teemuatlut, 2018-04-11:
It seems for LPC we need to explicitly include `Stream.h` but then you just run into another error about not finding `SoftwareSerial.h`. Fun times.

teemuatlut, 2018-04-11:
I'll move SW Serial to another folder that gets included by the build script and see how far we get from there.

teemuatlut, 2018-04-11:
Update and compile. Should be fixed now.

alexxy, 2018-04-11:
No. Now it wants HWSerial

```
Compiling .pioenvs/LPC1768/lib892/TMCStepper/source/TMC2208Stepper.o
.piolibdeps/TMCStepper/src/source/TMC2208Stepper.cpp:3:10: fatal error: HardwareSerial.h: No such file or directory

***********************************************************************
* Looking for HardwareSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:HardwareSerial.h"
* Web  > http://platformio.org/lib/search?query=header:HardwareSerial.h
*
***********************************************************************

#include <HardwareSerial.h>
^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/LPC1768/lib892/TMCStepper/source/TMC2208Stepper.o] Error 1
Compiling .pioenvs/LPC1768/lib892/TMCStepper/source/TMC2660Stepper.o

```

alexxy, 2018-04-11:
Seems like i need clean reset =) After it it works now

closed 2018-04-11 by alexxy