teemuatlut/TMCStepper #3: tmc2130 vs tmc5160

badVibes--, 2018-06-21:
hello, this is not an issue but i didn't know where else to ask. I saw you added support for 5160 and i was wondering if the differences are big and weather the code for 2130 could work for at least basic setup of 5160. im wondering because smoothieware is about to get support for 2130 but 5160 is probably way of.

teemuatlut, 2018-06-22:
The 5160 brings the advantages from each of the previous ones together and it will definitely be the best driver available.
Compared to 2130 it has more current delivery just like 2660 and it has the newer stealthChop2 that is present in 2208 (and 2224).
Compared to 2208 you get stallGuard and higher currents. I also consider SPI to be easier to work with.
The 5160 shouldn't need additional cooling at the typical current levels we use them.

Depending on the pricing and availability, I'd probably never recommend any of the other models anymore.
It's also likely that Marlin will be the first one to have dedicated support for the drivers and that may very well be the case for quite a while. The good news is that it shares a lot with 2130 and you'll likely be able to use them by defining them as 2130 drivers.

badVibes--, 2018-06-23:
>  The good news is that it shares a lot with 2130 and you'll likely be able to use them by defining them as 2130 drivers.

that really is good news. i saw that you wrote some functions for setting the RMS current of 5160 and i was worried that it was somehow different. All i want to do for starters is set the current and put them in stealth chop mode..

thank you.

teemuatlut, 2018-06-23:
Actually the current equation is one of the things that _did_ change. So if you configure Smoothieware (pretending 5160 is 2130) to 600mA, it's actually not going to be that. But I think you should still be able to get movement out of them and the current can set with the mindset "do I need more or do I need less", just keeping in mind that the value doesn't hold real world meaning.

badVibes--, 2018-06-24:
hmm, looking at the code only GLOBALSCALER was added to the equation which is 0 by default according to the data sheet and should have no effect if its not set to something different. So i guess the current setting should still work as intended?

teemuatlut, 2018-06-24:
TMC2130 also has 20mOhm added to the sense resistor value and TMC5160 doesn't have vsense setting that is used when setting current in TMC2130. V_fs is sligthly different but not much.

hasenbanck, 2018-10-07:
Just to add a little feedback from a hardware implementation standpoint:
The TMC2130 are easier and cheaper to implement. The TMC5160 need 8 external MOSFET for each stepper driver, which are the main culprit here.

I have seen pictures of Watterots version of the SilenStepStick TMC5160, so they should be usable in the same formfactor for people that use such driver boards.

ghent360, 2018-10-15:
as hasenback mentions the 5160 requires external mosfets. while these allow much higher current handling they complicate the design. the external mosfets add to the cost as well as the required board space. at high current the mosfets would need cooling as well. last but not least the bridge needs "tuning" of the switching speed.

so in the end it depends on your design goal. 5160 may be a good choice for some application, but 2130 is not obsolete yet.

WarlockD, 2019-02-03:
Don't forget about the TMC5161 , That was just realized.  Its just basically the 5160 with built in mosfets that handle 3.5A.  The Bob came out so going to pick one up at digikey to try it out.

thatdutchguy2, 2019-03-20:
> Don't forget about the TMC5161 , That was just realized. Its just basically the 5160 with built in mosfets that handle 3.5A. The Bob came out so going to pick one up at digikey to try it out.

Please keep us updated! I just saw that Watterott just came with a silentstepstick format of this stepper driver. Waiting for a reputable source to try them out and then might buy them too!

WarlockD, 2019-03-20:
It works just like the TMC5130's I have been experimenting with, both in step/dir and the RAMP generation.  Real issue is getting them in pololu socket.  TMC has some pictures of the TMC2160 (same as the TMC5160, just no ramp generator), in that format but there doesn't seem to be anyone making them.

I think there is just such a glut of TMC2130 and older chips that no one is trying to make them in bulk.  If someone kickstarts the TMC5161, I would throw in some cash hah.

thatdutchguy2, 2019-03-20:
Yes, but thats what i meant. Watterott is now producing them in the polulu style StepStick. They have the mosfets on the board so it’s just plug and play as far as I can see. Here’s a link: https://shop.watterott.com/SilentStepStick-TMC5160-Schrittmotortreiber

thatdutchguy2, 2019-03-21:
By the way, WarlockD, have you been experimenting with Marlin 2.0 software for 3D printers? because thats also how im going to use them probably. Going to buy the new SKR 1.3 board which will be a pretty good upgrade from my current motherboard which has built in stepper drivers

rotembo, 2019-03-27:
Please share how was your experience with the SKR 1.3 and the TMC 5160,

thatdutchguy2, 2019-03-27:
Will do! i expect there not be a problem because the TMC5160 is already supported in Marlin 2.0!

klcjr89, 2019-03-31:
I have been using the 5160 stepsticks from Watterott for awhile now but have been having random SPI issues, which thankfully haven't ruined print quality, but it needs to be solved.

https://github.com/MarlinFirmware/Marlin/issues/13544

teemuatlut, 2019-04-01:
The TMC2160, TMC5130, TMC5160 and TMC5161 all use the same inherited code from the 2130 class for SPI communication.

You can try disabling the monitoring feature as then there will be no periodic communication to the drivers.

If that doesn't work, try "running" the printer without the TMC5160 drivers plugged in. Of course the axis won't move but it might help narrowing down the issue to software or hardware.

klcjr89, 2019-04-01:
@teemuatlut The code looks to be different:

### TMC2130

```
template<char AXIS_LETTER, char DRIVER_ID>
  void tmc_init(TMCMarlin<TMC2130Stepper, AXIS_LETTER, DRIVER_ID> &st, const uint16_t mA, const uint16_t microsteps, const uint32_t thrs, const float spmm, const bool stealth) {
    st.begin();

    CHOPCONF_t chopconf{0};
    chopconf.tbl = 1;
    chopconf.toff = chopper_timing.toff;
    chopconf.intpol = INTERPOLATE;
    chopconf.hend = chopper_timing.hend + 3;
    chopconf.hstrt = chopper_timing.hstrt - 1;
    st.CHOPCONF(chopconf.sr);

    st.rms_current(mA, HOLD_MULTIPLIER);
    st.microsteps(microsteps);
    st.iholddelay(10);
    st.TPOWERDOWN(128); // ~2s until driver lowers to hold current

    st.en_pwm_mode(stealth);
    st.stored.stealthChop_enabled = stealth;

    PWMCONF_t pwmconf{0};
    pwmconf.pwm_freq = 0b01; // f_pwm = 2/683 f_clk
    pwmconf.pwm_autoscale = true;
    pwmconf.pwm_grad = 5;
    pwmconf.pwm_ampl = 180;
    st.PWMCONF(pwmconf.sr);

    #if ENABLED(HYBRID_THRESHOLD)
      st.TPWMTHRS(12650000UL*microsteps/(256*thrs*spmm));
    #else
      UNUSED(thrs);
      UNUSED(spmm);
    #endif

    st.GSTAT(); // Clear GSTAT
  }
#endif
```

### TMC5160
```
template<char AXIS_LETTER, char DRIVER_ID>
  void tmc_init(TMCMarlin<TMC5160Stepper, AXIS_LETTER, DRIVER_ID> &st, const uint16_t mA, const uint16_t microsteps, const uint32_t thrs, const float spmm, const bool stealth) {
    st.begin();

    int8_t timings[] = CHOPPER_TIMING; // Default 4, -2, 1

    CHOPCONF_t chopconf{0};
    chopconf.tbl = 1;
    chopconf.toff = timings[0];
    chopconf.intpol = INTERPOLATE;
    chopconf.hend = timings[1] + 3;
    chopconf.hstrt = timings[2] - 1;
    st.CHOPCONF(chopconf.sr);

    st.rms_current(mA, HOLD_MULTIPLIER);
    st.microsteps(microsteps);
    st.iholddelay(10);
    st.TPOWERDOWN(128); // ~2s until driver lowers to hold current

    #if ENABLED(ADAPTIVE_CURRENT)
      COOLCONF_t coolconf{0};
      coolconf.semin = INCREASE_CURRENT_THRS;
      coolconf.semax = REDUCE_CURRENT_THRS;
      st.COOLCONF(coolconf.sr);
    #endif

    st.en_pwm_mode(stealth);

    PWMCONF_t pwmconf{0};
    pwmconf.pwm_freq = 0b01; // f_pwm = 2/683 f_clk
    pwmconf.pwm_autoscale = true;
    pwmconf.pwm_grad = 5;
    pwmconf.pwm_ampl = 180;
    st.PWMCONF(pwmconf.sr);

    #if ENABLED(HYBRID_THRESHOLD)
      st.TPWMTHRS(12650000UL*microsteps/(256*thrs*spmm));
    #else
      UNUSED(thrs);
      UNUSED(spmm);
    #endif
    st.GSTAT(); // Clear GSTAT
  }
#endif
```

teemuatlut, 2019-04-21:
It's not different in a sense that matters. Nor do they have differences that are intentional. They should be about the same but it seems there's been a rebase mishap that I didn't catch. Thanks for pointing that out!

WarlockD, 2019-04-25:
Just checking but I just got mine from Watterott, have anyone else been able to use the internal ramp generator?  After looking at the schematics the cfg pins are forced into step/dir mode but when you run the internal ramp, it seems like it's running, at least the VACTUAL and XACTUAL are moving around but no motor action.

So does the TMC5160 not work in ramp mode unless the pins are set even if its configed in spi?  Just checking.  I haven't tested the dir/step mode yet, but I am sure it works.

teemuatlut, 2019-04-25:
Yea the driver is hardware wired for S/D mode so it won't work without PCB modification.
Interesting that the generator status values change but it doesn't really matter if it doesn't actually execute the stepping.

Maybe at some point someone will design a TMC5160 Ramp Generator based board like TRAMS was/is, but the downside from Marlin's point of view is that it'll also bypass some of the features like Linear Advance.

klcjr89, 2019-04-25:
@teemuatlut I'm not sure what you were referring to in regards to the quote: "They should be about the same but it seems there's been a rebase mishap that I didn't catch. Thanks for pointing that out!"

I'm struggling to figure out what works and what doesn't with tmc drivers, such as `ADAPTIVE_STEP_SMOOTHING` and if 8x microstepping works with interpolation mode.

teemuatlut, 2019-04-25:
The differences in `tmc_init`.

There should be no problem in running 8 microstepping with interpolation. I used to do that with my Z axis, but I have no personal experience with the step smoothing feature.

klcjr89, 2019-04-25:
I guess on the surface, it seems like there would be no reason not to enable adaptive step smoothing, because the Marlin documentation is vague otherwise.

WarlockD, 2019-05-01:
Not sure why Linear Advance wouldn't work, though it would require quite a rewrite.  Just using the simple trapezoid generator, (I did mess with the 6 point generator, but not sure how to modify Marlin's generator on the acceleration/deceleration bit), would just mean you have to q in the extruder movments into it.

It would mean everything would have to use the ramp generator, couldn't mix and match different chips.  And again, big rewrite not to mention calberation on the motors/printers on acceleration values.  At that point, you might as well go all the way to support a TMC4361 to get real closed loop working. 

I wish there was a mode of marlin to offload the ramp generator to another microcontroller.  It would solve the irq overhead on it.

sniperlucian, 2019-05-07:
> wish there was a mode of marlin to offload the ramp generator to another microcontroller. It would solve the irq overhead on it.

Thats exactly were klipper kicks in  - at least from the other way around.  ;)

alfredanil, 2019-05-12:
@I wish there was a mode of marlin to offload the ramp generator to another microcontroller.

esp32 would probably be an ideal solution for offloading ramp generation and it can also act as a web server for host.

alfredanil, 2019-05-12:
> I wish there was a mode of marlin to offload the ramp generator to another microcontroller.

esp32 would probably be an ideal solution to offload ramp generation. it could also run a web server for host.

charleslemaire0, 2019-05-25:
> You can try disabling the monitoring feature as then there will be no periodic communication to the drivers.
Hi,
how can I disable the monitoring feature?
Charles

teemuatlut, 2019-05-25:
Configuration_adv.h
Disable `MONITOR_DRIVER_STATUS`

charleslemaire0, 2019-05-25:
I mean in the TMCStepper Lib is there a way to reduce the SPI traffic?
It looks like the SPI of the TMC5160 is slow compare to the TMC2130...

teemuatlut, 2019-05-25:
If you have monitoring disabled then there is no SPI calls made after initial reboot.
Unless of course you manually use the gcodes.