teemuatlut/TMCStepper #10: rms_current with TMC5160

charleslemaire0, 2018-12-02:
Hello,

the TMC5160 has no vsense information but the method set rms consider the vsense.

FYI I use your lib in this project:

https://groups.io/g/TeenAstro/

the Rsense for the TMC5160 bob is 0.075

Charles

void TMCStepper::rms_current(uint16_t mA) {
  uint8_t CS = 32.0*1.41421*mA/1000.0*(Rsense+0.02)/0.325 - 0.5;
  // If Current Scale is too low, turn on high sensitivity R_sense and calculate again
  
  if (CS < 16) {
    vsense(true);
    CS = 32.0*1.41421*mA/1000.0*(Rsense+0.02)/0.180 - 0.5;
  } else { // If CS >= 16, turn off high_sense_r
    vsense(false);
  }
  irun(CS);
  ihold(CS*holdMultiplier);
  //val_mA = mA;
}

teemuatlut, 2018-12-02:
The methods are overloaded

https://github.com/teemuatlut/TMCStepper/blob/cefefbe1b098654b4d5ff7ede9acfb4ba987eadb/src/source/TMC5160Stepper.cpp#L15-L47

charleslemaire0, 2018-12-02:
My mistake. sorry.

Very good work.
If you can add the TCM260 that is used in the TOS-100 it would be great!
http://blog.trinamic.com/2013/02/01/getting-started-with-the-tos-100-arduino-stepper-motor-shield/

closed 2018-12-02 by charleslemaire0

BlueGene00, 2019-08-21:
> The methods are overloaded
> 
> https://github.com/teemuatlut/TMCStepper/blob/cefefbe1b098654b4d5ff7ede9acfb4ba987eadb/src/source/TMC5160Stepper.cpp#L15-L47

I was checking the rms_current function for using the TMC5160 and noticed that the formular contains additional 0.02 Ohm:

https://github.com/teemuatlut/TMCStepper/blob/master/src/source/TMCStepper.cpp

But this value can not be found in the the official manual of the chip and the calculation is wrong using this value.

It seems that there is currently no different calculation of the rms_current for the TMC5160:
https://github.com/teemuatlut/TMCStepper/blob/master/src/source/TMC5160Stepper.cpp

The function also does not seem to care if the user puts in a too high value. The register can only use values up to 31. If I put an unrealistic current value, the CS value of irun and ihold will exeed 31... Is this covered somewhere else? 

I would like to add something like at the end before using the irun and ihold functiond:
```
if(CS>31){
CS=31;
}
```

teemuatlut, 2019-08-21:
TMC5160 does **not** use `TMCStepper::rms_current`.
Instead you need to look at `TMC2160Stepper::rms_current` which it inherits, and which does not have the 0.02Ohms added.

Capping the rms current is currently left at the responsibility of the user.

BlueGene00, 2019-08-21:
Thanks for the very quick clarification.