teemuatlut/TMCStepper #11: Run at Velocity via SPI

FlowBo, 2018-12-02:
Hej,
I'm trying to implement a homing routine with Stallguard. (TMC5160)
Normally I'm running the drivers via SPI in Position mode. 
For homing I switch to Velocity mode. But since I'm not using the step/dir pins I wonder how to get the Stepper running at a constant velocity?
I have set the VMAX and AMAX but the motor does not move. Is there a specific command like "Start" or "run"?

teemuatlut, 2018-12-02:
You can refer to my unfinished TRAMS code. It's quite old at this point but as the chips are very similar and I did have homing working, you should be able to get started from there.

https://github.com/teemuatlut/Marlin/blob/trams-updated/Marlin/TRAMS.cpp#L77

FlowBo, 2018-12-04:
Thanks.
revisiting what I wanted to achieve I decided to go for positioning mode as it is also done in the example. Reasons:
- if something goes wrong while homing in velocity mode the motor would stall indefinitely. With positioning an end of the routine is defined
- To max travel distance for homing is defined by the physical structure

I post my code here for others as a starting point:
```
initSG(){
  driver.rms_current(200);
  driver.sgt(3);
  driver.sg_stop(true);
  driver.TCOOLTHRS(80000);
  driver.AMAX(10000); 
  driver.VMAX(100000);
  driver.XACTUAL(0);
  driver.XTARGET(int(maxTavelDistance*1.5));
}

updateHomingRoutine(){
  if (driver.status_sg() == true){
    driver.RAMPMODE(3);
    homed = true;
    driver.XACTUAL(0);
    driver.XTARGET(0);
    driver.sg_stop(false);
    // RESET YOUR MOTIONCURVE VALUES  AFTER HOMING
  }
}
```

closed 2018-12-04 by FlowBo