teemuatlut/TMCStepper #15: TMC2130 diag1 as aktiv high

dremeier, 2019-03-14:
I know the discussion here https://github.com/MarlinFirmware/Marlin/issues/10294. But on my Ultratronics (SAM3x) Board it is not working when the diag1 is in open drain mode even when the Sam3x input has the pullups aktiv. The only way to get the sensorless homing to work, is a external pullup resistor. So I would like to have the diag1 as aktiv high (pushpull).
Please can you tell me where and how to set this in you "TMCStepper" lib?

teemuatlut, 2019-03-18:
Diag1 active high can be forced with

```cpp
#define TMC_ADV() { \
  stepperX.diag1_pushpull(true); \
  stepperY.diag1_pushpull(true); \
}
```

This will modify the driver registers independent of the Marlin settings.

dremeier, 2019-03-19:
Hello Teemuatlut,
Thanks for your recommendation!

I put this in the Configuration_Motor_Driver.h (MarlinKimbra):
#define  TMC_ADV() { \
    stepperX->diag1_pushpull(true); \
    stepperY->diag1_pushpull(true); \
    stepperZ->diag1_pushpull(true); \
    stepperZ2->diag1_pushpull(true); \
 }
And it works like a charm! Perfect!

The MarlinKimbra works with your TMCStepper lib and Marlin works with your TMC2130Stepper lib. What are the different?
I´m using a Arduino Due (SAM3x) board. When i compile the sketch there is a state like: "TMCStepper works with avr Boards but probably not with sam Boards". But it looks like it is working with sam Board.

teemuatlut, 2019-03-20:
> The MarlinKimbra works with your TMCStepper lib and Marlin works with your TMC2130Stepper lib. What are the different?

As I'm expanding the number of TMC ICs supported, I didn't want to have 8 different libraries to support so `TMCStepper` supercedes both `TMC2130Stepper` and `TMC2208Stepper`, combining support for both and then some. Marlin changed to this new library after the release 1.1.9. MarlinKimbra is based on the newer code base from 2.0.x.

closed 2019-03-20 by dremeier