teemuatlut/TMCStepper #16: TMC4671

charleslemaire0, 2019-03-17:
Hi,

I am using your lib in my Project: https://groups.io/g/TeenAstro/wiki/home
Your are mentioned has contributor 🥇

Many user has requested servo motor support. Trinamic has a board that is interesting, the TMC4671, https://www.trinamic.com/support/eval-kits/details/tmc4671-bob/

Are you interested to explore that path?

Charles

teemuatlut, 2019-03-18:
Hey, that's a really cool looking project!
Happy that I've made something you've found useful =)

While I don't really know much about telescopes and everything related, I can help out if you can get down to specifics.

charleslemaire0, 2019-03-18:
Hi,

 

I am using the TMC lib for the TMC5160 and the TMC2130 if you can extend the lib to be able to support the TMC4671, it just fine. I will need the lib to configure the chip via spi (parameter like current, encoder resolution etc...) and then control the motor via the step/dir interface (i don't need th lib for that :)). The board is quite interesting as it supports both stepper and servo. Expert things like stall guard are for me not important, I just need the basic feature of the board.

 

I am not familiar with all the registry stuff and thanks your lib I saved me a lot of my time for the telescop related issues. If I was able to do that I would have not asked.

 

Sincerely,

 

Charles

 

Gesendet: Montag, 18. März 2019 um 21:42 Uhr

teemuatlut, 2019-03-18:
I'll see what I can do.
But I'd rather not give a time line as I wouldn't meet it anyhow =)

charleslemaire0, 2019-03-18:
You made a great job, take your time and have fun. If there is no fun all this have no sense.