teemuatlut/TMCStepper #17: tmc2160 communicate over UART-SPI bridge

alexgubanow, 2019-03-18:
Hi,
Currently developing board based on tmc2160, losed tmc5160 because it pointless without supporting encoder feedback on marlin side. So, to reduce amount of wire i decided to make uart-spi bridge, based on stm32f07. PCB already done, firmware only started. I will make two versions of driver, standalone and uart connected. Board suitable to mount on backside of nema17, first my attempt was to make pololu size, but pcb cost starts to be equal to moon flight.
My question, is it easy to you make support of tmc2160 with UART? or i have to make my own library?
I think we can base uart communicating on tmc2208 part.
PCB proj https://easyeda.com/alexgubanow/yadrv.
Firmware https://github.com/alexgubanow/YaDRV

teemuatlut, 2019-03-18:
No, I think it should be a relatively simple job. Mostly just rearranging the class inheritance structure so that the TMC2160Stepper also has access to UART communication methods. I'd likely separate the communication to their own classes.
I don't like making promises and even less about setting myself additional deadlines but I'll see if I can make a branch that you can test with.

alexgubanow, 2019-03-18:
So, on my side i have to make just clear UART-SPI bridge?

teemuatlut, 2019-03-18:
It would/should work just like how the TMC2208 interface works. But I have no idea of the SoftwareSerial implementation on the STM platform.

alexgubanow, 2019-03-18:
i will write it on basic HAL, so it would be pure hardware implementation

alexgubanow, 2019-03-19:
Sorry, my bad, only now realized about 8bit limit in stm32 hardware uart. I dont wanna to make software serial. I think i will make own implement of command sequnces based on simple UART, like RWbit+Address+32bData+CRC for writing and RWbit+Address+CRC for reading. Each 8bit will starts with 0 and ends with 1. What you think?