teemuatlut/TMCStepper #19: TMC5160 and configurator

calogit, 2019-03-25:
Hi !

This is not an issue, I'm the issue because i'm a really noob about code ...
But thank's for your cool and hard work about TMC chips.

My coding level can't go further than to use simple things that libraries allow, simplifying things.
So I'm using accelstepper, and recently this super multiple stepper library that include acceleration named  [Teensystep](https://github.com/luni64/TeensyStep) working on [teensy](https://www.pjrc.com/teensy/).

Now on TMC5160 there is no standalone Config pin option to setup the stepping mode, and no potentiometer to setup the current anymore.
Do you think there is a possible way to configure it as the [configurator](https://learn.watterott.com/silentstepstick/configurator/) allow it for the TMC2208 with the testboard ?
Or any other simple way via arduino code ?

thank's by advance

Mathias.

teemuatlut, 2019-04-21:
I believe the configurator works with an UART connection through a USB-UART bridge. All the TMC5160 implementations I know opted for the SPI connection for the driver. So I don't think that program will work with the driver.
Going further however, it's only a matter of sending the right commands through whatever connection you chose. But I haven't implemented the methods to control the OTP registers so you'd have to write those yourself.

calogit, 2019-04-21:
Thank's, for your answer.

If I setup my driver [with your sample code](examples/TMC_AccelStepper/TMC_AccelStepper.ino) using your TMCstepper library( mainly for current, stealthchop ,microstepping ), are the settings saved in the TMC chip ?

I mean if I power it off ,then take off my driver to an other board without SPI wirering, only ena,step,dir, does the driver will works with my previous setting ?

Can it be as simple as this ?

teemuatlut, 2019-04-21:
No, the starting values are cleared when you remove power from the IC.
You can modify some of the default values by making changes to the OTP register.

I don't think you can set any of the settings you're looking for from the TMC5160 OTP register.

calogit, 2019-04-21:
okai !
So my last question,
- Do you think your library can works on [teensy 3.xx board ?](https://www.pjrc.com/teensy/)
if yes without conflic with this [TeensyStep library ?](https://github.com/luni64/TeensyStep)

Thank's you very much !

teemuatlut, 2019-04-21:
The TMCStepper library handles the configuration of the driver, while libraries such as AccelStepper or TeensyStep handle acceleration profiles and step rates.
In other words, TMCStepper only uses the SPI/UART part of the driver while motion focused libraries only handle Step and Dir pins.

I don't think there should be any problem with running the library with a Teensy. The only prerequisites are a SPI library, a Serial class and a SoftwareSerial library.

FlowBo, 2019-04-21:
I have been able to use the library with the teensy 3.2. worked like a charm with spi.

calogit, 2019-04-21:
WOW so many thank's for your faster answer and taking time for me.
Guys you made my day.

calogit, 2019-04-21:
I would like to control multiple drivers.

On teensy page I found this:
_"Multiple SPI devices use the same SPI SCK, MISO and MOSI signals but each device will need it's own SS pin. "_
- Does SS pin the same thing as the CS_PIN ?

If yes, does this multiple stepper driver code correct ?:

```
#define R_SENSE    0.075

#include <TMCStepper.h>
TMC5160Stepper driver1(15, R_SENSE);
TMC5160Stepper driver2(20, R_SENSE);

#include <AccelStepper.h>
AccelStepper stepper1(1, 16, 17);
AccelStepper stepper2(1, 21, 22);

void setup() {
    SPI.begin();
    Serial.begin(9600);
    while(!Serial);
    Serial.println("Start...");
    
    pinMode(15, OUTPUT);
    digitalWrite(15, HIGH);
    driver1.begin();             // Initiate pins and registeries
    driver1.rms_current(600);    // Set stepper current to 600mA. The command is the same as command TMC2130.setCurrent(600, 0.11, 0.5);
    driver1.en_pwm_mode(1);      // Enable extremely quiet stepping
    driver1.pwm_autoscale(1);
    driver1.microsteps(16);

    stepper1.setMaxSpeed(500); 
    stepper1.setAcceleration(1000);

    pinMode(20, OUTPUT);
    digitalWrite(20, HIGH);
    driver2.begin();             
    driver2.rms_current(2000);    // Set stepper current to 2000mA.
    driver2.en_pwm_mode(1);      
    driver2.pwm_autoscale(1);
    driver2.microsteps(16);

    stepper2.setMaxSpeed(4000);
    stepper2.setAcceleration(20000); 

}

void loop() {

}
```

Tnak's again by advance

teemuatlut, 2019-04-21:
SS (Slave Select) is the same as CS (Chip Select).
Your code looks about right but you don't need to set the CS pin pin modes as that is handled in the `begin` method. You will of course need to add something to the loop to actually start stepping or use interrupts.

friedl1977, 2019-08-21:
Hi - 

My apologies in advance if the question has been addressed before, but I would also appreciate some help with switching from TMC2208 to TMC5160.

I am using a custom designed PCB and is happy to change the design if need be, but I am having trouble figuring out the wiring between the driver and MC.  With 2208 it was easy enough simply pulling CFG pins either high or low, but it seems not to be the case with the 5160?

I hope someone can shed some light on how to wire the driver directly to the MC without using a RAMPS board.

Many thanks in advance!

teemuatlut, 2019-08-22:
If you're making your own board I'd suggest you refer to the datasheet.
The main things you'll want to look at is the SPI connections and then the pins called SPI_MODE and SD_MODE.

friedl1977, 2019-08-22:
Hi - 

As the pins on the drivers I purchased  were labeled i.e. SDI/**CFG1**, SCK/**CFG2** etc. I was hoping I could, similar to the TMC2208, simply pull the CFG pins either High/Low.  From there then use STEP/DIR as apposed to SPI.  I assume this is not the case?   Sorry of this does not make much sense, I am quite new to all of this :) 

I noticed that the CLK pin on TMC5160 need to be pulled down to GND and not left floating.  I consulted support, so they sent me a 133 page document, quite daunting for someone new :D   From my understanding then the wiring between the TMC5160 and Particle photon should be as follow then: (bold being the TMC5160 pins)

**SDI** - > A5 (MOSI)
**SPI** -> A4 (MISO)
**SCK** -> A3 (SCK)
**CS** -> A2 (SS)

**VIO** -> 3V3
**GND** -> GND
**CLK** -> GND
**D7** -> STEP
**D0** - DIR

**EN** -> GND

Of course then A1, A2, B1,cB2  connections to the stepper as well as the external 24V and GND for motor power supply.

Thank you for your assistance so far.

teemuatlut, 2019-08-22:
If your custom PCB design uses Pololu style sockets for the drivers, like you would with the SilentStepSticks, then all you need to worry is routing the SPI pins, the S/D/E pins and finally wiring the CLK to GND.
So your planned routing seems to make sense.
As long as with `SPI -> A4 (MISO)` you mean `SDO -> A4 (MISO)`

*The 5160 SilentStepStick board can not be used in stand alone mode.*

friedl1977, 2019-08-22:
Hi - 

Thank you for the prompt response.  I am indeed using the SilentStepStick boards.  I have some from Bigtree-tech and I have some original ones from Trinamic.  To confirm, the config pins should be wired as follow:

- `SDI - > A5 (MOSI)` 
- `SDO -> A4 (MISO)` 
- `SCK -> A3 (SCK)`
- `CS -> A2 (SS)`

Just one last question, 
> The 5160 SilentStepStick board can not be used in stand alone mode.

Do you mean without connecting SPI, or that I connecting the driver directly to Micro controller.  I found this link connecting the TMC5160 (evaluation board) to an Arduino Mega.

[http://blog.trinamic.com/2019/02/22/explore-and-tune-stallguard2-with-the-tmc5160-eval-arduino-mega-2560-in-step-and-direction-mode/](url)

Thanks again, I really appreciate the help :)

Baramykin, 2020-12-23:
Hi,
I don't understand how to connect TMC5160 8 pcs to Teensy 4.1 UART mode.
I can't find the schematics.
I want to use TMC5160 V1.2 drivers.
Help me if possible, please.

teemuatlut, 2020-12-23:
I am not aware of any existing breakout boards for TMC5160 that connect using UART instead of SPI.

If you do however have a module that allows this, the UART interface is described in chapter 5.4 of the datasheet where you connect your serial port to SWIOP pin.

Mrugank12, 2022-01-07:
can anyone share wiring connection for TMC 5160 BOB to Arduino Mega 2650 for SPI because i am stuck here and cant make the motor turn so if anyonce could share the connections details and code it would be a great help as i am noob with elctronics

teemuatlut, 2022-01-07:
Your first priority should be to get communication working. You can try reading the `.version()` from the driver.

MCU | BOB
---|---
GND | GND
5V | VCC_IO
CS | CSN
SPI SCK | SCK
SPI MOSI | SDI
SPI MISO | SDO
GND | CLK16