teemuatlut/TMCStepper #21: BigTreeTech SKR 1.3 + MKS TMC2208 v1.1 Connection Error (UART)

Brainc0re, 2019-04-18:
I got my board today. For testing, i connected one stepper (Z). According to the manual i removed all jumpers from the STEP/DIR headers and set them on the UART headers. But i always get a "TMC CONNECTION ERROR" on the display...

[SKR V1.3-Instructions.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/SKR%20V1.3-Instructions.pdf)

Hardware: [https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/tree/master/hardware](url)

I have these drivers: https://www.coqaulores.de/wp-content/uploads/2018/07/StepStick-MKS-TMC2208_top.jpg (Same pinout but V1.1)

This is my current build:
https://drive.google.com/file/d/1NryYj-VFD_dcWFG64mShB-L8xsBiXHr1/view?usp=sharing

Do i have the wrong stepper drivers for UART?

teemuatlut, 2019-04-18:
Unfortunately I don't have the board so I could tell for sure.
As far as I'm aware, the UART should work with LPC1768 and TMC2208/TMC2224.

What does your `M122` output look like?
If they base their own development on SmoothieWare, it's possible that it's only a matter of using the wrong pins.

Brainc0re, 2019-04-18:
> What does your `M122` output look like?

That is my next problem ^^ I can't connect the board to a PC. Windows uses its own COM driver instead of the provided from BigTree :/

teemuatlut, 2019-04-18:
This may help you
http://marlinfw.org/docs/basics/install_rearm.html

Brainc0re, 2019-04-18:
Serial -1 did the trick. I also could install the driver on a W7 Lamptop in Testmode (ATM cant restart my desktop).

M122:

[image]

XylenC4, 2019-06-15:
I'll used LERDGE TMC2208 (comming from this board) and i'll had to bridge the UART pins on the bottom with a soldering iron. The BIGTREETECH TMC2208 has the same pins according to the manual you can find at: https://github.com/bigtreetech/BIGTREETECH-TMC2208-V3.0/blob/master/TMC2208-V3.0%20manual..pdf (2.2 UART MODE), if you bridge tim (i'll bridged all 3 pads) it should work fine!

lightface79, 2019-08-28:
I have the same problem with the Most recent build of Marlin 2 2019-08-28  

Using skr 1.3
And watterott tmc2008

Get the same respons with M122

Get the driver status and config 
But in the end it says error all low.
And my lcd says tmc connection error as soon as the card boots.

And i cant run the stepper in eany direktion.
And i can easaly din the stepper motor axel even if i just toold it to enable and move from prontoface or my lcd and rotary decoder.

Is there å solution?

Best regards jan  // sweden

lokus77, 2019-08-29:
Is not a marlim problem, is the driver sure. Marlin with SKR V1.3 runs nice in uart mode.

Each brand (and version) of TMC2208 has different settings to enable uart. Its hard sometimes...

I did a guide:  https://docs.google.com/spreadsheets/d/13LyuaG1WTgv0dE62VP0Jh8NLjLPRxc8z2F-cb96CR2g/edit?usp=sharing
(I have tried myself  Fysetc tmc2208 V1.0  and biqu tmc2209 v1.1. The rest of tmc2208 uart soldering info comes from telegram channel, friends, or forums).

asxavier, 2021-01-02:
> Is not a marlim problem, is the driver sure. Marlin with SKR V1.3 runs nice in uart mode.
>
> Each brand (and version) of TMC2208 has different settings to enable uart. Its hard sometimes...
>
> I did a [...]

Thank you very much! 
I was struggling to find a solution for my drivers, which are the TMC2208 Bigtreetech V2.0 in your guide. 
Would you mind sharing the source of that part of your guide, so I can learn more before putting everything together and firing up my board?