teemuatlut/TMCStepper #22: Possible Bug with open loop detection

truglodite, 2019-04-24:
Not sure this is a bug with the library...

I recently battled with an intermittent y-motor connector on a pair of TMC2130's on my printer, and was getting some weird symptoms while it was there. First off, everything has been working well, and in fact I'm printing right now... using TMC2130 w/spi and sensorless homing. 

First the problem arose at the start of a print, during a y-move as part of G29 bed leveling. The y-motor stopped short of the probing location, which would result in marlin throwing a "probing failure" error (since I use metal tape targets on a glass bed). After the failure, M122 returned "ola" and "olb" flags for the y-driver. Also, the y-driver register changed to 0's. The open loop flags would be the expected result with an intermittent connector. Not sure if the registers should also go too? It was a bit weird also, how it happened failed at the exact same Y location every time (y-motor connectors are not mechanically coupled, other than usual vibrations I swear).

Also, an even weirder result happened twice among the 50 or so times I induced this problem. 2 of the times the X motor stopped early and failed... m122 showed the x with ola and olb flags?! 

I'm not going to rule out that perhaps my x connectors also have some issues... but this certainly looks like the y motor connector caused a flag on the x driver. I looked in to the motor and diag pin signals with my storage oscope, but unfortunately with my probes attached the problem never happens (don't want to have to break the insulation with pins/skewers, so I used a y-harness that obviously fixes the connector problem). So all I saw was normal behavior on the scope.

Right now, it is printing a print with my probe y-harness attached, and I will replace the y-motor connector when it's done. So this is no big issue that's keeping me from printing, but I'm curious why/how the motor would fail at the same spot without real mechanical coupling of the wires (movement disconnecting the wire)... and if it is possible for an intermittent connector to fire an ol flag for another driver (like my case where x had the flag, but clearly the y-mot connector was the problem). If this isn't a soft bug, I'd love to know what was happening there... got to the point where I nearly kicked my printer in the rear! ;)

teemuatlut, 2019-04-25:
The open load error states don't cause the driver to stop (datasheet 12.3).
The library doesn't actually do anything with the open load indicators and they're just bits read from the driver. So I don't think it's a software bug because there's not software behind this =) Other than the driver firmware of course.

The driver should never return mere zeros if the communication connection is working properly, but the code is not as robust as I'd like it to be, so there's a possibility still.

truglodite, 2019-04-25:
OK thanks for the confirmation; I'll close the issue. 

Even though I'm using a separate 19V for Vmot, there is likely some inductive feedback to Vcc when a motor suddenly disconnect/reconnects. That could easily result in random issues, including a printer halt, and seeing zero registry with communication "ok".

closed 2019-04-25 by truglodite

teemuatlut, 2019-04-25:
You're in an unfortunate situation where the issue seems quite random and has no clear symptom like say not being able to connect. This also means it'll be a pain to diagnose.
Good luck and I hope you'll be able to solve your problems.

truglodite, 2019-04-25:
In a normal world, yes, I'd be up a creek with no paddle. However this, what should be a ghostly issue, is so repeatable I would bet $10 when I get home and remove my y-harness it will repeat. I may grab some waves from a few nodes before I swap connectors. If this is a bug that helps uncover soft/firm issues, it is presenting itself consistently and begging for diagnosis. I'll chime in if I find interesting g data.