teemuatlut/TMCStepper #26: Fast, Medium and slow CRC calculation speed option.

doggyfan, 2019-05-18:
It might help with fast SPI comms (or UART) if you want to reduce the slow CRC calculation bottle neck with the TMC chips.

You can increase the 8-bit CRC calculation by quite a LOT if you wanted, I've written a couple of table driven CRC routines (a fast and a medium speed) ..

```cpp
// define none or one of these
//#define CRC_FAST
//#define CRC_MEDIUM

// used if CRC_MEDIUM or CDC_FAST used
//#define BIT_REV_FAST

#define CRC8_POLY  0xE0    // 1110 0000  X^8 + X^2 + X^1 + X^0
#define CRC8_POLYR 0x07    // 0000 0111  X^8 + X^2 + X^1 + X^0

#if defined(CRC_FAST) || defined(CRC_MEDIUM)
	#if defined(CRC_FAST)
		constexpr uint8_t crc_table8[] =
		{	// fast CRC table
			0x00,0x91,0xe3,0x72,0x07,0x96,0xe4,0x75,0x0e,0x9f,0xed,0x7c,0x09,0x98,0xea,0x7b,
			0x1c,0x8d,0xff,0x6e,0x1b,0x8a,0xf8,0x69,0x12,0x83,0xf1,0x60,0x15,0x84,0xf6,0x67,
			0x38,0xa9,0xdb,0x4a,0x3f,0xae,0xdc,0x4d,0x36,0xa7,0xd5,0x44,0x31,0xa0,0xd2,0x43,
			0x24,0xb5,0xc7,0x56,0x23,0xb2,0xc0,0x51,0x2a,0xbb,0xc9,0x58,0x2d,0xbc,0xce,0x5f,
			0x70,0xe1,0x93,0x02,0x77,0xe6,0x94,0x05,0x7e,0xef,0x9d,0x0c,0x79,0xe8,0x9a,0x0b,
			0x6c,0xfd,0x8f,0x1e,0x6b,0xfa,0x88,0x19,0x62,0xf3,0x81,0x10,0x65,0xf4,0x86,0x17,
			0x48,0xd9,0xab,0x3a,0x4f,0xde,0xac,0x3d,0x46,0xd7,0xa5,0x34,0x41,0xd0,0xa2,0x33,
			0x54,0xc5,0xb7,0x26,0x53,0xc2,0xb0,0x21,0x5a,0xcb,0xb9,0x28,0x5d,0xcc,0xbe,0x2f,
			0xe0,0x71,0x03,0x92,0xe7,0x76,0x04,0x95,0xee,0x7f,0x0d,0x9c,0xe9,0x78,0x0a,0x9b,
			0xfc,0x6d,0x1f,0x8e,0xfb,0x6a,0x18,0x89,0xf2,0x63,0x11,0x80,0xf5,0x64,0x16,0x87,
			0xd8,0x49,0x3b,0xaa,0xdf,0x4e,0x3c,0xad,0xd6,0x47,0x35,0xa4,0xd1,0x40,0x32,0xa3,
			0xc4,0x55,0x27,0xb6,0xc3,0x52,0x20,0xb1,0xca,0x5b,0x29,0xb8,0xcd,0x5c,0x2e,0xbf,
			0x90,0x01,0x73,0xe2,0x97,0x06,0x74,0xe5,0x9e,0x0f,0x7d,0xec,0x99,0x08,0x7a,0xeb,
			0x8c,0x1d,0x6f,0xfe,0x8b,0x1a,0x68,0xf9,0x82,0x13,0x61,0xf0,0x85,0x14,0x66,0xf7,
			0xa8,0x39,0x4b,0xda,0xaf,0x3e,0x4c,0xdd,0xa6,0x37,0x45,0xd4,0xa1,0x30,0x42,0xd3,
			0xb4,0x25,0x57,0xc6,0xb3,0x22,0x50,0xc1,0xba,0x2b,0x59,0xc8,0xbd,0x2c,0x5e,0xcf
		};
	#else	// fast'ish CRC table
		constexpr uint8_t crc_table8_1[] = {0x00,0x91,0xe3,0x72,0x07,0x96,0xe4,0x75,0x0e,0x9f,0xed,0x7c,0x09,0x98,0xea,0x7b};
		constexpr uint8_t crc_table8_2[] = {0x00,0x1c,0x38,0x24,0x70,0x6c,0x48,0x54,0xe0,0xfc,0xd8,0xc4,0x90,0x8c,0xa8,0xb4};
	#endif

	#if defined(BIT_REV_FAST)
		constexpr uint8_t bit_rev_table8[] =
		{	// fast 8-bit reverse table
			0x00,0x80,0x40,0xc0,0x20,0xa0,0x60,0xe0,0x10,0x90,0x50,0xd0,0x30,0xb0,0x70,0xf0,
			0x08,0x88,0x48,0xc8,0x28,0xa8,0x68,0xe8,0x18,0x98,0x58,0xd8,0x38,0xb8,0x78,0xf8,
			0x04,0x84,0x44,0xc4,0x24,0xa4,0x64,0xe4,0x14,0x94,0x54,0xd4,0x34,0xb4,0x74,0xf4,
			0x0c,0x8c,0x4c,0xcc,0x2c,0xac,0x6c,0xec,0x1c,0x9c,0x5c,0xdc,0x3c,0xbc,0x7c,0xfc,
			0x02,0x82,0x42,0xc2,0x22,0xa2,0x62,0xe2,0x12,0x92,0x52,0xd2,0x32,0xb2,0x72,0xf2,
			0x0a,0x8a,0x4a,0xca,0x2a,0xaa,0x6a,0xea,0x1a,0x9a,0x5a,0xda,0x3a,0xba,0x7a,0xfa,
			0x06,0x86,0x46,0xc6,0x26,0xa6,0x66,0xe6,0x16,0x96,0x56,0xd6,0x36,0xb6,0x76,0xf6,
			0x0e,0x8e,0x4e,0xce,0x2e,0xae,0x6e,0xee,0x1e,0x9e,0x5e,0xde,0x3e,0xbe,0x7e,0xfe,
			0x01,0x81,0x41,0xc1,0x21,0xa1,0x61,0xe1,0x11,0x91,0x51,0xd1,0x31,0xb1,0x71,0xf1,
			0x09,0x89,0x49,0xc9,0x29,0xa9,0x69,0xe9,0x19,0x99,0x59,0xd9,0x39,0xb9,0x79,0xf9,
			0x05,0x85,0x45,0xc5,0x25,0xa5,0x65,0xe5,0x15,0x95,0x55,0xd5,0x35,0xb5,0x75,0xf5,
			0x0d,0x8d,0x4d,0xcd,0x2d,0xad,0x6d,0xed,0x1d,0x9d,0x5d,0xdd,0x3d,0xbd,0x7d,0xfd,
			0x03,0x83,0x43,0xc3,0x23,0xa3,0x63,0xe3,0x13,0x93,0x53,0xd3,0x33,0xb3,0x73,0xf3,
			0x0b,0x8b,0x4b,0xcb,0x2b,0xab,0x6b,0xeb,0x1b,0x9b,0x5b,0xdb,0x3b,0xbb,0x7b,0xfb,
			0x07,0x87,0x47,0xc7,0x27,0xa7,0x67,0xe7,0x17,0x97,0x57,0xd7,0x37,0xb7,0x77,0xf7,
			0x0f,0x8f,0x4f,0xcf,0x2f,0xaf,0x6f,0xef,0x1f,0x9f,0x5f,0xdf,0x3f,0xbf,0x7f,0xff
		};
	#else
		constexpr uint8_t bit_rev_table4[]  = {0x00,0x08,0x04,0x0c,0x02,0x0a,0x06,0x0e,0x01,0x09,0x05,0x0d,0x03,0x0b,0x07,0x0f};
		uint8_t reverse_bits8(uint8_t x)
		{	// fast'ish 8-bit reverse
			return (bit_rev_table4[x & 0x0f] << 4) | bit_rev_table4[(x >> 4) & 0x0f];
		}
	#endif
#endif
/*
void makeCRC_tables()
{
	#if defined(CRC_FAST)
		for (uint16_t i = 0; i < 256; i++)
		{
			uint8_t crc = i;
			for (uint8_t j = 8; j > 0; j--)
				crc = (crc & 1) ? (crc >> 1) ^ CRC8_POLY : crc >> 1;
			crc_table8[i] = crc;
		}
	#elif defined(CRC_MEDIUM)
		for (uint8_t i = 0; i < 16; i++)
		{
			uint8_t crc = i;
			for (uint8_t j = 8; j > 0; j--)
				crc = (crc & 1) ? (crc >> 1) ^ CRC8_POLY : crc >> 1;
			crc_table8_1[i] = crc;
		}
		for (uint16_t i = 0; i < 16; i++)
		{
			uint8_t crc = i << 4;
			for (uint8_t j = 8; j > 0; j--)
				crc = (crc & 1) ? (crc >> 1) ^ CRC8_POLY : crc >> 1;
			crc_table8_2[i] = crc;
		}
	#endif
}
*/
uint8_t TMC2208Stepper::calcCRC(uint8_t datagram[], uint8_t len) {
	uint8_t crc = 0;
	uint8_t *data = datagram;

	#if defined(CRC_FAST)
		for (uint8_t i = len; i > 0; i--)
		{
			crc ^= *data++;
			crc = crc_table8[crc];
		}
	#elif defined(CRC_MEDIUM)
		for (uint8_t i = len; i > 0; i--)
		{
			crc ^= *data++;
			crc = crc_table8_1[crc & 0x0f] ^ crc_table8_2[(crc >> 4) & 0x0f];
		}
	#else
		for (uint8_t i = len; i > 0; i--)
		{
			uint8_t c = *data++;
			for (uint8_t j = 8; j > 0; j--)
			{
				crc = ((crc >> 7) ^ (c & 1)) ? (crc << 1) ^ CRC8_POLYR : crc << 1;
				c >>= 1;
			}
		}
	#endif

	#if defined(CRC_FAST) || defined(CRC_MEDIUM)
		#if defined(BIT_REV_FAST)
			crc = bit_rev_table8[crc];
		#else
			crc = reverse_bits8(crc);
		#endif
	#endif

	return crc;
}
```

teemuatlut, 2019-05-19:
Any significant changes in compile size?
I don't want to start adding configurable options inside the library as I'd like the code to be used as is and "just work" from the end user perspective.
You have both fast and medium, do we need both? If fast works, why have medium? Or if fast doesn't work for all, I'd rather just have a medium that works for all.

SPI doesn't use CRC and with UART our issue is closer to running too fast.
Did you time the execution differences in the current implementation compared to your faster code?

doggyfan, 2019-05-19:
oh yeah, compile size depends on which one you use. The fast version compile size is more obviously (256-byte lookup table), that would be the fastest but biggest.

The medium speed CRC and medium bit reverse is a nice compromise, it wouldn't really affect compile size, but you'd certainly spend less time doing the CRC.

I'm not saying you must, was just showing you how to do do CRC somewhat faster and how to create tables for doing so, especially on the AVR.

closed 2019-05-23 by doggyfan