teemuatlut/TMCStepper #30: DRV_STATUS on 2660

ghent360, 2019-05-19:
Hi,

The DRVSTATUS register on 2660 is only 20 bit long. Because SPI reads full bytes, we end up reading 4 extra bits and the DRV_STATUS() method returns value that is shifted 4 bits to the left.

Should the DRV_STATUS() method on 2660 shift the value so bit 0 of the chip register ends up in bit 0 of the return value?

teemuatlut, 2019-05-19:
Can you give an example of a response? I don't have any 2660 drivers hooked up at the moment.

ghent360, 2019-05-19:
When I trip the stall guard DRV_STATUS() returns 0x10.
If my motor is disconnected it reads 0x400.
If I stop driving the step pin, I read 0x800.

teemuatlut, 2019-05-19:
I added a simple shift at the end of read command. Let me know if it helps. The commit is in the master branch.

ghent360, 2019-06-12:
Works fine now.

closed 2019-06-12 by ghent360