teemuatlut/TMCStepper #32: Error compiling TMC2130 in Marlin on SKR 1.3 with TMC_DEBUG enabled

Shiver999, 2019-05-27:
With the update to 0.3.5, I seem to have trouble compiling Marlin 2.0 on the SRK 1.3 with TMC_DEBUG enable for TMC2130 drivers.  Commenting out TMC_DEBUG allows it to complete compiling.

Error:
Marlin\src\feature\tmc_util.cpp: In instantiation of 'void tmc_status(TMC&, TMC_debug_enum) [with TMC = TMCMarlin<TMC2130Stepper, 'X', '0', (AxisEnum)0>]':
Marlin\src\feature\tmc_util.cpp:655:31:   required from here
Marlin\src\feature\tmc_util.cpp:579:38: error: 'class TMCMarlin<TMC2130Stepper, 'X', '0', (AxisEnum)0>' has no member named 'get_pwm_thrs'; did you mean 'en_pwm_mode'?
uint32_t tpwmthrs_val = st.get_pwm_thrs();
~~~^~~~~~~~~~~~
en_pwm_mode

teemuatlut, 2019-05-27:
See https://github.com/MarlinFirmware/Marlin/pull/14134

Shiver999, 2019-05-27:
Ahh, thanks, enabling Hybrid fixed the issue.

closed 2019-05-27 by Shiver999