teemuatlut/TMCStepper #38: `test_mode`

thalesfsp, 2019-06-07:
Hi!

  [Here](https://github.com/teemuatlut/TMCStepper/blob/06468aca644fdfe7dd15c17b0832520ab1ba0fb9/src/source/TMC2208_bitfields.h#L19) it is set to `1`, but [here](https://github.com/teemuatlut/TMCStepper/blob/13465b1b9350ffd562ee5c0d2368024ad3444864/src/source/GCONF.cpp#L53-L60) it says that should be`0` for normal operation. I tried to set `driver.test_mode(false);` and Arduino IDE says: `'class TMC2208Stepper' has no member named 'test_mode'`

Thank you!

teemuatlut, 2019-06-07:
It's a [bitfield](https://www.geeksforgeeks.org/bit-fields-c/), not a value assignment.
So not a bug but a feature of C/C++ language 😊

closed 2019-06-12 by teemuatlut

thalesfsp, 2019-06-24:
@teemuatlut BitField is really cool! Do you `keybase` user?

teemuatlut, 2019-06-25:
Don't know what that is, I guess that would be no =)