teemuatlut/TMCStepper #40: fatal error: SoftwareSerial.h: No such file or directory

D33jay, 2019-06-16:
There seems to be an issue with a missing SoftwareSerial.h file when compiling. 

It starts with:

In file included from .piolibdeps\TMCStepper_ID5513\src\source\CHOPCONF.cpp:1:0:
.piolibdeps\TMCStepper_ID5513\src/TMCStepper.h:14:29: fatal error: SoftwareSerial.h: No such file or directory

referring to TMSStepper.h line 14.

Is there a library that has to be included?

I tried it with the current Marlin2.0.x branch and with my own project.

teemuatlut, 2019-06-16:
You need to have a SoftwareSerial library installed for whatever platform you're compiling for.

D33jay, 2019-06-16:
ah, ok. I was not sure if I have to include a library manually. Thanks man

closed 2019-06-16 by D33jay

thinkyhead, 2019-10-26:
Do we have a guide posted on how to install SoftwareSerial when needed? I wonder how Travis CI is grabbing it for the tests… (Trying to get STM32 builds fixed while we get the new timer code sorted out.)

teemuatlut, 2019-10-26:
Typically the SoftwareSerial library is shipped with the platform, the same as libraries like `SPI.h`.
Until just recently this wasn't the case for STM and so the platform wasn't considered SW Serial capable. Then the library got included in the platform package but it requires extra work on PIO setup to use that update and then you'll run into new compile issues.
Another thing that piles on top of the issues is there are two STM platforms for PIO with Arduino frameworks. `STM32Duino` has the library included but "Maple" supposedly doesn't.
Overall it's a headache I'll gladly leave for more invested parties.

thinkyhead, 2019-10-26:
That's an annoying change. Do you have a link to the extra steps we need to do so I can get our STM32 test builds (and end-user builds) working again?

teemuatlut, 2019-10-26:
Basically you'll need to upgrade to the newer platform version and then sort out the compile issues.
https://github.com/MarlinFirmware/Marlin/pull/15655

For now I'd just disable the STM tests and test them locally outside of Travis.

thinkyhead, 2019-10-26:
At the moment the number of nonworking STM32 builds is very annoying. Can't build STM32F4, STM32F7, ARMED, BIGTREE_BTT002, or BIGTREE_SKR_PRO. I'm more inclined to fix them all at an older version like 5.4.3 that wasn't as broken. We will have to set fixed versions of most libraries for our tagged releases anyway.

teemuatlut, 2019-10-26:
If you wish to use an older version, then you need a working external SoftwareSerial library.
But at least with the old version you can have the tests complete but building for STM+TMCs will fail due to the missing dependency.

MagoKimbra, 2019-10-26:
The new TMC library has added support for STM including the softwareserial library included in the new STM library. So either they update both or go back both for TMC and for STM otherwise you would always have errors.