teemuatlut/TMCStepper #42: SPI Cascade Mode

bdring, 2019-06-16:
There was talk of supporting SPI cascade mode in the TMC2130 repo. I do not see that in the API. Is that still a possibility?

It would be a great enhancement. I am trying to implement 5+ drivers on an I/O limited ESP32.

Thanks for your great work!

teemuatlut, 2019-06-17:
It's a possibility but not a priority.
I could try putting together another branch for you to test though. I don't currently myself have a lot of time to do testing.

bdring, 2019-06-17:
Thanks, that would be great. I am happy to do testing. I have had good success using the library in normal SPI mode.

bdring, 2019-07-03:
[image]

I made a controller to test the daisy chain SPI mode. I did a quick hack to the library to test the layout. It just loops 4 times inside the `write(...)` function. Therefore, all drivers get the same configuration. It works great so far. I only have (2) drivers, so I need to order a few more.

@teemuatlut If you have a design pattern in mind to handle this in your library, I would be happy to hack in some test code.

bdring, 2019-07-04:
I am trying a new method of daisy chaining with minimal impact on the library.

I added an `axis_index` and `axis_count` variable to the TMCStepper object. They have defaults that cause the library to work in its traditional mode. 

If you use daisy chaining, you do a `set_axis_count(...)` to something above 1. You also select the axis you want to talk/listen to using `set_axis()`.

The `write` and `read` functions are rewritten work in the traditional mode when the `axis_count  == 1` and target the `axis_index` driver when the `axis_index` is greater than 1.

The `axis_index == 0` is the first driver in the chain.

This requires more data to be transferred than targeting all drivers at once, but it is probably fine. You probably are more interested in saving data lines than transfer speed if you decided to use daisy chain mode.

bdring, 2019-07-08:
The new method works for SPI write. I think there are still some issues with SPI read. I have a [fork of the TMCStepper](https://github.com/bdring/TMCStepper) library with my changes.

I have [a branch of Grbl_ESP32](https://github.com/bdring/Grbl_Esp32/tree/TMCStepper) that works with it.

I am now working on adding 3 more axes (A, B & C) to Grbl_ESP32.