teemuatlut/TMCStepper #45: Compatibility of TMC2209 with Linear advence on Marlin

Gerggg, 2019-06-19:
Does the addition of the TMC2209 correct the compatibility problem we encounter between TMC2208 and linear advence? as a reminder, the extruder motor shuts off when TMC2208 is used with Marlin's Linear Advence feature.

teemuatlut, 2019-06-19:
No there should be no difference in that.
Square step mode would have a much better likelihood of correcting the behavior.

Gerggg, 2019-06-19:
Thank you for your answer, can you specify the option to activate the Square step mode ?

teemuatlut, 2019-06-19:
https://github.com/MarlinFirmware/Marlin/blob/dd06017f642b0396389e85796e447ecee47fdd4b/Marlin/Configuration_adv.h#L1833-L1837

Gerggg, 2019-06-20:
Great, I'll try this. Thank you.

drewzh, 2019-06-21:
> Great, I'll try this. Thank you.

Out of curiosity, as I'm about to start looking at linear advance - did you have any luck?

Gerggg, 2019-06-21:
yes, it works perfectly well.  I just have a strange noise when retracting but otherwise it's absolute silence.

Gerggg, 2019-06-22:
good sorry but if the noise level is good, the level of inpression quality I did not have a good result with the TMC2209 (a lot of layer shift)

Gerggg, 2019-07-05:
The problem has been fixed since version 0.4.5.

closed 2019-07-05 by Gerggg