teemuatlut/TMCStepper #46: problem update

gauix4, 2019-06-20:
[image]

I do not understand I'm fine doing PIO update it is well updated but when I debug it does not take into account

teemuatlut, 2019-06-20:
Check your `platformio.ini` file.

gauix4, 2019-06-20:
@teemuatlut I left the original TMCStepper@<1.0.0
I tested with https://github.com/teemuatlut/TMCStepper/archive/master.zip
it does not change much

gauix4, 2019-06-20:
[image]

teemuatlut, 2019-06-20:
Try deleting the .pioenvs and .piolibdeps folders. That should prompt PIO to redownload all dependencies.

gauix4, 2019-06-20:
@teemuatlut  I do not have access to his files on Mac by searching a little on Google and on the right to the left I just found that Atom is not compatible with Mac above IOS10

gauix4, 2019-06-20:
@teemuatlut  it does not change anything

teemuatlut, 2019-06-20:
I see you've highlighted the line 35 on the pio file, but that's not the section LPC is using.
Go further down to LPC1768 section and make the changes there.

https://github.com/MarlinFirmware/Marlin/blob/4df4c47994f248428ae30218800803e467327b6e/platformio.ini#L149-L169

gauix4, 2019-06-20:
@teemuatlut yes yes I know I had just taken the first one that had come

gauix4, 2019-06-20:
I think it will go so fast to take back a blank Marlin and complete it

teemuatlut, 2019-06-20:
You may also encounter issues using PIO on Atom and switching to Visual Code is recommended as it is their preferred platform.

gauix4, 2019-06-20:
@teemuatlut yes that's what I saw more I'm on Mac and they stopped the development at iOS10

gauix4, 2019-06-20:
@teemuatlut  after several tests the TMC5160 that I activate or not STEALTHCHOP it does not change anything makes noise

gauix4, 2019-06-20:
@teemuatlut 
first start the engines are silent you turn as I should disconnect the power I reconnect they make a lot of noise and nothing works.
I have beautiful re flash they make a lot of noise now
it is to understand nothing

closed 2019-07-03 by gauix4