teemuatlut/TMCStepper #47: Unable to get Stallguard example working with Trinamic's TMC2130-eval board

platisd, 2019-06-26:
First of all, thanks for your great work on this library and the support you are providing! :+1: 

I am trying to get stallguard detection to work using this library and the [TMC2130-eval board by Trinamic](https://www.trinamic.com/fileadmin/assets/Products/Eval_Drawings/TMC2130-Eval_v13_01_Schematic.pdf), but I run into some problems and eventually fail to detect any stalling as the related value always seems to be `0`. Before I continue I should note that using Trinamic's evaluation software ("TMCL-IDE") produces the desired results.

To begin with, the example `Stallguard.ino` does not compile out of the box using Arduino IDE 1.8.9 when compiling for an Arduino Mega. Specifically, several variables are never declared. To get it to compile, one would need to add the following declarations somewhere in the beginning:

```cpp
constexpr uint8_t SG_RESULT_bp  = 0;
constexpr uint8_t CS_ACTUAL_bp  = 16;
constexpr uint32_t SG_RESULT_bm = 0x3FFUL;
constexpr uint32_t CS_ACTUAL_bm = 0x1F0000UL;
```
I assume the values are correct since I took them from the older `TMC2130Stepper` library.
Moving on, the next necessary change (naturally) is adjusting the pin configurations accordingly for my setup. Specifically, now it looks like this:
```cpp
#define EN_PIN        21    // Enable
#define DIR_PIN       20    // Direction
#define STEP_PIN      19    // Step
#define STEP_PORT     PORTD // Match with STEP_PIN
#define STEP_BIT_POS  2     // Match with STEP_PIN
#define CS_PIN        37    // Chip select
```

After compiling and running this, the first problem I encountered was that the motor would not move at all. Looking at the step signals sent to the board, what ultimately did the trick was changing the *order* of the pin toggling within the interrupt. It looks like this now:
```cpp
ISR(TIMER1_COMPA_vect){
  STEP_PORT &= ~(1 << STEP_BIT_POS);
  STEP_PORT |= 1 << STEP_BIT_POS;
}
```
Now the motor is moving, however the printout I get is always something like `0 0 381` where the last value is the current and the second one should indicate the stall.

Now that I have hopefully explained my setup, my questions are the following:
* Should the example sketch compile without declaring those constants? And if so, am I missing something here? (using the latest `master` of this repository)
* Should the change in the *order* of the pin toggling be necessary? Is there something special going on with this [evaluation board](https://www.trinamic.com/support/eval-kits/details/tmc2130-eval/) I am using?
* Finally, any ideas/hints on why do I get the observed behavior?

Any help would be tremendously appreciated! :small_airplane:

teemuatlut, 2019-06-26:
The stallGuard example was copied over from TMC2130 as-is and it looks like it needs a small update to be compatible with the new architecture.

Changing the pin write order means that the pin is HIGH for the most of the time and then briefly pulses LOW within the ISR. A better solution would be to toggle the pin.

platisd, 2019-06-26:
Thanks for your response. Good to know that the example may need to be modified to be compile-able. If you want, I can create a pull request with the "fix" later.

In regards to the pin toggling, yes, I've tried that and it also makes the motor move. My main problem, in both cases, is the stall value being always returned as `0` with the current hypothesis being there is some, subtle but very essential, difference between the official board by Trinamic and the boards everyone else online is using.

Any input in regards to that?

teemuatlut, 2019-06-26:
I'd try printing out the full response to make sure it's not zero. That would indicate a possible comms issue.

platisd, 2019-06-26:
Judging by the current reading changing (as expected pretty much) when the motor is running and when it isn't, I'd assume that SPI communication is working. When printing the `DRV_STATUS` result (along with the other info printed by the sketch I get:

> 0 0 381
DRV_STATUS: 00000000000100001000000000000000

teemuatlut, 2019-06-26:
I updated the stallguard example and made new ones for TMC2209/2660.
I don't think it will solve your problem but at least it shows how to parse information using the register data structs.

platisd, 2019-06-27:
Thanks for the update. The example indeed compiles now. :+1: 
I got a SilentStepStick to try with the example and it works exactly as intended, so there is indeed something different/strange in this `TMC2130-eval` board. :disappointed: 
I am insisting on using this evaluation board since I have a custom board, designed using the evaluation board as a reference, which (perhaps naturally) also does not work. Anyway, will keep digging into this but if you think of something feel free to let me know. :smiley:

closed 2019-06-27 by platisd

platisd, 2019-06-27:
For anyone ever observing similar problems, the problem was eventually nailed down to *not* having grounded some configuration chips on the TMC2130 of the boards in question. :+1: