teemuatlut/TMCStepper #48: Only including SoftwareSerial when actually necessary?

kinsi55, 2019-06-30:
After switching to TMC2130's today I've been greeted by [this limitation](https://github.com/MarlinFirmware/Marlin/pull/12606) of MarlinFW where I cannot use Interrupts for the axis endstops and would have to fallback to polling.

Yes this would theoretically be fine since the TMC supports endstop-less homing, but I'm only using them for the X and Y axis (And I'd like to keep using the mechanical stoppers because they're there anyways)

Since I dont actually need to make use of SoftwareSerial, in my case this was easily resolved by simply harcoding `SW_CAPABLE_PLATFORM` to false in the library and commenting out Marlins Sanity check from the linked PR, but this obviously is not really a decent solution, especially none that could be contributed.

Would there maybe be some possibility to have an alternative include to the main one with SoftwareSerial disabled? Then in Marlin it could evaluate wether it actually needs it and include the appropriate library instead of hardcoding an error due to missing control.

teemuatlut, 2019-07-11:
Any Marlin configuration pre-processor definitions cannot be passed to external libraries.
You'll run into issues when you'd get to the library internal compile units like `TMC2208Stepper.cpp` which expects SW Serial capabilities based on the target framework/platform, but it has no way to know which features are enabled in the application.

kinsi55, 2019-07-11:
Almost thought so, I wasnt able to come up with a way to have it work trough a singular include either. I'm not really all that much into C, but maybe it would be possible / make sense to have a seperate library without SWS included, and then Marlin would take care of including it if necessary?

closed 2019-09-05 by teemuatlut

lonelymyp, 2020-09-24:
@kinsi55 How did you do this?
I am using tmc2209 for x and y axes and want to autolevel the z-axis with interrupts.
Can you tell me how to get around this error in the settings?