teemuatlut/TMCStepper #56: error: no matching function for call to TMC2209Stepper

TheZoker, 2019-08-01:
I'm not sure, if this is a Marlin or TMCStepper issue.

When I try to compile Marlin for my SKR E3 DIP with TMC2209, I get the following error (next to a lot of warnings):
[image]

Here is the complete log:
https://gist.github.com/TheZoker/036b74969940e4888068b0baee3f761c

TheZoker, 2019-08-05:
@teemuatlut Any idea about this?

teemuatlut, 2019-08-05:
v0.4.6 should fix this when it hits PIO library manager in a few days.
If you don't want to wait, you can update your `platform.ini` to get the latest version with a direct git link.

The underlying reason is that STM32 wasn't yet defined as a Software Serial supported platform.

TheZoker, 2019-08-05:
@teemuatlut Thanks! Now I directly tagged the master branch and plattform.io installed version 0.4.6
But now I get this error:
```
In file included from Marlin/src/module/stepper_indirection.h:47:0,
                 from Marlin/src/module/stepper.h:221,
                 from Marlin/src/Marlin.cpp:37:
.piolibdeps/BIGTREE_SKR_MINI/TMCStepper/src/TMCStepper.h:18:11: fatal error: SoftwareSerial.h: No such file or directory

************************************************************************
* Looking for SoftwareSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SoftwareSerial.h"
* Web  > https://platformio.org/lib/search?query=header:SoftwareSerial.h
*
************************************************************************

  #include <SoftwareSerial.h>
           ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

Here is my config:
https://github.com/TheZoker/Marlin/commit/dc3f0171feec3c67657cc0d491ffbd72c12b13bb

TheZoker, 2019-08-08:
@teemuatlut Any idea where this issue comes from?

teemuatlut, 2019-08-08:
It's trying to include a SoftwareSerial library that doesn't exist on your computer.
Maybe @GerogeFu has a suggestion which one to use.

TheZoker, 2019-08-08:
Alright, thanks
But isn't the purpose of with dependencies (/libraries), that everything needed to build the .bin file is handled by platformio?

teemuatlut, 2019-08-08:
Yup, but that part hasn't been specified yet. All previous platforms, pre STM, have had the SoftwareSerial library included in the framework.

TheZoker, 2019-08-08:
Seems like this could fix the issue, right? 
https://github.com/MarlinFirmware/Marlin/commit/b76bf93a2848d309192ba477fb421c5a52a57bcf

salacpavel, 2019-08-20:
So its just to confirm I have compiled succesfully today on very same platform, but there is "TMC CONNECTION ERROR" on screen now and M122 reports all 4 UART comm beeing down.