teemuatlut/TMCStepper #59: ESP32 is no SoftwareSerial Compatible Platform

p0wl, 2019-08-12:
Hi!

first of all, thanks for your great library. We are using it to configure 3xTMC2209 using UART from a ESP32 and it's working quite nice.

The problem is that we have to monkey patch TMCStepper.h because it tries to include SoftwareSerial.h. I saw your latest commit that would probably fix it, but I'm questioning why you included ESP32 as a `SW_CAPABLE_PLATFORM` in the first place (in https://github.com/teemuatlut/TMCStepper/commit/0d11a14affd101da42087cb314f12e5d46a55b0e). I read all the issues in the repository about it, but I did not find anyone who said that it was needed. It was expressed by @bdring that it caueses problems for her/him as well: https://github.com/teemuatlut/TMCStepper/issues/35#issuecomment-503322972

Also, there are other libraries who are explicitly NOT using SoftwareSerial on an esp32: https://github.com/brianrho/GT5X/blob/54fe601441d1d962da2fc8e5f081a0dd5acebf37/examples/count_templates/count_templates.ino#L1-L5

I would suggest removing it from the list of `SW_CAPABLE_PLATFORM`s. What do you think?

vivian-ng, 2019-08-15:
@teemuatlut I agree with @p0wl, please consider removing ESP32 from the list of SW serial capable platforms. Alternatively, include a note about a compatible SoftwareSerial.h implementation for the ESP32 that can work with this TMCStepper library. Or maybe a [dummy SoftwareSerial.h](https://github.com/MarlinFirmware/Marlin/pull/14861) like what was done for STM32F1 in Marlin.

Or just keep this issue open so that anyone with problems will come across it. 😅

p0wl, 2019-08-16:
Created a PR, this is the only required change: https://github.com/teemuatlut/TMCStepper/pull/62

closed 2019-08-18 by teemuatlut

p0wl, 2019-08-19:
Awesome, thanks for merging!

kirito2908, 2023-10-21:
Well, ESP is not included in SW_CAPABLE_PLATFORM in my case but for some reason i can't set current for 3 stepper motors with 3 individual TMC2209. I can make them move fine with this library but i want to set current for each Stepper motor. Have any idea about what's wrong ? Or what do i need to give so anyone can help me ? Please let me know.

PR #62 (2019-08-18): Remove ESP32 from list of SW_CAPABLE_PLATFORMs
since there is no default SoftwareSerial library for the ESP32

fixes #59