teemuatlut/TMCStepper #61: Knocking noise when moving TMC2208 (UART) and MKS GEN L

michaelsheridan23, 2019-08-15:
This has already been brought up in the Marlin GitHub, https://github.com/MarlinFirmware/Marlin/issues/14183.

Has anyone on this hub seen this issue? Or have a resolution for this issue? 

This only occurs with Marlin 2.0.x and the TMCStepper library. With Marlin 1.1.9 and the TMC2208 library the issue does not occur. The current work around is to disable  "“//#define MONITOR_DRIVER_STATUS" function. This gets the knocking noise to go away but also disables the ability to utilize the UART features.

teemuatlut, 2019-08-15:
Will you be able to provide test information if I create a branch for you try out which reverts many of the UART changes introduced by TMCStepper?
It may not happen today but perhaps during this week.

michaelsheridan23, 2019-08-15:
I would be happy to help.

michaelsheridan23, 2019-09-03:
@teemuatlut have you made any updates you would like me to test?

teemuatlut, 2019-09-04:
https://github.com/teemuatlut/TMCStepper/tree/revert_uart

michaelsheridan23, 2019-09-04:
Does this remove the UART capability? Or does it revert it to the old settings? I am not sure, the readme was a little light on details.

teemuatlut, 2019-09-04:
It reverts the read/write methods to what they are in TMC2208Stepper library which is used by Marlin 1.1.9.

michaelsheridan23, 2019-09-04:
Thanks for the clarity. Hopefully I will get the chance to try it tonight.

michaelsheridan23, 2019-09-06:
Unfortunately, the "_revert" files did not fix the issue. There is still a consistent thumbing when moving any axis with TMC2208s and UART Enabled.

jacekk015, 2019-10-03:
@teemuatlut @michaelsheridan23 
[Marlin-bugfix-2.0.x_Knocking_recent_30_09_2019.zip](https://github.com/teemuatlut/TMCStepper/files/3688236/Marlin-bugfix-2.0.x_Knocking_recent_30_09_2019.zip)
[Marlin-bugfix-2.0.x_OK_April_2019.zip](https://github.com/teemuatlut/TMCStepper/files/3688237/Marlin-bugfix-2.0.x_OK_April_2019.zip)

I've used Marlin 2.0 already with TMC2208 and MKS GEN L.
Same knocking happens after recent update of Marlin - I've added BLTouch so it was an occasion to update whole software package.
Last working normally Marlin 2.0 has Marlin.ino dated 13 April 2019.

BTW. After I've started 2 days ago ArduinoIDE it asked me to update libraries so right now I have 0.5.1 TMCStepper version.

The steppers knocking is present only on new Marlin 2.0 with TMCStepper 0.5.1
My old Marling 2.0 with new TMCStepper 0.5.1 works OK
So it looks like it's not library dependable issue.
All stepper knocks: X, Y, Z

Attached both ZIP packages. I've reuploaded right now the old April package with BLTouch support - no knocking - even if it's compiled using recent TMCStepper library.

michaelsheridan23, 2019-10-07:
[image]
[image]

Took a quick comparison of some of the files using Beyond Compare and these sections look interesting. But honestly I have no idea what I am looking at. Hopefully someone smarter than me can find something more conclusive.

jacekk015, 2019-10-07:
@michaelsheridan23 I've seen solutions saying to disable TMC_DEBUG

michaelsheridan23, 2019-10-08:
I can confirm with @jacekk015 the April 12, 2019 commit, Tree #57aaa9110b, does not exihibit the knocking noise with the latest version for TMCStepper, 0.5.1. Here are the files configured for my ender 3.
[Marlin-bugfix-2.0.x - MKSv1 Ender 3 - 2019-04-12.zip](https://github.com/teemuatlut/TMCStepper/files/3701905/Marlin-bugfix-2.0.x.-.MKSv1.Ender.3.-.2019-04-12.zip)

teemuatlut, 2019-10-16:
Thanks. Knowing a specific commit is actually something I can work with and compare the changes in the file where the code is executed. Also knowing the platform narrows things down.
If disabling `TMC_DEBUG` helps the issue, it further rules out parts of the code.

jacekk015, 2019-10-16:
@teemuatlut TMC_DEBUG disable don't help anything - just checked myself with downloaded version from 12.10.2019.
I probably mistaken TMC_DEBUG with
`#define MONITOR_DRIVER_STATUS`
But I can't disable this since my TMC2208 runs on UART.
However I can check this personally in a minute.

teemuatlut, 2019-10-17:
And what about the effects of `TMC_DEBUG` with monitoring enabled?

jacekk015, 2019-10-17:
There’s no effect at all

token47, 2019-12-16:
I'm also suffering of this problem. Migrated from 1.1.9 to 2.0.0, configs are roughly equivalent but got that knocking noise and a bed leveling weirdness. Spent almost three hours debugging and changing options one by one until discovered that disabling "MONITOR_DRIVER_STATUS" makes the problem go away.

I'm willing to make tests if needed.

jacekk015, 2019-12-16:
@token47 
Without MONITOR_DRIVER_STATUS you can't use hybrid mode. So better stick with the old build.

token47, 2019-12-16:
I'm not using hybrid mode anyway. I'm using fixed stealthchopv2 on X and Y and spreadcycle on E (Z is still an A4988). But it was nice to have the current, temperature and connection monitoring and be able to see detailed status on M122.

rjaros87, 2019-12-16:
@jacekk015 so from my understanding, solution for this issue is to migrate to the new board? eg. 32-bit? To have Marlin 2.0 + TMC2208 (and newer).

jacekk015, 2019-12-16:
@rjaros87 Can't tell you that, since I have a 8-bit MKS GEN L, but that's what they think about here:
https://github.com/MarlinFirmware/Marlin/issues/14183

BTW I just stayed with my 2.0.0 April build - it just works.

token47, 2019-12-16:
Let me take the opportunity to ask two things:
1) is it expected that SPI boards (like 2209) will not suffer from this issue? They would use hardware SPI and just extra CS pins.
2) In the specific case of 2208, in my board, it uses sense resistors labeled R100, but in marlin config X_RSENSE is default 0.11 (which would mean 10% more). Anyone else noticed this?

teemuatlut, 2019-12-17:
Driver monitoring isn't required for Hybrid mode.
You should match the `_RSENSE` values in the configuration to your physical hardware.
2209 is UART, not SPI and so has the same limitations as the 2208.

There may be some things that can be adjusted to mitigate the issue but first I would need to know if the issue is with Marlin or with TMCStepper. Both introduce changes that may affect this behavior, but I've looked at the changes on Marlin side and didn't see anything immediately suspicious. Then again, it's also been said here that the issue goes away with an earlier build of Marlin 2.0 but still using the current TMCStepper version.

michaelsheridan23, 2019-12-17:
I can confirm it is likely with Marlin. I have tested the April build with the latest TMCStepper and it works fine. However any newer build of Marlin causes the issue again.

Miraculix200, 2020-05-15:
Is there any fix for this problem yet? Except turning off MONITOR_DRIVER_STATUS?

I'm getting that noise on a Robin Nano (STM32F1) with 4 TMC2209 stepper drivers. The stepper drivers each get a dedicated pin on the board, where both RX and TX are connected to (UART, software serial). 

Turning off MONITOR_DRIVER_STATUS removes the noise, and changing MONITOR_DRIVER_STATUS_INTERVAL_MS also changes the interval of the noise.

I doubt it is helpful, but there is a difference between the two TMC2209s from Fysetc (v2.1) and the two 2209s from Watterott. With the ones from Fysetc the noise is barely audible. You have to listen very closely and know what you're listening for to hear it. With the ones from Watterott the noise is really loud. Both types of 2209s are running in SpreadCycle mode.

teemuatlut, 2020-05-15:
The noise should be only with AVR boards so you'll have to take your issue with whomever made the serial implementation you're using.