teemuatlut/TMCStepper #69: Constructor empty in library. Please.

MagoKimbra, 2019-09-04:
Hi, I'm the MK4duo programmer.
I need to create a public TMC2130 stepper class driver and then create an array of some normal and other 2130 drivers.
To do this in the driver class costructor I call the TMC2130steper library for normal queries I don't want to send any parameters. To do this I need the empty constructor in the TMC2130 costructors and other drivers as well.
I have tried modifying the downloaded libreia and it works, but I would like it to be also on the online one in this way the users who download it have no problems.
For example TMC2130Stepper constructor:
TMC2130Stepper () {}           <--------- This
TMC2130Stepper (uint16_t pinCS, float RS = default_RS);
TMC2130Stepper (uint16_t pinCS, uint16_t pinMOSI, uint16_t pinMISO, uint16_t pinSCK);
TMC2130Stepper (uint16_t pinCS, float RS, uint16_t pinMOSI, uint16_t pinMISO, uint16_t pinSCK);

And This:
TMCStepper() {}        <--------- This
TMCStepper(float RS) : Rsense(RS) {};

My example:

```
#include <TMCStepper.h>

class driver {
  public:
    driver(int PIN_ENABLE, int PIN_DIR, int PIN_STEP) :
      pin_enable(PIN_ENABLE),
      pin_dir(PIN_DIR),
      pin_step(PIN_STEP)
    {}
 
    int pin_enable;
    int pin_dir;
    int pin_step;
};

class mkdriver : public TMC2130Stepper, public driver {

  public:
    mkdriver(int PIN_ENABLE, int PIN_DIR, int PIN_STEP) :
      driver(PIN_ENABLE, PIN_DIR, PIN_STEP)
    {}
 
    mkdriver(int PIN_ENABLE, int PIN_DIR, int PIN_STEP, int CS_PIN, float RS) :
      driver(PIN_ENABLE, PIN_DIR, PIN_STEP),
      TMC2130Stepper(CS_PIN, RS)
    {}

  public:
    void test() {}
};

mkdriver* driver[10];

void setup() {
  driver[0] = new mkdriver(5, 6, 7);            // Normal driver
  driver[1] = new mkdriver(8, 9, 10, 10, 0.1);  // TMC2130 Driver
}

```

Thank's for your response...

teemuatlut, 2019-09-04:
Sure I can create a protected constructor like that, but as an alternative could it be done with something like:
```cpp
#include <TMCStepper.h>

class driver {
...
};

class mkdriver : public driver {

  public:
    mkdriver(int PIN_ENABLE, int PIN_DIR, int PIN_STEP) :
      driver(PIN_ENABLE, PIN_DIR, PIN_STEP)
    {}
 
    mkdriver(int PIN_ENABLE, int PIN_DIR, int PIN_STEP, int CS_PIN, float RS) :
      driver(PIN_ENABLE, PIN_DIR, PIN_STEP)
    {
          TMC2130Stepper drv = new TMC2130Stepper(CS_PIN, RS);
          tmc_instance = &drv;
    }

  public:
    void test() {}
    TMC2130Stepper *tmc_instance = NULL;
};

mkdriver* driver[10];

void setup() {
  driver[0] = new mkdriver(5, 6, 7);            // Normal driver
  driver[1] = new mkdriver(8, 9, 10, 10, 0.1);  // TMC2130 Driver
}
```

This would allow you to also later test if the driver is a TMC driver by checking the pointer against null.
The downside I suppose would be that you'll have to write methods for accessing the underlying TMC class.

I don't know if that compiles or not but it's quite close to what the library does with SoftwareSerial.

MagoKimbra, 2019-09-04:
O very good... Yes this is a fast solution.
In MK4duo have TMCStepper class for methods accessing TMC stepper, now for access i use driver[1].drv.function...
For the check if driver has tmc or not i can use  if (driver[1].drv) driver[1].drv.function.
Thank's...

closed 2019-10-24 by MagoKimbra