teemuatlut/TMCStepper #74: TMS2208/9 uart: too much delay after write

ghent360, 2019-09-14:
Hi,

I'm trying to use hardware serial port on STM32 in half-duplex mode. This uses a single pin on the chip for both TX and RX. (see PR #643 on stm32duino/Arduino_Core_STM32).

I'm encountering the issue that after the write operation is complete there is delay(replyDelay) (which is ~2ms). This causes the receiver not to be turned on and the receiver bytes are missing.

Does this delay need to be there? In my experiment by default the data follows after ~2-3 clocks on the serial port which is several uS.

If I do driver.senddelay(14) the delay becomes ~1ms max, so IMO 2ms delay is way too much.

This would not be an issue with a dedicated RX line, because the receiver is always on and even if the data comes fast it gets buffered by the software uart layer.

Alas for half-duplex mode I have to switch the receiver on immediately after the transmission of the last byte is complete to avoid missing the incoming data.

I'm have removed two calls to delay(replyDelay) and added the following changes:

uint64_t _sendDatagram(SERIAL_TYPE &serPtr, uint8_t datagram[], const uint8_t len, uint16_t timeout) {
        while (serPtr.available() > 0) serPtr.read(); // Flush
        for(int i=0; i<=len; i++) serPtr.write(datagram[i]);
-       delay(TMC2208Stepper::replyDelay);
+       int16_t res = serPtr.read();
 
        // scan for the rx frame and read it
        uint32_t ms = millis();
        uint32_t sync_target = ((uint32_t)datagram[0]<<16) | 0xFF00 | datagram[2];
-       uint32_t sync = 0;
+       uint32_t sync = (res > 0) ? res : 0;
 
        do {

Does this sound reasonable? I can prepare a PR if there are no objections.

teemuatlut, 2019-09-14:
I'm not against a change like this, but it needs to be tested with AVR and LPC as well, as they are the two main platforms and I definitely wouldn't want to break those.
You'll need to explain the logic behind your changes. Both the added lines seem unnecessary.

ghent360, 2019-09-25:
I don't have AVR board with pluggable steppers. I ordered one and I'll test there and on SKR 1.3 (LPC) when it arrives.

The reasoning for the changes is as follows:
In half duplex mode the STM32 IP needs to switch receiver or transmitter before data can be transmitted in the corresponding direction.

For sending data it is easy - every time you write, firs you check if the transmitter is enabled, if not enable it, then write as usual.

For receiving data, enabling the receiver is done on the first call to the read() method. In that case the result from the read() method would be -1, because we just turned the receiver on and there is no data yet. However this call needs to be done as soon as we are done writing data, any delay would risk the data to be sent back with the receiver off.

Moving the setPrt.read() immediately after the "for" cycle accomplishes that goal. The rest of the core is to make sure that in case of full duplex communication we don't drop the first byte.

teemuatlut, 2019-09-25:
Here's how the code then breaks down
```cpp
for(int i=0; i<=len; i++) serPtr.write(datagram[i]);

int16_t res = serPtr.read();
uint32_t sync = (res > 0) ? res : 0;

do {
	int16_t res = serPtr.read();

	sync <<= 8;
	sync |= res & 0xFF;
	sync &= 0xFFFFFF;

} while (sync != sync_target);
```
If I do the same omissions to upstream code, plus remove the delay as per your suggestion, then it looks like this:
```cpp
for(int i=0; i<=len; i++) serPtr.write(datagram[i]);

uint32_t sync = 0;

do {
	int16_t res = serPtr.read();

	sync <<= 8;
	sync |= res & 0xFF;
	sync &= 0xFFFFFF;

} while (sync != sync_target);
```
This is what I mean with the _added lines_ seeming unnecessary =)

ghent360, 2019-10-04:
In addition to my STMF4 board I did test the code on SKR 1.3 (LPC1768) as well as MKS GEN L v2.0 (ATmega256). TMC2209 drivers work on both boards and I was able to change the driver current settings.

My main point was removing the two calls to delay(TMC2208Stepper::replyDelay). I'll trust your judgement on the rest of the proposed changes.