teemuatlut/TMCStepper #75: Bigtreetech changes to TMC2208 SW_SERIAL implementation

salacpavel, 2019-09-14:
Hello,

could it be possible to merge changes Bigrtreetech comited to their fork in order to your main tree, so we (SKR users) are not stuck with 0.4.5 forever because of their inabilit to issue a proper pull request?
This is the only change they made: https://github.com/bigtreetech/TMCStepper/commit/b43f65aac27316504305808262bff5e817223320

teemuatlut, 2019-09-14:
They are testing if an *unsigned integer* is greater than a negative value.

I'm planning to do away with write only mode anyway but I'd need to test if the simple change works with AVR first before releasing it as a feature. Basically platform agnostic half-duplex capabilities.
I know what they're trying to do and I might make the change as a temporary step if that solves some issues.

salacpavel, 2019-09-15:
Awesome :) Tank you, sir!

AbelCS, 2019-10-15:
> They are testing if an _unsigned integer_ is greater than a negative value.
>
> I'm planning to do away with write only mode anyway but I'd need to test if the simple change works with AVR first before [...]

It would be great for all SKR users.

LinoBarreca, 2019-10-22:
No wait @salacpavel is wrong.
What BTT did in reality is to include a SoftwareSerial implementation for stm32

https://github.com/bigtreetech/TMCStepper/blob/master/src/source/SW_UART.h (&cpp)

I think this can be solved at project level (references) into Marlin itself.

LinoBarreca, 2019-10-22:
addendum:
use this workaround:
edit platformio.ini to add
`  SoftwareSerialM=https://github.com/FYSETC/SoftwareSerialM/archive/master.zip`
to lib_deps in the BIGTREE_SKR_PRO section
Open the "TMCStepper.h" in the library folder of your build environment (where you download libraries, mine is in C:\Users\[name]\.platformio\lib\TMCStepper[XYZ])

Edit line 
`#if (__cplusplus >= 201703L) && defined(__has_include)`
into
`#if defined(__has_include)`

Save.
Build.

extesy, 2019-10-22:
@LinoBarreca Take a look at https://github.com/stm32duino/Arduino_Core_STM32/pull/645

LinoBarreca, 2019-10-23:
> 
> @LinoBarreca Take a look at [stm32duino/Arduino_Core_STM32#645](https://github.com/stm32duino/Arduino_Core_STM32/pull/645)

I see it has been merged to official stm32duino.
Problem here is that the ST STM32 platform (on PlatformIO) uses package framework-arduinoststm32 but the version 3.10601.190716 hints that it has been built on 16 july 2019.
Probably it's just a matter of time 'till they update the framework (or maybe we can open a feature request for it). Meanwhile using another SoftwareSerial implementation works just fine.