teemuatlut/TMCStepper #77: Cannot Define Software Spi in 0.5.1

peaceblackhair, 2019-09-27:
Hello,

`TMC2130Stepper driver = TMC2130Stepper(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK); // Software SPI`

i cannot define software spi in 0.5.1 version.I'm getting this error 

`src/main.cpp:28:81: error: call of overloaded 'TMC2130Stepper(int, double, int, int, int)' is ambiguous`

but in version 4.5 everything is working fine.

teemuatlut, 2019-09-27:
Is there more to the error message?
Is this one of the examples or your own project?

peaceblackhair, 2019-09-27:
no.im getting in example projects of your library.Can you try to compile please ?

peaceblackhair, 2019-09-27:
[image]

Calibrate_spreadCycle.ino example i think is about version 0.5.1 because i downgrade to 0.4.5 and it worked.

teemuatlut, 2019-09-27:
I can compile fine with Arduino IDE 1.8.9.

And there shouldn't be any ambiguity with
```cpp
TMC2130Stepper(uint16_t pinCS, float RS = default_RS, int8_t link_index = -1);
TMC2130Stepper(uint16_t pinCS, uint16_t pinMOSI, uint16_t pinMISO, uint16_t pinSCK, int8_t link_index = -1);
TMC2130Stepper(uint16_t pinCS, float RS, uint16_t pinMOSI, uint16_t pinMISO, uint16_t pinSCK, int8_t link_index = -1);
```

It would help if your compiler gave the possible constructors it's trying to match against.

Danitegue, 2020-01-11:
Hello, I´m also having this issue in the Software_SPI example:

exit status 1
call of overloaded 'TMC2130Stepper(int, double, int, int, int)' is ambiguous

A bit of background:
I installed your dedicated "TMC2130Stepper" library first. I only had to replace the basic call in order to have customized MOSI,MISO, and CSK pins, in a copy of the Software_SPI example:

instead of:
//TMC2130Stepper TMC2130 = TMC2130Stepper(EN_PIN, DIR_PIN, STEP_PIN, CS_PIN);

I used:
TMC2130Stepper TMC2130 = TMC2130Stepper(EN_PIN, DIR_PIN, STEP_PIN, CS_PIN, MOSI_PIN, MISO_PIN, SCK_PIN);

The driver worked fine, it moved the motor without problems. Great job!

But then I saw that your "TMCStepper" repository was newer. So I uninstalled (deleted) the "TMC2130Stepper" library, and installed the "TMCStepper". 

When trying the Software_SPI example of this other library, this error happens, and does not allow to compile. Any idea why?

Danitegue, 2020-01-11:
the overload error dissapear by replacing:

`TMC2130Stepper driver = TMC2130Stepper(CS_PIN,R_SENSE, SW_MOSI, SW_MISO, SW_SCK); // Software SPI`

by

`TMC2130Stepper driver = TMC2130Stepper(CS_PIN, float(R_SENSE), SW_MOSI, SW_MISO, SW_SCK); // Software SPI`

but no idea why

teemuatlut, 2020-01-11:
I'm not getting this error with TMCStepper / Simple.ino example.
How are you defining the R sense value?

Danitegue, 2020-01-12:
Ok, I have seen the problem:

in your TMCStepper / Simple.ino example, the R sense var is defined in this
way:

"#define R_SENSE 0.11*f*"

but in the TMCStepper / Software_SPI.ino example, the R sense var is
defined as:

"#define R_SENSE 0.11"

So, is missing the last "f"