teemuatlut/TMCStepper #78: [FR] Add support for STM32F4 platform

extesy, 2019-09-29:
@teemuatlut Since STM32 has native support for SoftwareSerial now (https://github.com/stm32duino/Arduino_Core_STM32/pull/645), would it be possible to add full STM32F4 support to TMCStepper library? Right now `TMC_Platforms.h` only knows about STM32F1.

teemuatlut, 2019-10-04:
Done https://github.com/teemuatlut/TMCStepper/commit/2d14f340a0966a16ac66ecfa435434aac3dd0f7d

closed 2019-10-26 by teemuatlut

extesy, 2019-10-26:
@teemuatlut A few questions:
1. Why is there no mention of STM32F4 in `TMC_platforms.h`?
2. The block for `TARGET_STM32F1` seems identical to the block for `else` so should it be removed as redundant?
3. The comment for `else` block says `DUE:116kHz` and I don't think that's correct anymore.

teemuatlut, 2019-10-31:
The STM section used to contain a few more definitions for the UART muxing but that was later expanded to the rest of the platforms.