teemuatlut/TMCStepper #84: 2130 stallguard questions

i-make-robots, 2019-10-21:
Hi!  Thanks for your great work on these libraries.

I'm extra slow today.  I have TMCStepper library working with the StallGuard demo.  I've added a wire to listen to diag1 on pin 42.  Regardless of motor speed or STALL_VALUE I don't get a stall warning or lost_steps > 0.  Any idea what I'm missing?
[sketch_oct21a.ino.txt](https://github.com/teemuatlut/TMCStepper/files/3752537/sketch_oct21a.ino.txt)

teemuatlut, 2019-10-21:
The stall status isn't automatically directed to a diag output.
You need to use the `diag0_stall` or `diag1_stall` methods to enable that.
See `p25 GCONF` and `p72 DIAG Outputs` for more information.

i-make-robots, 2019-10-21:
I guess I'm not understanding how to set the rms_current correctly.  the equation on p56 makes my eyes crossed.  i have a nema17 with peak 300mA.  how would I get the right rms_current?  rn i'm just guessing numbers and that's going to cause an accident.

teemuatlut, 2019-10-22:
`rms_current` handles the calculation automatically so you don't need to worry about the equation.

i-make-robots, 2019-10-22:
so i don't need to call rms_current?  I tried 300/1.4 (~214) and up to 250, it generates many false positives.  I tried 300 up to 300*1.4 and it never drops to zero.  I'm sorry I'm so slow to grok this.

cstevens-mettest, 2019-10-22:
I'm also having a similar issue. I'm currently trying to use a TMC 2130 to drive a stepper motor on an MKS Gen L board. I plan on using the TMCStepper library with the AccelStepper library to create a closed loop system that can account for lost steps. Do I need to use `diag1_stall` or can I use `diag1_steps_skipped()` to find the amount of steps skipped? How do I ensure that I'm listening to Diag1 (Pin 6)? I've attached my modified stallguard example below.
[StallGuardTest.txt](https://github.com/teemuatlut/TMCStepper/files/3759134/StallGuardTest.txt)