teemuatlut/TMCStepper #85: Library on STM32

MagoKimbra, 2019-10-24:
Hi, I've been editing MK4duo firmware for STM32, Board RUMBA32 for a few weeks, and everything works perfectly. TMC2130 in SPI mode, with BLtouch and other functions.
A couple of days ago, I updated the TMC library and then downloaded some updates to the board. The BLTOUCH has stopped working. It took me several days to figure out what it was, between modifications and other things, eventually removing the support for the TMC2130 the BLtouch is back to work ... I couldn't understand why the TMC drivers with the BLtouch centered, after several tests I put the previous library TMCstepper seems to me to be 0.5.0 and at that point even with the TMC2130 drivers the bltouch is working again. At this point I investigated and saw that you include the <SoftwareSerial.h> library if it is enabled ARDUINO_ARCH_STM32.
At this point I went to see in the SoftwareSerial.h library of the arduino stm32 and unfortunately this uses the timer 7 which is the same one used for the servos, that's why the bltouch stopped ...
Now surely the problem I will also communicate to the developers of arduino stm32, but for now if it is possible you could include the SoftwareSerial library only if you use serial drivers and in software mode and not always as now ???
Thank you and sorry for the quality of my English.

teemuatlut, 2019-10-24:
External libraries have no way to know which configuration options are enabled as they don't have access to the project header files. So I don't have anything to test against to decide whether or not I can include SoftwareSerial.h. The only defines I can access is those provided in the compile command or the ones from the platform.
It's possible to do some trickery with PlatformIO but as the library is used for other purposes as well other than just Marlin, I'm hesitant to go that route.

It would be easier to change the BLTouch timer than a platform library.
The underlying issue is with libraries that add globals...

MagoKimbra, 2019-10-24:
I just didn't think that the library doesn't know what is set or not in the firmware. I have nothing to tell them to avoid using the same timer for the softwareserial library and the bltouch.
Thank's and sorry!!!

MagoKimbra, 2019-10-24:
Yes.  I see.. Thank's AnHadt.

teemuatlut, 2019-10-25:
https://github.com/teemuatlut/TMCStepper/blob/5fc23b2b0b95594fb974da23e77372f2d3996d99/src/TMCStepper.h#L22-L27

I don't have any STM boards myself so let me know if it breaks something. Or if it solves anything.

MagoKimbra, 2019-10-25:
No it cannot be resolved in your library. Each board with STM32 has its variants.h with different timer management. This is so in rumba32:
#define TIMER_TONE              TIM6
#define TIMER_SERVO             TIM7
So 6 and 7 are used for tone and servo routines. On other cards it is different. It must be modified in the variants.h of each card, the definition must be set. I wrote to those of the arduino stm32 library and now let's see what we can do ...
I thank you for your interest, only that I became crazy to understand why suddenly the bltouch didn't go anymore, so for no reason. But rightly your library has nothing to do, my mistake ask you something that has nothing to do with ...