teemuatlut/TMCStepper #90: [Feature Request] Support for Trinamic TMC2225 stepper drivers

bagobor, 2019-11-09:
[Feature Request] Support for Trinamic TMC2225 stepper drivers

https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2225_Datasheet_V109.pdf

"The TMC2225 is a Step/Dir Driver for Two-Phase Bipolar Stepper Motors up to 2A peak - StealthChop for Quiet Movement - UART Interface Option."

@trinamic As I understand it; TMC2202, TMC2208, TMC2220, TMC2224 and TMC2225 all belong to the same stepper driver family from Trinamic?; "The TMC22xx family". See:

https://www.digikey.com/en/datasheets/trinamic-motion-control-gmbh/trinamic-motion-control-gmbh-tmc220x_tmc222x_datasheet

@bigtreetech TMC2225 v1.0 drivers are supposedly available from BIGTREETECH now?
https://www.reddit.com/r/BIGTREETECH/comments/dlgfcw/new_driverbigtreetech_tmc2225/

Update: @bigtreetech TMC2225 v1.0 is now available from BIQU Official Stores on Aliexpress:

https://www.aliexpress.com/item/4000298434549.html
https://www.aliexpress.com/item/4000296704105.html
https://www.aliexpress.com/item/4000296898203.html

Update: It looks like a drop-in replacement for 2208 drivers. -> https://wiki.fysetc.com/TMC2225/

teemuatlut, 2019-11-10:
As they're the same as 2208, I'm not looking to add it as a separately supported item.

bagobor, 2019-11-10:
@bagobor 2224 is listed as supported so why not add 2225 as well?

bagobor, 2019-11-10:
Is it just a TMC2224 but in a different package?

Hedda, 2019-11-15:
@teemuatlut if already supported then maybe list so in the README.md file?

https://github.com/teemuatlut/TMCStepper/blob/master/README.md

Hedda, 2019-11-15:
@teemuatlut ...and if TMC2225 is already supported then maybe also TMC2202 and TMC2220 is already supported too since they all belong to the same family?

bjoernwedi, 2019-11-24:
But the 2225s have stall guard II Like 2209ers, right?

bagobor, 2019-11-25:
@bjoernwedi it's not mentioned in the specs: https://www.digikey.com/en/datasheets/trinamic-motion-control-gmbh/trinamic-motion-control-gmbh-tmc220x_tmc222x_datasheet

bagobor, 2020-02-29:
correct link https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC220x_TMC2224_datasheet_Rev1.09.pdf
No any mention about TMC2225  :/ so this issue can be closed now. 
@Hedda FIY

closed 2020-02-29 by bagobor