teemuatlut/TMCStepper #91: My tmc2660 stepper motor driver is very hot, how can I cut the current when it can stop, can you help?

diegcccco, 2019-11-14:
I use the tmc2660 stepper driver on my 3D printer. I use the TM2626Stepper library. When I start my 3-D printer, it is warming up because the motors come to a standstill and how do I stop it at the beginning.
Previously, I used the tmc26xstepper library, so it was easier to set the definitions.
Now I have switched to the new version of mk4duo I had to use the tmc2660stepper library, but I cannot configure it according to the configuration settings in the previous library. How can I solve the heating problem of stepper motors?

teemuatlut, 2019-11-14:
I don't have experience with the MK4Duo fork.
If your motors are running too hot, turn down the current.

diegcccco, 2019-11-15:
> I don't have experience with the MK4Duo fork.
> If your motors are running too hot, turn down the current.

Thank you for helping, can I set up your stallguard if I activate it?