teemuatlut/TMCStepper #93: all low with the new version

cabezacabra, 2019-11-14:
Hii, i just updated the library and when i upload the new marlin with this library (0.60) all drivers send me error all low
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing Z2 connection... Error: All LOW
Testing E connection... Error: All LOW
I just update because the E driver is always on stealthchop mode even when i have it deactivated for it.

teemuatlut, 2019-11-14:
What's your platform?

cabezacabra, 2019-11-14:
Im using mks gen l board with 2208, and arduino in linux (v1.8.9) and windows (v1.8.10)

teemuatlut, 2019-11-14:
Were you using write only mode? As in RX and TX were the same?

cabezacabra, 2019-11-14:
No, rx and tx are 2 diferent, with the y cable with a resistor

teemuatlut, 2019-11-14:
I'll look more into it tomorrow. It's quite possible that Marlin needs an update because when I run the Simple example I can get a good response. The test config was with AVR and 2209 with Half Duplex.

swilkens, 2019-11-15:
Same problem on a SKR Mini E3 v1.2. (env STM32F103RC_bigtree)

With 0.6.0, marlin reports TMC CONNECTION ERROR, M122 showing error state... But the steppers are working.

With 0.5.2 all errors go away.

TheZoker, 2019-11-15:
Same with the SKR DIP E3

swilkens, 2019-11-15:
Fix for now is to modify platform.io to replace `TMCStepper@>=0.5.0,<1.0.0` to `TMCStepper@=0.5.2` untill a patch exists.

sjasonsmith, 2019-11-15:
I had time to take a capture comparing a working X driver and failing Y driver this morning.
I don't have time to root-cause the failure right now, but a few things stand out:

1. Half-duplex is already working with Marlin and TMCStepper 0.5.2, at least for STM32 (and I believe LPC). As @LinoBarreca said in the referenced Marlin issue, the SoftwareSerial Marlin uses for the SKR Pro matches the patches that will be released in Arduino_Core_STM32 1.8, and is effectively the same as the FYSETC SoftwareSerial used for STM32F1 boards, as well as the implementation BTT has in their TMCStepper form.

1. The command sent to the Y driver in the image below seems to sending a CRC value of 0, even though it should match the CRC sent to the X driver (since the commands are identical).

1. There is no stop bit sent after the CRC byte to the Y driver. A lof of TMC stepper modules pull the UART line down so that the modules will default to standalone mode if there is no UART connected. The LPC and STM32 SoftwareSerial libraries are careful not to switch to input prior to the stop bit transmitting and a "Response Delay" elapsing, so that the TMC starts driving without the line entering the indeterminate state shown in the capture, which is caused by competing pull resistors.

[image]

teemuatlut, 2019-11-15:
https://github.com/teemuatlut/Marlin/tree/update_tmc_serial_constructors

https://github.com/teemuatlut/TMCStepper/tree/master

sjasonsmith, 2019-11-16:
I just re-tested using the two branches you linked above, with both an SKR Pro (STM32) and SKR 1.3 (LPC1768).

The SKR Pro still has the same behavior as the pictures I posted. Communication fails for all drivers except X. I don't know what makes X special.

The SKR 1.3 worked, because it was using two serial pins per driver. When I switched to one serial pin per driver, communication failed for all drivers.

Commenting out the pinMode and digitalWrite calls in TMC2208Stepper::_sendDatagram allowed single-pin serial to communicate successfully with both boards.

teemuatlut, 2019-11-16:
Pin mode change now restricted to AVR only as LPC does that natively.
https://github.com/teemuatlut/TMCStepper/commit/7c36557beab437356dd690b06aaf0aea0a5be567

sjasonsmith, 2019-11-16:
Thanks! I will re-test it shortly. I’m sure it will be fine for the SKR boards now. I’ll test it on an AVR as well.

sjasonsmith, 2019-11-17:
I verified that this now works great with both SKR boards (LPC and STM32).

AVR is not working for me. It reports the TMC Connection being OK, but the registers are actually reading back incorrect values, such as 8 microsteps instead of the configured 16.

Here is an example of two back-to-back M122 commands, with different values reported. I've removed uninteresting lines where there were no differences between the output, to reduce the length:

```
M122
                X       Y       Z       E
...
msteps          8       8       8       8
...
Driver registers:
                X       0xC0:1B:00:00
                Y       0xC0:10:00:00
                Z       0xC0:10:00:00
                E       0xC0:1A:00:00

Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
ok

M122
                X       Y       Z       E
...
msteps          256     8       8       8
...
Driver registers:
                X       0xC0:1B:00:00
                Y       0xC0:14:00:00
                Z       0xC0:16:00:00
                E       0xC0:10:00:00

Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
ok
```

sjasonsmith, 2019-11-17:
I am able to modify code to allow it to work on my board. I will post my modifications in a PR in a few minutes, I'm currently capturing some images so you'll be able to see how it changed the behavior on the wire.

It is possible that with some hardware configurations it will already work fine with that you have, mine is just another data point from switching over a board that was already working in 2-wire (and using modules which already worked in 1-wire on 32-bit boards).

sjasonsmith, 2019-11-17:
So my "working" modifications are only working most of the time. I'm seeing a bad status about 5% of the time that I am currently debugging.

sjasonsmith, 2019-11-17:
Well, I feel silly. I spent a ton of time trying to debug why my fix sometimes failed, without considering that 2-wire SoftwareSerial was not that reliable to begin with.

At 115200, querying driver status already fails fairly often with 2-pin serial on my MKS Gen L. Failures usually occur within 100 DRVSTATUS queries. I'm not sure if this has always been the case, or if recent Arduino framework changes may have worsened things.

Since that was already an issue with 2-pin serial, I reduced my baud rate to 50,000, where things are much more stable. With that I validated my change and posted PR #98.

PGrave, 2019-11-18:
SKR Pro V1.1; BTT TMC2209 V1.1; test OK with 0.6.1:

Send: M122
Recv: 		X	Y	Z	E	E1
Recv: 		X	0xC0:08:00:00
Recv: 		Y	0xC0:08:00:00
Recv: 		Z	0xC0:08:00:00
Recv: 		E	0xC0:08:00:00
Recv: 		E1	0xC0:08:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... OK
Recv: Testing E1 connection... OK
Recv: ok

swilkens, 2019-11-19:
Confirmed working on my SKR mini E3 V1.2 using 2209 stepper drivers on latest marlin 2.0.x bugfix.

No more connection errors, M112 reports OK:
```

>>> M122
SENDING:M122
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
```

cabezacabra, 2019-11-22:
Yes now is working perfectly. Ty

closed 2019-11-22 by cabezacabra

jonasuea, 2020-02-13:
#ifndef E1_SERIAL_TX_PIN
    #define E1_SERIAL_TX_PIN 20
  #endif
  #ifndef E1_SERIAL_RX_PIN
    #define E1_SERIAL_RX_PIN 21
  #endif

This error appears
Driver registers:
		X	0xC0:09:00:00
		Y	0xC0:09:00:00
		Z	0xC0:0C:00:00
		E	0xC0:09:00:00
		E1	0x00:00:00:00	 Bad response!
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
Testing E1 connection... Error: All LOW

These pins are not recognizing, can I change to others?

>>> M43
SENDING:M43
PIN:   0   Port: E0        RXD0                                   protected
PIN:   1   Port: E1        TXD0                                   protected
PIN:   2   Port: E4        E0_DIAG_PIN                            Input  = 1    TIMER3B   PWM:     0    WGM: 1    COM3B: 1    CS: 3    TCCR3A: 1    TCCR3B: 3    TIMSK3: 0
PIN:   3   Port: E5        X_MIN_PIN                              protected
.                          X_DIAG_PIN                             protected
.                          X_STOP_PIN                             protected
PIN:   4   Port: G5        FIL_RUNOUT_PIN                         Input  = 0    TIMER0B   PWM:   128    WGM: 3    COM0B: 3    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   compare interrupt enabled   overflow interrupt enabled
PIN:   5   Port: E3        <unused/unknown>                       Input  = 0                           TIMER3A   PWM:     0    WGM: 1    COM3A: 1    CS: 3    TCCR3A: 1    TCCR3B: 3    TIMSK3: 0
PIN:   6   Port: H3        <unused/unknown>                       Input  = 0                           TIMER4A   PWM:     0    WGM: 1    COM4A: 1    CS: 3    TCCR4A: 1    TCCR4B: 3    TIMSK4: 0
PIN:   7   Port: H4        FAN1_PIN                               protected
.                          MOSFET_D_PIN                           protected
PIN:   8   Port: H5        HEATER_BED_PIN                         protected
.                          RAMPS_D8_PIN                           protected
PIN:   9   Port: H6        FAN_PIN                                protected
.                          RAMPS_D9_PIN                           protected
PIN:  10   Port: B4        HEATER_0_PIN                           protected
.                          RAMPS_D10_PIN                          protected
PIN:  11   Port: B5        SERVO0_PIN                             Input  = 0    TIMER1A   PWM:  2000    WGM: 4    COM1A: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode   compare interrupt enabled
PIN:  12   Port: B6        SERVO1_PIN                             Input  = 0    TIMER1B   PWM:     0    WGM: 4    COM1B: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode
PIN:  13   Port: B7        LED_PIN                                Input  = 0    TIMER0A   PWM:     0    WGM: 3    COM0A: 3    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   overflow interrupt enabled
 .                  TIMER1C is also tied to this pin                  TIMER1C   PWM:     0    WGM: 4    COM1C: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode
PIN:  14   Port: J1        Y_MIN_PIN                              protected
.                          Y_DIAG_PIN                             protected
.                          Y_STOP_PIN                             protected
PIN:  15   Port: J0        E1_DIAG_PIN                            Input  = 1
PIN:  16   Port: H1        LCD_PINS_RS                            Output = 0
PIN:  17   Port: H0        LCD_PINS_ENABLE                        Output = 0
PIN:  18   Port: D3        Z_MIN_PIN                              protected
.                          Z_DIAG_PIN                             protected
.                          Z_STOP_PIN                             protected
PIN:  19   Port: D2        <unused/unknown>                       Input  = 1
PIN:  20   Port: D1        E1_SERIAL_TX_PIN                       Output = 1
PIN:  21   Port: D0        E1_CS_PIN                              Input  = 1
.                          E1_SERIAL_RX_PIN                       Input  = 1
PIN:  22   Port: A0        <unused/unknown>                       Input  = 0
PIN:  23   Port: A1        LCD_PINS_D4                            Output = 1
PIN:  24   Port: A2        E0_ENABLE_PIN                          protected
PIN:  25   Port: A3        LCD_PINS_D5                            Input  = 1
PIN:  26   Port: A4        E0_STEP_PIN                            protected
PIN:  27   Port: A5        LCD_PINS_D6                            Input  = 1
PIN:  28   Port: A6        E0_DIR_PIN                             protected
PIN:  29   Port: A7        LCD_PINS_D7                            Input  = 1
PIN:  30   Port: C7        E1_ENABLE_PIN                          protected
PIN:  31   Port: C6        BTN_EN1                                Input  = 0
PIN:  32   Port: C5        SERVO3_PIN                             Input  = 0
PIN:  33   Port: C4        BTN_EN2                                Input  = 0
[... 27 lines cut ...]
PIN:  53   Port: B0        AVR_SS_PIN                             Output = 1
.                          SDSS                                   Output = 1
.                          SS_PIN                                 Output = 1
PIN:  54   Port: F0 (A 0)  X_STEP_PIN                             protected
PIN:  55   Port: F1 (A 1)  X_DIR_PIN                              protected
PIN:  56   Port: F2 (A 2)  Y_ENABLE_PIN                           protected
[... cut ...]
.                          Z_SERIAL_RX_PIN                        Input  = 1
PIN:  66   Port: K4 (A12)  E0_CS_PIN                              Input  = 1
.                          MAX6675_SS_PIN                         Input  = 1
.                          E0_SERIAL_RX_PIN                       Input  = 1
PIN:  67   Port: K5 (A13)  TEMP_0_PIN                             protected
PIN:  68   Port: K6 (A14)  TEMP_BED_PIN                           protected
PIN:  69   Port: K7 (A15)  TEMP_1_PIN                             Analog in =  1023

sjasonsmith, 2020-02-13:
@jonasuea, I am guessing from your comment than you are using an MKS Gen L 2.0. They selecting pins for their 5th driver that are not compatible with SoftwareSerial. It is a board design problem, and not related to this issue.

jonasuea, 2020-02-13:
@sjasonsmith https://github.com/jonasuea/MKS-GEN-2.0/issues/1

Danny-Von-TechTips, 2020-11-21:
> Confirmed working on my SKR mini E3 V1.2 using 2209 stepper drivers on latest marlin 2.0.x bugfix.
> 
> No more connection errors, M122 reports OK:
> 
> ```
> >>> M122
> SENDING:M122
> Testing X connection... OK
> Testing Y connection... OK
> Testing Z connection... OK
> Testing E connection... OK
> ```

How did you fix the TMC Connection Error?

Danny-Von-TechTips, 2020-11-21:
@swilkens How did you fix the TMC Connection Error?

swilkens, 2020-11-22:
> @swilkens How did you fix the TMC Connection Error? I Have an SKR mini E3 V2 and E motor isn't working with an All Low Error.
> [image]

This issue no longer exists on current marlin with current TMCStepper library, though I am using a V1.2 board. There are some differences between the v1.2 and 2.0 board.

Danny-Von-TechTips, 2020-11-22:
@swilkens How did you fix your Connection Error?

nachosan100, 2020-12-20:
I am still having the same issue with the nightly build from last night (2020-12-19). Fresh Install of VS Code.
This is my M122:

Recv: 		X	Y	Z	E
Recv: Address		0	2	1	3
Recv: Enabled		false	false	false	false
Recv: Set current	580	580	580	650
Recv: RMS current	550	994	994	1160
Recv: MAX current	776	1402	1402	1636
Recv: Run current	17/31	17/31	17/31	20/31
Recv: Hold current	8/31	8/31	8/31	10/31
Recv: CS actual	8/31	0/31	0/31	0/31
Recv: PWM scale
Recv: vsense		1=.18	0=.325	0=.325	0=.325
Recv: stealthChop	true	false	false	false
Recv: msteps		16	256	256	256
Recv: interp		true	false	false	false
Recv: tstep		max	0	0	0
Recv: PWM thresh.
Recv: [mm/s]
Recv: OT prewarn	false	false	false	false
Recv: triggered
Recv:  OTP		false	false	false	false
Recv: pwm scale sum	10	0	0	0
Recv: pwm scale auto	0	0	0	0
Recv: pwm offset auto	36	0	0	0
Recv: pwm grad auto	14	0	0	0
Recv: off time	4	0	0	0
Recv: blank time	24	16	16	16
Recv: hysteresis
Recv:  -end		2	-3	-3	-3
Recv:  -start		1	1	1	1
Recv: Stallguard thrs	0	0	0	0
Recv: uStep count	8	0	0	0
Recv: DRVSTATUS	X	Y	Z	E
Recv: sg_result	0	0	0	0
Recv: stst			*	*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:08:00:00
Recv: 		Y	0x00:00:00:00	 Bad response!
Recv: 		Z	0x00:00:00:00	 Bad response!
Recv: 		E	0x00:00:00:00	 Bad response!
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... Error: All LOW
Recv: Testing Z connection... Error: All LOW
Recv: Testing E connection... Error: All LOW

Does anybody have any further ideas how to get around this?

swilkens, 2020-12-21:
> @swilkens How did you fix your Connection Error?

The issue was fixed in source code. Remember I have a v1.2, not a 2.0

Ozanismailoz, 2021-11-21:
hello i am getting the same error but i can't solve it. can you help me?