teemuatlut/TMCStepper #97: Teensy 4.0 Compile Error

neilmendoza, 2019-11-15:
Doesn't like SoftwareSerial.h, see below. It compiles fine if I remove the SoftwareSerial include on line 23 of TMCStepper.h.

Dependency Graph
|-- <TMCStepper> 0.6.0
|   |-- <SPI> 1.0
|   |-- <SoftwareSerial> 1.0
Building in release mode
Compiling .pio/build/teensy40/src/main.cpp.o
Compiling .pio/build/teensy40/libeb3/SPI/SPI.cpp.o
Compiling .pio/build/teensy40/lib729/SoftwareSerial/SoftwareSerial.cpp.o
Compiling .pio/build/teensy40/libf94/TMCStepper_ID5513/source/CHOPCONF.cpp.o
Compiling .pio/build/teensy40/libf94/TMCStepper_ID5513/source/COOLCONF.cpp.o
Compiling .pio/build/teensy40/libf94/TMCStepper_ID5513/source/DRVCONF.cpp.o
Compiling .pio/build/teensy40/libf94/TMCStepper_ID5513/source/DRVCTRL.cpp.o
Compiling .pio/build/teensy40/libf94/TMCStepper_ID5513/source/DRVSTATUS.cpp.o
/Users/neil/.platformio/packages/framework-arduinoteensy/libraries/SoftwareSerial/SoftwareSerial.cpp: In constructor 'SoftwareSerial::SoftwareSerial(uint8_t, uint8_t, bool)':
/Users/neil/.platformio/packages/framework-arduinoteensy/libraries/SoftwareSerial/SoftwareSerial.cpp:75:11: error: 'Serial8' was not declared in this scope
   port = &Serial8;
           ^
/Users/neil/.platformio/packages/framework-arduinoteensy/libraries/SoftwareSerial/SoftwareSerial.cpp:95:8: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
  txreg = portOutputRegister(digitalPinToPort(txPin));
        ^
/Users/neil/.platformio/packages/framework-arduinoteensy/libraries/SoftwareSerial/SoftwareSerial.cpp:96:8: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
  rxreg = portInputRegister(digitalPinToPort(rxPin));
        ^
*** [.pio/build/teensy40/lib729/SoftwareSerial/SoftwareSerial.cpp.o] Error 1

teemuatlut, 2019-11-17:
I couldn't reproduce this with Arduino IDE and the Teensyduino add on.
It also sounds like the errors aren't coming from the TMCStepper library.

neilmendoza, 2019-11-17:
OK, thanks for checking it out. Guess it must be related to PlatformIO somehow so will just stick with commenting out soft serial include for now.