teemuatlut/TMCStepper #99: TMC2209 Torque Setting

neilmendoza, 2019-11-17:
I'm running a TMC2209 connected to a motor rated at 2A. If I set the RMS current to 2000mA...
```
    driver.I_scale_analog(false); 
    driver.rms_current(2000);        
    driver.microsteps(64);          
    driver.en_spreadCycle(false);
    driver.pwm_autoscale(true);  
```
...the motor has very little torque. If I set it to 1600mA, it gives me the kind of torque I'd expect for a motor of that size.

teemuatlut, 2019-11-17:
What's your R sense value?
A typical 0.11Ohm value results in max rms current of 1.77A.

neilmendoza, 2019-11-17:
I'm using [Bigtreetech v1.2 TMC2209 drivers](https://www.ebay.com/itm/BIGTREETECH-TMC2209-V1-2-Stepper-Motor-Driver-UART-SKR-V1-3-Pro-Control-Board-/183969349016) which claim to have an R sense of 0.11Ohm so guess that's the reason (even though they claim up to 2A). Thanks!

teemuatlut, 2019-11-17:
It's likely that their marketing guys just plugged that value from the datasheet without actually knowing what factors into the current calculation.
I've added a check against using current values above the valid range to prevent confusion in the future. https://github.com/teemuatlut/TMCStepper/commit/3b67947cc3f3539e331de61dc988a6a3dc7a5064

closed 2019-12-28 by neilmendoza