teemuatlut/TMCStepper #104: TMC5160 Stallguard/Sensorless homing support

mamama1, 2019-12-09:
Hi

I wasn't quite sure where to post this (whether in the Marlin or in this repo), is Stallguard/Sensorless homing currently working with the TMC5160 or is this yet to be implemented?
I'm asking because we are finally starting to develop a 3D printer board which has those TMC5160 drivers onboard.

Thanks!

teemuatlut, 2019-12-10:
Yes it should work just as well as with 2130.

mamama1, 2019-12-10:
Great, thanks for your reply! Is it implemented the same way, through DIAG1 pin like with the TMC2130 steppers?

Rincewind08, 2020-05-05:
Okay, so check it out.  You need to add a pin.  There are four holes, three of them are for the quadrature encoder that you can hook up (see as5601 if interested, budget option.  You can also use I2C Encoders and do it on Marlin side - another topic entirely.)

So when you are looking at your 5160 v1.2, you will see four golden holes going across one side.  For the ones I have, I opened the box to find a pinout from bigtreetech, and it showed me that the DIAG1 (StallGuard2 Output pin) was the second one over from the Enable pin.  So it goes x DIAG1 x x, starting from the enable pin.  The other three pins, should you want to solder them up too, should face upwards, but the pin you solder to DIAG1 will go down into the board.

Then you simply have to find the sweet spot for your motors in terms of current and StallGuard sensitivity.  I could dive off into that talk for days, but for instance, I am using 1.7amp RMS Nema 17s, and I have found that with a homing amperage of 600ma and a StallGuard sensitivity of 3, I am able to get a soft and truly silent homing to the exact same position every time.  All of these settings will be found in the Configuration_adv.h file.  I have found that the homing current value is typically 1/5th to 1/3rd of your full stepping current for the motor.  Solder that single pin on, double check the schematics for your stepper, making sure to connect to DIAG1.  Solder it going downwards into the board, adding the solder from the top.  Do not be sloppy and touch any components or any other pads around that pin's hole, or you will have what used to be a functioning 5160, but is now just some angry mosfets.

Hope this helps.  

#if HAS_TRINAMIC_CONFIG

  #define HOLD_MULTIPLIER    0.7  // Scales down the holding current from run current
  #define INTERPOLATE       true  // Interpolate X/Y/Z_MICROSTEPS to 256

  #if AXIS_IS_TMC(X)
    #define X_CURRENT       1400        // (mA) RMS current. Multiply by 1.414 for peak current.
    #define X_CURRENT_HOME  600  // (mA) RMS current for sensorless homing
    #define X_MICROSTEPS     8    // 0..256
    #define X_RSENSE          0.075
    #define X_CHAIN_POS      -1    // <=0 : Not chained. 1 : MCU MOSI connected. 2 : Next in chain, ...
  #endif

  #if AXIS_IS_TMC(Y)
    #define Y_CURRENT       1400
    #define Y_CURRENT_HOME  600
    #define Y_MICROSTEPS     8
    #define Y_RSENSE          0.075
    #define Y_CHAIN_POS      -1
  #endif

  #if AXIS_IS_TMC(Z)
    #define Z_CURRENT       650
    #define Z_CURRENT_HOME  Z_CURRENT
    #define Z_MICROSTEPS     32
    #define Z_RSENSE          0.075
    #define Z_CHAIN_POS      -1
  #endif

  #if AXIS_IS_TMC(E0)
    #define E0_CURRENT      800
    #define E0_MICROSTEPS    32
    #define E0_RSENSE         0.075
    #define E0_CHAIN_POS     -1
  #endif

  #define TMC_USE_SW_SPI
  //#define TMC_SW_MOSI       -1
  //#define TMC_SW_MISO       -1
  //#define TMC_SW_SCK        -1

  /**
   * Software enable
   *
   * Use for drivers that do not use a dedicated enable pin, but rather handle the same
   * function through a communication line such as SPI or UART.
   */
  //#define SOFTWARE_DRIVER_ENABLE

  /**
   * TMC2130, TMC2160, TMC2208, TMC2209, TMC5130 and TMC5160 only
   * Use Trinamic's ultra quiet stepping mode.
   * When disabled, Marlin will use spreadCycle stepping mode.
   */
  #define STEALTHCHOP_XY
  #define STEALTHCHOP_Z
  #define STEALTHCHOP_E

  #define CHOPPER_TIMING CHOPPER_DEFAULT_24V

  #define MONITOR_DRIVER_STATUS

  #if ENABLED(MONITOR_DRIVER_STATUS)
    #define CURRENT_STEP_DOWN     50  // [mA]
    #define REPORT_CURRENT_CHANGE
    //#define STOP_ON_ERROR
  #endif

  /**
   * TMC2130, TMC2160, TMC2208, TMC2209, TMC5130 and TMC5160 only
   * The driver will switch to spreadCycle when stepper speed is over HYBRID_THRESHOLD.
   * This mode allows for faster movements at the expense of higher noise levels.
   * STEALTHCHOP_(XY|Z|E) must be enabled to use HYBRID_THRESHOLD.
   * M913 X/Y/Z/E to live tune the setting
   */
  #define HYBRID_THRESHOLD

  #define X_HYBRID_THRESHOLD     65  // [mm/s]
  #define Y_HYBRID_THRESHOLD     65
  #define Z_HYBRID_THRESHOLD      10

  #define SENSORLESS_HOMING // StallGuard capable drivers only

  #if EITHER(SENSORLESS_HOMING, SENSORLESS_PROBING)
    // TMC2209: 0...255. TMC2130: -64...63
    #define X_STALL_SENSITIVITY  3
    #define Y_STALL_SENSITIVITY  3
    //#define Z_STALL_SENSITIVITY  8
    //#define SPI_ENDSTOPS              // TMC2130 only
    #define IMPROVE_HOMING_RELIABILITY
  #endif

MarcelRobitaille, 2020-10-27:
It would be nice to have an example of how to home without a sensor (outside of marlin).

@Rincewind08, it seems that this is to integrate into marlin?