teemuatlut/TMCStepper #105: TMC2209 Stallguard Exemple

siteswapjuggler, 2019-12-14:
Hi and thanks a lot for you work,

I tried the stall guard exemple on TMC2209 v1.2 from Big Tree Tech connected to an Arduino Mega. I don't understand what happen exactly the Serial prints give me three numbers : 0 0 55.

The motors turns and react to Serial command but nothing seems to happen from a PDN UART point of view. Do I miss something ?

Sincerely

teemuatlut, 2019-12-14:
The first zero is just to force the bottom of the graph scale as it would try to automatically adjust to the current min point shown.

The second is the current stallGuard measurement.

The third is the current rms coil current and demonstrates the coolStep feature.

siteswapjuggler, 2019-12-14:
I understand what they should do but none of those numbers are actually changing, and there values are the same as when I'm totally unconnected. 

Can you confirm that I should see changes when I manually block the motor movement ? Is there a basic command to confirm that the PDN_UART communication works well ?

siteswapjuggler, 2019-12-15:
Problem solved : a had a misplaced dupont wire. For documentation purpose 0 0 55 are the values sent when the communication doesn't work. The real values should reflect the rms value set with driver.rms_current() and a stall guard value that change over time and drop when the motor is blocked.

Thanks again for your work and rapid answer yesterday.

++

closed 2019-12-15 by siteswapjuggler

wuram806, 2020-07-14:
Hello
I have same trouble.
Stallguard SG RESULT value is zero.
I use arduino mega and big tree tmc2209.
I connected rx,tx,dir,step,en, power pin.
Can i see your circuit diagram?(arduino-tmc2209)

siteswapjuggler, 2020-07-14:
I didn't document at the time, it's the classic configuration and my mistake was the enable pin not well connected. Make sure that vdd and vm are both connected otherwise you will not have any answer from the driver.

wuram806, 2020-07-15:
thank you very much!
but.. I already connected vm and vdd.TT
Please refer to the below for the my circuit connection..
Did I make any mistakes?

1)uno 5v : tmc vdd
2)uno gnd : tmc gnd
3)dc 24v power + : tmc vm
4)dc 24v power - : tmc gnd
5)uno rx : tmc pdn uart (nearest EN PIN)
6)uno tx : tmc pdn uart (nearest EN PIN, use 1k ohm)
7)uno gpio pins : tmc step / en / dir
8)tmc a1/2,b1/2 : step motor