teemuatlut/TMCStepper #107: [BUG] LPC176x TMC SOFTWARE SPI too fast

Windaub57, 2019-12-24:
### Bug Description

TMC software SPI is running too fast and 40bits setup messages not fully sent to drivers, resulting in random and anarchic driver configuration.
Also drivers reporting (reading) is fully corrupted.

Already discussed that topic in MARLIN Git and seems that the issue is related to TMCStepper.

### My Configurations

Marlin BUGFIX 2.0 from 23/12/2019, LPC176x boards (dev one), TMC2130 SPI drivers

### Steps to Reproduce

1. Setup any LPC176x boards with TMC SOFT SPI DRIVERS in MARLIN firmware

**Expected behavior:** proper communication with driver, setup and feedback

**Actual behavior:** complete random communication, drivers not properly setup, feedback corrupted

#### Additional Information

used a logic analyser to check messages sent. Modified writeSCK_H and writeSCK_L timing in TMC_platforms.h to make it slower and got proper communication back. See attached photos before after.

Also SPI_SPEED has no effect on communication speed.

[image]
[image]

XBrav, 2020-01-14:
@Windaub57, what exact modifications did you make? I'm having similar issues on the STM32 platform with 5161s randomizing.

Windaub57, 2020-01-14:
i changed in tmc_platforms.h
  #define writeSCK_H LPC176x::delay_ns(**40**); LPC176x::gpio_set(sck_pin)
  #define writeSCK_L LPC176x::delay_ns(**40**); LPC176x::gpio_clear(sck_pin)
to
  #define writeSCK_H LPC176x::delay_ns(**240**); LPC176x::gpio_set(sck_pin)
  #define writeSCK_L LPC176x::delay_ns(**240**); LPC176x::gpio_clear(sck_pin)

XBrav, 2020-01-14:
Thanks. I'll find the specific call for the STM32's delay to see if it makes a difference. Any issues after the change?

Windaub57, 2020-01-14:
no, everything is running smooth since ;)

xavsbl, 2020-01-18:
Hi there, many thanks to @Windaub57 . I stuffed a Sunlu S8 with a BTT SKR1.4Turbo and was struggling with the TMC5160 v1.2 i put on XY. Tried many things, but without results. Drivers were reporting error, m122 was sometime reporting like there were no motors attached to driver,or bad registers, randomly happenning. Sometimes motors were running well with no error and right after that when homing one of them was running at very low speed and then m122 would say nothing is wrong.

This afternoon i came across this issue and gave it a try. Modified tmc_platform.h like said. 
that worked. I tried to go back with 40ns to try if the problem was here again. And yep it was. Then reverting to 240ns and again no errors, m122 reporting working motors and good reading of registers.

Thank you !