teemuatlut/TMCStepper #111: MSCURACT_t out of scope

AlpineWhite, 2020-01-21:
I am trying to compile this from source in a project, and I am hitting compiler errors for MSCURACT_t such as:
In member function ‘uint32_t TMCStepper::MSCURACT()’:
 error: ‘MSCURACT_t’ has not been declared
 uint32_t TMCStepper::MSCURACT() { return read(MSCURACT_t::address); }

In member function ‘int16_t TMCStepper::cur_a()’:
error: ‘MSCURACT_t’ was not declared in this scope
   MSCURACT_t r{0};
 error: ‘r’ was not declared in this scope
   r.sr = MSCURACT();

(and the same for cur_b)

I compared  to the other structs and tried adding it next to TSTEP_t and MSCNT_t which then threw errors:
In member function ‘int16_t TMCStepper::cur_a()’:
error: too many initializers for ‘TMCStepper::MSCURACT_t’
  MSCURACT_t r{0};
 error: ‘struct TMCStepper::MSCURACT_t’ has no member named ‘sr’
   r.sr = MSCURACT();
 error: ‘struct TMCStepper::MSCURACT_t’ has no member named ‘cur_a’
   int16_t value = r.cur_a;

(and the same in cur_b)

Checking against similar structs and their usage in TMCStepper.cpp, the others are defined in the bitfields.h file. However, in the 5130, 2160, nor 5160 file can I find any MSCURACT_t struct buildout / unionizations.

Could a fix be as simple as:

struct MSCURACT_t {
  constexpr static uint8_t address = 0x6B;
  union {
    uint32_t sr : 32;
    struct {
      uint16_t cur_a : 16,
              cur_b : 16;
    };
  };
};

Any help is appreciated.

teemuatlut, 2020-01-21:
What's your full error log?

AlpineWhite, 2020-01-22:
CmakeOutput.log reads:

'
Building CXX object CMakeFiles/cmTC_5ca99.dir/feature_tests.cxx.o
/usr/bin/c++    -std=c++17 -o CMakeFiles/cmTC_5ca99.dir/feature_tests.cxx.o -c /home/pi/nanodlpshield/CMakeFiles/feature_te$
Linking CXX executable cmTC_5ca99
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_5ca99.dir/link.txt --verbose=1
/usr/bin/c++       -rdynamic CMakeFiles/cmTC_5ca99.dir/feature_tests.cxx.o  -o cmTC_5ca99
make[1]: Leaving directory '/home/pi/nanodlpshield/CMakeFiles/CMakeTmp'

    Feature record: CXX_FEATURE:1cxx_aggregate_default_initializers
    Feature record: CXX_FEATURE:1cxx_alias_templates
    Feature record: CXX_FEATURE:1cxx_alignas
    Feature record: CXX_FEATURE:1cxx_alignof
    Feature record: CXX_FEATURE:1cxx_attributes
    Feature record: CXX_FEATURE:1cxx_attribute_deprecated
    Feature record: CXX_FEATURE:1cxx_auto_type
    Feature record: CXX_FEATURE:1cxx_binary_literals
    Feature record: CXX_FEATURE:1cxx_constexpr
    Feature record: CXX_FEATURE:1cxx_contextual_conversions
    Feature record: CXX_FEATURE:1cxx_decltype
    Feature record: CXX_FEATURE:1cxx_decltype_auto
    Feature record: CXX_FEATURE:1cxx_decltype_incomplete_return_types
    Feature record: CXX_FEATURE:1cxx_default_function_template_args
    Feature record: CXX_FEATURE:1cxx_defaulted_functions
    Feature record: CXX_FEATURE:1cxx_defaulted_move_initializers
    Feature record: CXX_FEATURE:1cxx_delegating_constructors
    Feature record: CXX_FEATURE:1cxx_deleted_functions
    Feature record: CXX_FEATURE:1cxx_digit_separators
    Feature record: CXX_FEATURE:1cxx_enum_forward_declarations
    Feature record: CXX_FEATURE:1cxx_explicit_conversions
    Feature record: CXX_FEATURE:1cxx_extended_friend_declarations
    Feature record: CXX_FEATURE:1cxx_extern_templates
    Feature record: CXX_FEATURE:1cxx_final
    Feature record: CXX_FEATURE:1cxx_func_identifier
    Feature record: CXX_FEATURE:1cxx_generalized_initializers
    Feature record: CXX_FEATURE:1cxx_generic_lambdas
    Feature record: CXX_FEATURE:1cxx_inheriting_constructors
    Feature record: CXX_FEATURE:1cxx_inline_namespaces
    Feature record: CXX_FEATURE:1cxx_lambdas
    Feature record: CXX_FEATURE:1cxx_lambda_init_captures
    Feature record: CXX_FEATURE:1cxx_local_type_template_args
    Feature record: CXX_FEATURE:1cxx_long_long_type
    Feature record: CXX_FEATURE:1cxx_noexcept
    Feature record: CXX_FEATURE:1cxx_nonstatic_member_init
    Feature record: CXX_FEATURE:1cxx_nullptr
    Feature record: CXX_FEATURE:1cxx_override
    Feature record: CXX_FEATURE:1cxx_range_for
    Feature record: CXX_FEATURE:1cxx_raw_string_literals
    Feature record: CXX_FEATURE:1cxx_reference_qualified_functions
    Feature record: CXX_FEATURE:1cxx_relaxed_constexpr
    Feature record: CXX_FEATURE:1cxx_return_type_deduction
    Feature record: CXX_FEATURE:1cxx_right_angle_brackets
    Feature record: CXX_FEATURE:1cxx_rvalue_references
    Feature record: CXX_FEATURE:1cxx_sizeof_member
    Feature record: CXX_FEATURE:1cxx_static_assert
    Feature record: CXX_FEATURE:1cxx_strong_enums
    Feature record: CXX_FEATURE:1cxx_template_template_parameters
    Feature record: CXX_FEATURE:1cxx_thread_local
    Feature record: CXX_FEATURE:1cxx_trailing_return_types
    Feature record: CXX_FEATURE:1cxx_unicode_literals
    Feature record: CXX_FEATURE:1cxx_uniform_initialization
    Feature record: CXX_FEATURE:1cxx_unrestricted_unions
    Feature record: CXX_FEATURE:1cxx_user_literals
    Feature record: CXX_FEATURE:1cxx_variable_templates
    Feature record: CXX_FEATURE:1cxx_variadic_macros
    Feature record: CXX_FEATURE:1cxx_variadic_templates
'

Error output to console reads:
[... 58 lines cut ...]
Only edits to the lib are removal of UART steppers (this is going on a Pi and using the 5160 motion controller over SPI) and replacing SW_SPI functions with bcm2835 library equivalents

void SW_SPIClass::init() {
  int SPI_enum = bcm2835_spi_begin();
  bcm2835_spi_setBitOrder(BCM2835_SPI_BIT_ORDER_MSBFIRST);
  bcm2835_spi_setDataMode(BCM2835_SPI_MODE3);
  bcm2835_spi_setClockDivider(BCM2835_SPI_CLOCK_DIVIDER_256);
  bcm2835_spi_chipSelect(BCM2835_SPI_CS0);
  bcm2835_spi_setChipSelectPolarity(BCM2835_SPI_CS0, LOW);
}

uint8_t SW_SPIClass::transfer(uint8_t ulVal) {
  uint8_t value = 0;
  value = bcm2835_spi_transfer(ulVal);
  return value;
}

uint16_t SW_SPIClass::transfer16(uint16_t data) {
  uint16_t returnVal = 0x0000;
  returnVal |= transfer((data>>8)&0xFF) << 8;
  returnVal |= transfer(data&0xFF) & 0xFF;
  return returnVal;
}

SW_SPI was forced by changing Due elif to else

teemuatlut, 2020-01-22:
The `MSCURACT_t` bitfield is borrowed from 2208, so it's possible that this was missed in your changes.

---

You might also be interested in the [weak_io_functions](https://github.com/teemuatlut/TMCStepper/tree/weak_io_functions) branch as that would allow the application to override certain methods from the library.

AlpineWhite, 2020-01-26:
You sir, are a wizard and a saint.  That worked perfectly (at compile time).  I still have to wait to build my custom board.  
Thank you.

closed 2020-01-26 by AlpineWhite