teemuatlut/TMCStepper #112: TMC2208 stand alone with arduino Nano IOT 33

Russell108, 2020-01-21:
Hi I cannot seen to get the simple example  to compile I get the following error

use of deleted function 'TMC2208Stepper::TMC2208Stepper(uint16_t, uint16_t, float)'

<code
/**
   Author Teemu Mäntykallio
   Initializes the library and runs the stepper
   motor in alternating directions.
*/

#include <TMCStepper.h>

//#define EN_PIN           4 // Enable
#define DIR_PIN          4 // Direction
#define STEP_PIN         5 // Step
//#define CS_PIN           42 // Chip select
//#define SW_MOSI          66 // Software Master Out Slave In (MOSI)
//#define SW_MISO          44 // Software Master In Slave Out (MISO)
//#define SW_SCK           64 // Software Slave Clock (SCK)
#define SW_RX            30 // TMC2208/TMC2224 SoftwareSerial receive pin
#define SW_TX            31 // TMC2208/TMC2224 SoftwareSerial transmit pin
//#define SERIAL_PORT Serial1 // TMC2208/TMC2224 HardwareSerial port
#define DRIVER_ADDRESS 0b00 // TMC2209 Driver address according to MS1 and MS2
//
#define R_SENSE 0.11f // Match to your driver
// SilentStepStick series use 0.11
// UltiMachine Einsy and Archim2 boards use 0.2
// Panucatt BSD2660 uses 0.1
// Watterott TMC5160 uses 0.075

// Select your stepper driver type
//TMC2130Stepper driver(CS_PIN, R_SENSE);                           // Hardware SPI
//TMC2130Stepper driver(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK); // Software SPI
//TMC2660Stepper driver(CS_PIN, R_SENSE);                           // Hardware SPI
//TMC2660Stepper driver(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK);
//TMC5160Stepper driver(CS_PIN, R_SENSE);
//TMC5160Stepper driver(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK);

//TMC2208Stepper driver(&SERIAL_PORT, R_SENSE);                     // Hardware Serial
TMC2208Stepper driver(SW_RX, SW_TX, R_SENSE);                     // Software serial
//TMC2209Stepper driver(&SERIAL_PORT, R_SENSE, DRIVER_ADDRESS);
//TMC2209Stepper driver(SW_RX, SW_TX, R_SENSE, DRIVER_ADDRESS);

void setup() {
  // pinMode(EN_PIN, OUTPUT);
  pinMode(STEP_PIN, OUTPUT);
  pinMode(DIR_PIN, OUTPUT);
  //  digitalWrite(EN_PIN, LOW);      // Enable driver in hardware

  // Enable one according to your setup
  //SPI.begin();                    // SPI drivers
  //SERIAL_PORT.begin(115200);      // HW UART drivers
  driver.beginSerial(115200);     // SW UART drivers

  driver.begin();                 //  SPI: Init CS pins and possible SW SPI pins
  // UART: Init SW UART (if selected) with default 115200 baudrate
  driver.toff(5);                 // Enables driver in software
  driver.rms_current(600);        // Set motor RMS current
  driver.microsteps(16);          // Set microsteps to 1/16th

  //driver.en_pwm_mode(true);       // Toggle stealthChop on TMC2130/2160/5130/5160
  //driver.en_spreadCycle(false);   // Toggle spreadCycle on TMC2208/2209/2224
  // driver.pwm_autoscale(true);     // Needed for stealthChop
}

bool shaft = false;

void loop() {
  // Run 5000 steps and switch direction in software
  for (uint16_t i = 5000; i > 0; i--) {
    digitalWrite(STEP_PIN, HIGH);
    delayMicroseconds(160);
    digitalWrite(STEP_PIN, LOW);
    delayMicroseconds(160);
  }
  shaft = !shaft;
  driver.shaft(shaft);
}
code/>

teemuatlut, 2020-01-21:
The Arduino NANO IOT33 is not a platform that is currently marked as supporting Software Serial through an included library.

Russell108, 2020-01-21:
Ok

My ignorance

Is it possible to connect the serial Pins of the arduino Nano IOT 33 to
A TMC stepper driver using cables

I am not sure the difference between software & hardware serial

Thanks

teemuatlut, 2020-01-21:
https://www.arduino.cc/en/Tutorial/SamdSercom
This article would suggest it's possible to add more Serial ports that you could use. I'm not sure if it'll be compatible with the API but all TMCStepper cares about is that the Serial instance is of type `Stream` or inherits from it and that it has the few required methods available.
It might require some tinkering as the library was not created with SAMD in mind but I'm sure it'll eventually be possible.