teemuatlut/TMCStepper #113: call to overloaded

slipperylemon, 2020-02-04:
error on compile:
call of overloaded 'TMC5160Stepper(int, double, int, int, int)' is ambiguous

Versions 1.9.0 beta and 1.8.11
linux mint 64

error message:
Arduino: 1.9.0-beta (Linux), Board: "Arduino/Genuino Uno"

Simple:33:222: error: call of overloaded 'TMC5160Stepper(int, double, int, int, int)' is ambiguous
 TMC5160Stepper driver(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK);
                                                                                                                                                                                                                              ^
In file included from /tmp/arduino_modified_sketch_99715/Simple.ino:7:0:
/home/chris/Arduino/libraries/TMCStepper/src/TMCStepper.h:707:3: note: candidate: TMC5160Stepper::TMC5160Stepper(uint16_t, float, uint16_t, uint16_t, uint16_t, int8_t)
   TMC5160Stepper(uint16_t pinCS, float RS, uint16_t pinMOSI, uint16_t pinMISO, uint16_t pinSCK, int8_t link_index = -1);
   ^~~~~~~~~~~~~~
/home/chris/Arduino/libraries/TMCStepper/src/TMCStepper.h:706:3: note: candidate: TMC5160Stepper::TMC5160Stepper(uint16_t, uint16_t, uint16_t, uint16_t, int8_t)
   TMC5160Stepper(uint16_t pinCS, uint16_t pinMOSI, uint16_t pinMISO, uint16_t pinSCK, int8_t link_index = -1);
   ^~~~~~~~~~~~~~
Multiple libraries were found for "TMCStepper.h"
 Used: /home/chris/Arduino/libraries/TMCStepper
Multiple libraries were found for "SPI.h"
 Used: /home/chris/arduino-PR-beta1.9-BUILD-116/hardware/arduino/avr/libraries/SPI
Multiple libraries were found for "SoftwareSerial.h"
 Used: /home/chris/arduino-PR-beta1.9-BUILD-116/hardware/arduino/avr/libraries/SoftwareSerial
exit status 1
call of overloaded 'TMC5160Stepper(int, double, int, int, int)' is ambiguous

teemuatlut, 2020-02-04:
Make sure to define the `R_SENSE` as a float either with an `f` suffix or a strong variable type.

```cpp
#define R_SENSE 0.11f
or
constexpr float R_SENSE = 0.11;
```

slipperylemon, 2020-02-04:
Thank you. That fixed it.

closed 2020-02-04 by slipperylemon