teemuatlut/TMCStepper #119: TMC5160 Problem

RandyLJackson, 2020-03-26:
Hello there,
currently I am working on a Arduino project, with an arduino and a TMC5160 Stepper Driver. 

The goal of this projcet is, to let a Nema 23 Stepper turn in both directions. Easy as that!

My problem is that I don´t know if my code is wrong or the connection between the components.

It could also be, that I included the library (TMCStepper) wrong in my programm.

I send pictures of my wireing and of course the code I am useing.

This is one of my first attempts to use a microcontroller, thats probably the reason why I cannot get this system up and running.

The controller is powered by usb and the powersource for the stepper is 24V DC 2.7 A.

Thank you for your help!

I know it might be a silly mistake, but everyone has to start small ^^.
`// Einfügen der Stepper Treiber Bibilotek
#include <TMCStepper.h>

//Pinbelegung findet global in dem Code statt
#define BTN_LEFT    2   //Button für die Linksdreheung des Schrittmotors
#define BTN_RIGHT   3   //Button für die Rechtsdreheung des Schrittmotors

//Pinbelegung der einzelnen Stepper anschlüsse
#define EN_PIN    7 //enable
#define DIR_PIN   8 //direction
#define STEP_PIN  9 //step

#define CS_PIN   10 //CS chip select
#define MOSI_PIN 11 //SDI/MOSI (ICSP: 4, Uno: 11, Mega: 51)
#define MISO_PIN 12 //SDO/MISO (ICSP: 1, Uno: 12, Mega: 50)
#define SCK_PIN  13 //CLK/SCK  (ICSP: 3, Uno: 13, Mega: 52)

// Definieren der Sensibilität des inneren Widerstandes. Kommt auf den Treiber an den man verwendet bei uns 0.075f
#define R_SENSE   0.075f //TMC5160: 0.075

//Auswahl des Treibers und der Ansteuerungsmethode // Teilt der Bibilotek mit wecher Treiber verwendet wird

TMC5160Stepper driver = TMC5160Stepper(CS_PIN, R_SENSE, MOSI_PIN, MISO_PIN, SCK_PIN);

int LEFT = LOW;
int RIGHT = LOW;
bool dir = true;

void setup() {
  
  pinMode(BTN_LEFT, INPUT);
  digitalWrite(BTN_LEFT, LOW);
  pinMode(BTN_RIGHT, INPUT);
  digitalWrite(BTN_RIGHT, LOW);

  //Pin Bestimmung für die Ansteuerung des Motors
  pinMode(EN_PIN, OUTPUT);
  digitalWrite(EN_PIN, HIGH); //deactivate driver (LOW active)
  pinMode(DIR_PIN, OUTPUT);
  digitalWrite(DIR_PIN, LOW); //direction: LOW or HIGH,
  pinMode(STEP_PIN, OUTPUT);
  digitalWrite(STEP_PIN, LOW);

  
//Die Treiber Configuration setzten
  driver.begin();
  driver.toff(4); //off time
  driver.blank_time(24); //blank time
  driver.en_pwm_mode(1); //enable extremely quiet stepping
  driver.microsteps(256); //256 microsteps
  driver.rms_current(400); //400mA RMS  //Die maximale Stromstärke

}

void loop() {
  
  LEFT = digitalRead(BTN_LEFT);
  RIGHT = digitalRead(BTN_RIGHT);
 

      //Bedingung das nicht beide Buttons gleichzeitig gedrückt werden können
  
    if ((LEFT) == (RIGHT) == HIGH )
  
      LEFT == LOW;
      RIGHT == LOW;
  
    //Ausgsbe des Buttons der den Motor nach links laufen lässt
  
    if ((LEFT) == HIGH) {
      digitalWrite(DIR_PIN, HIGH);   //gibt das Signal das er sich nach links derehen soll
      digitalWrite(EN_PIN, LOW);      //aktiviert Motor
    }
    else{
      digitalWrite(DIR_PIN, LOW);   //(Standart)
      digitalWrite(EN_PIN, HIGH);     //deaktiviert den Motor wider
    }
  
    //Ausgsbe des Buttons der den Motor nach rechts laufen lässt
  
    if ((RIGHT) == HIGH) {
  
      digitalWrite(STEP_PIN, LOW);   //gibt das Signal das er sich nach rechts derehen soll
      digitalWrite(EN_PIN, LOW);      //aktiviert Motor
    }
    else {
  
      digitalWrite(DIR_PIN, LOW);     //(Standart)
      digitalWrite(EN_PIN, HIGH);     //deaktiviert den Motor wider
  }

 //make steps
  digitalWrite(STEP_PIN, HIGH);
  delayMicroseconds(10);
  digitalWrite(STEP_PIN, LOW);
  delayMicroseconds(10);
}
[image]
[image]

`

teemuatlut, 2020-03-26:
I'll write out what your `loop` function does in pseudo code.

* Read both switches
* If left and right switches are the same, and if the **result of the comparison** is high/true/1, then compare left switch to low/0. _Nothing is set._
* Compare right button, but do nothing.
* If left button is high, set dir pin and enable motor, else disable motor.
* Same with right
* Run steps

So take a closer look at your first if statement.

RandyLJackson, 2020-03-26:
OK ^^ Thanks now i know that this part has his hifficulties.

I tried reparing the code but it still does not work.

```
if ((LEFT == HIGH) && (RIGHT == HIGH)){
  
      digitalWrite(EN_PIN, HIGH);
    }
   
```
I hope that i did it right.

But it is still not working.

So I tried to use a example from Watterott SilentStepStick, to see if the Stepper is working correctly:

Here the copyed code:
```
/*
  Trinamic TMC5160 Example
  
  TMCStepper library required:
  https://github.com/teemuatlut/TMCStepper
*/

#include <TMCStepper.h>

// Note: You also have to connect GND, 5V/VIO and VM.
//       A connection diagram can be found in the schematics.
#define EN_PIN    7 //enable
#define DIR_PIN   8 //direction
#define STEP_PIN  9 //step

#define CS_PIN   10 //CS chip select
#define MOSI_PIN 11 //SDI/MOSI (ICSP: 4, Uno: 11, Mega: 51)
#define MISO_PIN 12 //SDO/MISO (ICSP: 1, Uno: 12, Mega: 50)
#define SCK_PIN  13 //CLK/SCK  (ICSP: 3, Uno: 13, Mega: 52)

#define R_SENSE   0.075f //TMC5160: 0.075

TMC5160Stepper tmc = TMC5160Stepper(CS_PIN, R_SENSE, MOSI_PIN, MISO_PIN, SCK_PIN);

void setup()
{
  //set pins
  pinMode(EN_PIN, OUTPUT);
  digitalWrite(EN_PIN, HIGH); //deactivate driver (LOW active)
  pinMode(DIR_PIN, OUTPUT);
  digitalWrite(DIR_PIN, LOW); //direction: LOW or HIGH
  pinMode(STEP_PIN, OUTPUT);
  digitalWrite(STEP_PIN, LOW);

  pinMode(CS_PIN, OUTPUT);
  digitalWrite(CS_PIN, HIGH);
  pinMode(MOSI_PIN, OUTPUT);
  digitalWrite(MOSI_PIN, LOW);
  pinMode(MISO_PIN, INPUT);
  digitalWrite(MISO_PIN, HIGH);
  pinMode(SCK_PIN, OUTPUT);
  digitalWrite(SCK_PIN, LOW);

  //init serial port
  Serial.begin(9600); //init serial port and set baudrate
  while(!Serial); //wait for serial port to connect (needed for Leonardo only)
  Serial.println("\nStart...");

  //set driver config
  tmc.begin();
  tmc.toff(4); //off time
  tmc.blank_time(24); //blank time
  //tmc.en_pwm_mode(1); //enable extremely quiet stepping
  tmc.microsteps(16); //16 microsteps
  tmc.rms_current(400); //400mA RMS

  //outputs on (LOW active)
  digitalWrite(EN_PIN, LOW);
}

void loop()
{
  static uint32_t last_time=0;
  uint32_t ms = millis();

  if((ms-last_time) > 1000) //run every 1s
  {
    last_time = ms;

    if(tmc.diag0_error()){ Serial.println(F("DIAG0 error"));    }
    if(tmc.ot())         { Serial.println(F("Overtemp."));      }
    if(tmc.otpw())       { Serial.println(F("Overtemp. PW"));   }
    if(tmc.s2ga())       { Serial.println(F("Short to Gnd A")); }
    if(tmc.s2gb())       { Serial.println(F("Short to Gnd B")); }
    if(tmc.ola())        { Serial.println(F("Open Load A"));    }
    if(tmc.olb())        { Serial.println(F("Open Load B"));    }
  }

  //make steps
  digitalWrite(STEP_PIN, HIGH);
  delayMicroseconds(10);
  digitalWrite(STEP_PIN, LOW);
  delayMicroseconds(10);
}

```

So came another problem with it.
Everything is connected correctly but the serial monitor is pointing out that both Loads are open (A and B).

I wired the whole system again but the same warning poped out.
I send a image were you can see the wireing.
[image]
[image]

Thank you for your help^^

RandyLJackson, 2020-04-08:
Hello @teemuatlut  ,

do you have a working code for a arduino Uno with a SPI connection, so I can see that my System is working?

changedsoul, 2020-07-10:
Have you figured out why its not working? I have the same issue and cannot figure this out.

fenixil, 2020-12-13:
@RandyLJackson , @changedsoul I had a problem with SPI too. I managed it to work by powering VM. I see on your images that it is not powered, so this might be the case too. I described [my findings here](https://github.com/fenixil/robostroller/tree/master/blog/20201211_TMC5160_debugging) but still have no clue why SPI does not work without VM being powered.

laminarturbulent, 2020-12-24:
@fenixil I think I found the answer: https://learn.watterott.com/silentstepstick/faq/#what-is-the-difference-between-silentstepsticks-with-3-5v-and-5v-logic-voltage

> The SilentStepSticks with a variable logic voltage (VIO) of 3-5V use the internal linear regulator of the TMCxxxx to generate from the motor supply voltage (VM) a 5V voltage for the internal digital and analog circuit (about 20mA).

Which also makes me a little confused as to why the VCC_IO pin exists if there is a regulator to get logic level voltage from VM.

Additionally: https://learn.watterott.com/silentstepstick/faq/#silentstepsticks-with-variable-3-5v-logic-voltage

> **SilentStepSticks with variable 3-5V logic voltage**
> 
> At power-up the motor supply voltage VM should come up first and then the logic supply voltage VIO. On power-down the logic supply voltage VIO should turned off at first and then the motor supply voltage VM, because the internal logic of the TMCxxxx driver is powered from VM.

This may be why my BIGTREETECH TMC5160 V1.2 died as I turned off VM before unplugging my ESP32 (I also did not have VCC_IO plugged in, not sure if that has anything to do with it though).

fenixil, 2020-12-28:
@laminarturbulent great catch! I checked [TMC5160 datasheet](https://github.com/fenixil/robostroller/blob/master/datasheets/TMC/TMC5160A_Datasheet_Rev1.14.pdf) and, indeed, the 'brain' is powered by VCC which is connected to VM power in the reference diagram:
[image]

Pin | TQFP | QFN | Function
--- | ------ | ----- | ------------
VCC_IO | 20 | 20 | 3.3V to 5V IO supply voltage for all digital pins.
VCC | 29 | 30 | 5V supply input for digital circuitry within chip. Provide 100nF or bigger capacitor to GND (GND plane) near pin. Shall be supplied by 5VOUT. A 2.2 or 3.3 Ohm resistor is recommended for decoupling noise from 5VOUT. **When using an external supply, make sure, that VCC comes up before or in parallel to 5VOUT or VCC_IO, whichever comes up later**!

VCC_IO is just a logic level for the digital pins, but not a power supply for "digital circuitry within chip".

Jdparker4837, 2021-03-10:
Did you ever get this working?  I finally got mine working using the code here.  The only difference I see is according to the TMC5160 Datasheet, the CLK pin should be connected to ground to use internal clock.

yasir-shahzad, 2023-09-16:
> @RandyLJackson , @changedsoul I had a problem with SPI too. I managed it to work by powering VM. I see on your images that it is not powered, so this might be the case too. I described [my findings [...]

The SPI does not work without being powered by VM. The reason is that the TMC5160 has a 5V regulator whose output is connected to the VCC of the IC in the schematic, while VCC_io is only used as a reference for the logic level