teemuatlut/TMCStepper #121: TMC5160 SilentStepStick in spi mode?

nilsgraebner, 2020-04-02:
Hello,
it is more a question than an issue. I am wondering if it is possible to use the TMC5160 SilentStepStick in spi mode. So that there is no need for the step and dir connection. As far as I understood all the examples the velocity is defined by the pulse frequency which is generated by the Arduini. At https://learn.watterott.com/silentstepstick/comparison/ is written that the TMC5160 is able to work with spi connection or is this just for the setup?

teemuatlut, 2020-04-02:
Yes it is possible. See chapter 12 in the datasheet.

nilsgraebner, 2020-04-02:
thanks!
but then is the next question how? and is it possible with this library? maybe someone has an example code for this.

teemuatlut, 2020-04-02:
I'm not aware of any examples.
The ramp generator is described in the datasheet and the register relating to it are in chapter 6.3.1.
Yes the library does expose the necessary register calls.

nilsgraebner, 2020-04-02:
ok so then i have to do something like integrating this register to your library. i will try to do my best

closed 2020-08-07 by teemuatlut