teemuatlut/TMCStepper #126: TMC2208 rms_current set get discrepancy

imwhocodes, 2020-04-20:
While testing some stepper I noticed that they became rather hot even if the TMC2208 was configured with a conservative rms current using `rms_current(uint16_t mA, float mult)`

Checking current consumption it was higher than excepted so i logged some information

Here is the log:
`SET:<value>` is the value configured with `rms_current(<value>, 0.1)`
`GET:<value>` is the value obtained right after with `rms_current()`

```
SET: 0      GET: 55
SET: 5      GET: 55
SET: 10     GET: 55
SET: 15     GET: 55
SET: 20     GET: 55
SET: 25     GET: 55
SET: 30     GET: 55
SET: 35     GET: 55
SET: 40     GET: 55
SET: 45     GET: 55
SET: 50     GET: 55
SET: 55     GET: 55
SET: 60     GET: 55
SET: 65     GET: 110
SET: 70     GET: 110
SET: 75     GET: 110
SET: 80     GET: 110
SET: 85     GET: 110
SET: 90     GET: 110
SET: 95     GET: 165
SET: 100    GET: 165
SET: 105    GET: 165
SET: 110    GET: 165
SET: 115    GET: 165
SET: 120    GET: 165
SET: 125    GET: 220
SET: 130    GET: 220
SET: 135    GET: 220
SET: 140    GET: 220
SET: 145    GET: 220
SET: 150    GET: 220
SET: 155    GET: 276
SET: 160    GET: 276
SET: 165    GET: 276
SET: 170    GET: 276
SET: 175    GET: 276
SET: 180    GET: 276
SET: 185    GET: 331
SET: 190    GET: 331
SET: 195    GET: 331
SET: 200    GET: 331
SET: 205    GET: 331
SET: 210    GET: 331
SET: 215    GET: 386
SET: 220    GET: 386
SET: 225    GET: 386
SET: 230    GET: 386
SET: 235    GET: 386
SET: 240    GET: 386
SET: 245    GET: 441
SET: 250    GET: 441
SET: 255    GET: 441
SET: 260    GET: 441
SET: 265    GET: 441
SET: 270    GET: 441
SET: 275    GET: 441
SET: 280    GET: 497
SET: 285    GET: 497
SET: 290    GET: 497
SET: 295    GET: 497
SET: 300    GET: 497
SET: 305    GET: 497
SET: 310    GET: 552
SET: 315    GET: 552
SET: 320    GET: 552
SET: 325    GET: 552
SET: 330    GET: 552
SET: 335    GET: 552
SET: 340    GET: 607
SET: 345    GET: 607
SET: 350    GET: 607
SET: 355    GET: 607
SET: 360    GET: 607
SET: 365    GET: 607
SET: 370    GET: 662
SET: 375    GET: 662
SET: 380    GET: 662
SET: 385    GET: 662
SET: 390    GET: 662
SET: 395    GET: 662
SET: 400    GET: 718
SET: 405    GET: 718
SET: 410    GET: 718
SET: 415    GET: 718
SET: 420    GET: 718
SET: 425    GET: 718
SET: 430    GET: 773
SET: 435    GET: 773
SET: 440    GET: 773
SET: 445    GET: 773
SET: 450    GET: 773
SET: 455    GET: 773
SET: 460    GET: 828
SET: 465    GET: 828
SET: 470    GET: 828
SET: 475    GET: 828
SET: 480    GET: 828
SET: 485    GET: 828
SET: 490    GET: 883
SET: 495    GET: 883
SET: 500    GET: 883`
````

The driver is a SilentStepStick TMC2208 by Watterott, here is the configuration pushed:

```
TMC_SERIAL_CON.begin(9600);
TMC2208Stepper tmc_driver { &TMC_SERIAL_CON , 0.11};
tmc_driver.begin();
tmc_driver.I_scale_analog(false);
tmc_driver.en_spreadCycle(false);
tmc_driver.multistep_filt(true);
tmc_driver.pwm_autoscale(true);
tmc_driver.pwm_autograd(true);
tmc_driver.rms_current(400, 0.5);
tmc_driver.microsteps(32);
```

I get the quantization caused by the internal math of the trinamic, but is the linear growth of the actual computed value excepted behaviour? 

Thanks so much,
Luca

teemuatlut, 2020-04-20:
What are the raw CS values?

imwhocodes, 2020-04-20:
Thank for the fast answer:

Here they are:

```
SET: 0      GET: 55     IRUN: 0
SET: 5      GET: 55     IRUN: 0
SET: 10     GET: 55     IRUN: 0
SET: 15     GET: 55     IRUN: 0
SET: 20     GET: 55     IRUN: 0
SET: 25     GET: 55     IRUN: 0
SET: 30     GET: 55     IRUN: 0
SET: 35     GET: 55     IRUN: 0
SET: 40     GET: 55     IRUN: 0
SET: 45     GET: 55     IRUN: 0
SET: 50     GET: 55     IRUN: 0
SET: 55     GET: 55     IRUN: 0
SET: 60     GET: 55     IRUN: 0
SET: 65     GET: 110    IRUN: 1
SET: 70     GET: 110    IRUN: 1
SET: 75     GET: 110    IRUN: 1
SET: 80     GET: 110    IRUN: 1
SET: 85     GET: 110    IRUN: 1
SET: 90     GET: 110    IRUN: 1
SET: 95     GET: 165    IRUN: 2
SET: 100    GET: 165    IRUN: 2
SET: 105    GET: 165    IRUN: 2
SET: 110    GET: 165    IRUN: 2
SET: 115    GET: 165    IRUN: 2
SET: 120    GET: 165    IRUN: 2
SET: 125    GET: 220    IRUN: 3
SET: 130    GET: 220    IRUN: 3
SET: 135    GET: 220    IRUN: 3
SET: 140    GET: 220    IRUN: 3
SET: 145    GET: 220    IRUN: 3
SET: 150    GET: 220    IRUN: 3
SET: 155    GET: 276    IRUN: 4
SET: 160    GET: 276    IRUN: 4
SET: 165    GET: 276    IRUN: 4
SET: 170    GET: 276    IRUN: 4
SET: 175    GET: 276    IRUN: 4
SET: 180    GET: 276    IRUN: 4
SET: 185    GET: 331    IRUN: 5
SET: 190    GET: 331    IRUN: 5
SET: 195    GET: 331    IRUN: 5
SET: 200    GET: 331    IRUN: 5
SET: 205    GET: 331    IRUN: 5
SET: 210    GET: 331    IRUN: 5
SET: 215    GET: 386    IRUN: 6
SET: 220    GET: 386    IRUN: 6
SET: 225    GET: 386    IRUN: 6
SET: 230    GET: 386    IRUN: 6
SET: 235    GET: 386    IRUN: 6
SET: 240    GET: 386    IRUN: 6
SET: 245    GET: 441    IRUN: 7
SET: 250    GET: 441    IRUN: 7
SET: 255    GET: 441    IRUN: 7
SET: 260    GET: 441    IRUN: 7
SET: 265    GET: 441    IRUN: 7
SET: 270    GET: 441    IRUN: 7
SET: 275    GET: 441    IRUN: 7
SET: 280    GET: 497    IRUN: 8
SET: 285    GET: 497    IRUN: 8
SET: 290    GET: 497    IRUN: 8
SET: 295    GET: 497    IRUN: 8
SET: 300    GET: 497    IRUN: 8
SET: 305    GET: 497    IRUN: 8
SET: 310    GET: 552    IRUN: 9
SET: 315    GET: 552    IRUN: 9
SET: 320    GET: 552    IRUN: 9
SET: 325    GET: 552    IRUN: 9
SET: 330    GET: 552    IRUN: 9
SET: 335    GET: 552    IRUN: 9
SET: 340    GET: 607    IRUN: 10
SET: 345    GET: 607    IRUN: 10
SET: 350    GET: 607    IRUN: 10
SET: 355    GET: 607    IRUN: 10
SET: 360    GET: 607    IRUN: 10
SET: 365    GET: 607    IRUN: 10
SET: 370    GET: 662    IRUN: 11
SET: 375    GET: 662    IRUN: 11
SET: 380    GET: 662    IRUN: 11
SET: 385    GET: 662    IRUN: 11
SET: 390    GET: 662    IRUN: 11
SET: 395    GET: 662    IRUN: 11
SET: 400    GET: 718    IRUN: 12
SET: 405    GET: 718    IRUN: 12
SET: 410    GET: 718    IRUN: 12
SET: 415    GET: 718    IRUN: 12
SET: 420    GET: 718    IRUN: 12
SET: 425    GET: 718    IRUN: 12
SET: 430    GET: 773    IRUN: 13
SET: 435    GET: 773    IRUN: 13
SET: 440    GET: 773    IRUN: 13
SET: 445    GET: 773    IRUN: 13
SET: 450    GET: 773    IRUN: 13
SET: 455    GET: 773    IRUN: 13
SET: 460    GET: 828    IRUN: 14
SET: 465    GET: 828    IRUN: 14
SET: 470    GET: 828    IRUN: 14
SET: 475    GET: 828    IRUN: 14
SET: 480    GET: 828    IRUN: 14
SET: 485    GET: 828    IRUN: 14
SET: 490    GET: 883    IRUN: 15
SET: 495    GET: 883    IRUN: 15
SET: 500    GET: 883    IRUN: 15
```

Happy to help

imwhocodes, 2020-04-20:
I did some research and given the formula:
`((CS + 1) / 32) * (VSense / (0.11Ω + 20milliΩ))  * (1 / √2)`

Seem like that `Vsense` is always **0.325** and never **0.180**, this explains the results

I also checked the value of `vsense()` and it is always `false`

Even if forcefully set it to `vsense(true)` after `rms_current(uint16_t mA, float mult)` it stay setted on false 

```
SET: 0      GET: 55     IRUN: 0     VSENSE: false
SET: 5      GET: 55     IRUN: 0     VSENSE: false
SET: 10     GET: 55     IRUN: 0     VSENSE: false
SET: 15     GET: 55     IRUN: 0     VSENSE: false
SET: 20     GET: 55     IRUN: 0     VSENSE: false
SET: 25     GET: 55     IRUN: 0     VSENSE: false
SET: 30     GET: 55     IRUN: 0     VSENSE: false
SET: 35     GET: 55     IRUN: 0     VSENSE: false
SET: 40     GET: 55     IRUN: 0     VSENSE: false
SET: 45     GET: 55     IRUN: 0     VSENSE: false
SET: 50     GET: 55     IRUN: 0     VSENSE: false
SET: 55     GET: 55     IRUN: 0     VSENSE: false
SET: 60     GET: 55     IRUN: 0     VSENSE: false
SET: 65     GET: 110    IRUN: 1     VSENSE: false
SET: 70     GET: 110    IRUN: 1     VSENSE: false
SET: 75     GET: 110    IRUN: 1     VSENSE: false
SET: 80     GET: 110    IRUN: 1     VSENSE: false
SET: 85     GET: 110    IRUN: 1     VSENSE: false
SET: 90     GET: 110    IRUN: 1     VSENSE: false
SET: 95     GET: 165    IRUN: 2     VSENSE: false
SET: 100    GET: 165    IRUN: 2     VSENSE: false
SET: 105    GET: 165    IRUN: 2     VSENSE: false
SET: 110    GET: 165    IRUN: 2     VSENSE: false
SET: 115    GET: 165    IRUN: 2     VSENSE: false
SET: 120    GET: 165    IRUN: 2     VSENSE: false
SET: 125    GET: 220    IRUN: 3     VSENSE: false
SET: 130    GET: 220    IRUN: 3     VSENSE: false
SET: 135    GET: 220    IRUN: 3     VSENSE: false
SET: 140    GET: 220    IRUN: 3     VSENSE: false
SET: 145    GET: 220    IRUN: 3     VSENSE: false
SET: 150    GET: 220    IRUN: 3     VSENSE: false
SET: 155    GET: 276    IRUN: 4     VSENSE: false
SET: 160    GET: 276    IRUN: 4     VSENSE: false
SET: 165    GET: 276    IRUN: 4     VSENSE: false
SET: 170    GET: 276    IRUN: 4     VSENSE: false
SET: 175    GET: 276    IRUN: 4     VSENSE: false
SET: 180    GET: 276    IRUN: 4     VSENSE: false
SET: 185    GET: 331    IRUN: 5     VSENSE: false
SET: 190    GET: 331    IRUN: 5     VSENSE: false
SET: 195    GET: 331    IRUN: 5     VSENSE: false
SET: 200    GET: 331    IRUN: 5     VSENSE: false
SET: 205    GET: 331    IRUN: 5     VSENSE: false
SET: 210    GET: 331    IRUN: 5     VSENSE: false
SET: 215    GET: 386    IRUN: 6     VSENSE: false
SET: 220    GET: 386    IRUN: 6     VSENSE: false
SET: 225    GET: 386    IRUN: 6     VSENSE: false
SET: 230    GET: 386    IRUN: 6     VSENSE: false
SET: 235    GET: 386    IRUN: 6     VSENSE: false
SET: 240    GET: 386    IRUN: 6     VSENSE: false
SET: 245    GET: 441    IRUN: 7     VSENSE: false
SET: 250    GET: 441    IRUN: 7     VSENSE: false
SET: 255    GET: 441    IRUN: 7     VSENSE: false
SET: 260    GET: 441    IRUN: 7     VSENSE: false
SET: 265    GET: 441    IRUN: 7     VSENSE: false
SET: 270    GET: 441    IRUN: 7     VSENSE: false
SET: 275    GET: 441    IRUN: 7     VSENSE: false
SET: 280    GET: 497    IRUN: 8     VSENSE: false
SET: 285    GET: 497    IRUN: 8     VSENSE: false
SET: 290    GET: 497    IRUN: 8     VSENSE: false
SET: 295    GET: 497    IRUN: 8     VSENSE: false
SET: 300    GET: 497    IRUN: 8     VSENSE: false
SET: 305    GET: 497    IRUN: 8     VSENSE: false
SET: 310    GET: 552    IRUN: 9     VSENSE: false
SET: 315    GET: 552    IRUN: 9     VSENSE: false
SET: 320    GET: 552    IRUN: 9     VSENSE: false
SET: 325    GET: 552    IRUN: 9     VSENSE: false
SET: 330    GET: 552    IRUN: 9     VSENSE: false
SET: 335    GET: 552    IRUN: 9     VSENSE: false
SET: 340    GET: 607    IRUN: 10    VSENSE: false
SET: 345    GET: 607    IRUN: 10    VSENSE: false
SET: 350    GET: 607    IRUN: 10    VSENSE: false
SET: 355    GET: 607    IRUN: 10    VSENSE: false
SET: 360    GET: 607    IRUN: 10    VSENSE: false
SET: 365    GET: 607    IRUN: 10    VSENSE: false
SET: 370    GET: 662    IRUN: 11    VSENSE: false
SET: 375    GET: 662    IRUN: 11    VSENSE: false
SET: 380    GET: 662    IRUN: 11    VSENSE: false
SET: 385    GET: 662    IRUN: 11    VSENSE: false
SET: 390    GET: 662    IRUN: 11    VSENSE: false
SET: 395    GET: 662    IRUN: 11    VSENSE: false
SET: 400    GET: 718    IRUN: 12    VSENSE: false
SET: 405    GET: 718    IRUN: 12    VSENSE: false
SET: 410    GET: 718    IRUN: 12    VSENSE: false
SET: 415    GET: 718    IRUN: 12    VSENSE: false
SET: 420    GET: 718    IRUN: 12    VSENSE: false
SET: 425    GET: 718    IRUN: 12    VSENSE: false
SET: 430    GET: 773    IRUN: 13    VSENSE: false
SET: 435    GET: 773    IRUN: 13    VSENSE: false
SET: 440    GET: 773    IRUN: 13    VSENSE: false
SET: 445    GET: 773    IRUN: 13    VSENSE: false
SET: 450    GET: 773    IRUN: 13    VSENSE: false
SET: 455    GET: 773    IRUN: 13    VSENSE: false
SET: 460    GET: 828    IRUN: 14    VSENSE: false
SET: 465    GET: 828    IRUN: 14    VSENSE: false
SET: 470    GET: 828    IRUN: 14    VSENSE: false
SET: 475    GET: 828    IRUN: 14    VSENSE: false
SET: 480    GET: 828    IRUN: 14    VSENSE: false
SET: 485    GET: 828    IRUN: 14    VSENSE: false
SET: 490    GET: 883    IRUN: 15    VSENSE: false
SET: 495    GET: 883    IRUN: 15    VSENSE: false
SET: 500    GET: 883    IRUN: 15    VSENSE: false
```

teemuatlut, 2020-04-20:
Sounds like a communication issue. When setting a new rms value there are no read operations from the driver IC, but when reading back the rms value the vsense value does get read. If the response is all zeros then it will come back as if `vsense = 0` and the rms calculation is based on that.

imwhocodes, 2020-04-20:
Ok, actually my TMCs are only connected to the serial TX pin of the microcontroller (Teensy4) and there isn't any multiplexer for the RX signal.

This would explain why `vsense` is always zero

But the thing is that the motors are still running hot (i'm talking more than 100C) even if I configure them well under specification (tested on multiple motor of different brand ecc) and they stay cool only when the the `rms_current` is setted to a value that will equal the one calculated with `VSense = 0.325`

Can it be that some some read operation spoil the local copy of the register and when some other write happen `vsense = 0` is erroneously sent over to the TMC?

I'm pretty sure that the TX communication is stable, all other command (microstep, spreadcycle on/off ecc) are correctly recived

teemuatlut, 2020-04-21:
The write-only capabilities were deprecated a version or two ago so it shouldn't overwrite cached shadow registers with read data anymore.

The IRUN value tells us that the formula works as expect.
CS = 12 and RSENSE = 0.11 gives an RMS current of 0.4A.

Another register that would be of interest to you is the `cs_actual` from DRV_STATUS but for that you need to get read capabilities working. That register would tell you the actual live Current Selection value at that moment.

Can you tell me any of the parameters for your motor?

imwhocodes, 2020-04-23:
Ok, I MITM the sent packets and the vsense flag is actually configured to true

Now i'm having a somewhat different problem: I connected the **pnd** pin directly to the **RX** pin of the Tennsy4.0, the **TX** pin is connected to **pnd** pin with a 1Kohm resistor but I'm still unable to read values from the TMC2208 (Watterott SilentStepStick, J12 soldered)

The code is the same as before but only one TMC connected now

imwhocodes, 2020-04-27:
Sorry but I didn't have it on hands,
these are the specs of the motor:

[https://en.nanotec.com/fileadmin/files/Datenblaetter/Schrittmotoren/ST4118/X/ST4118X0404-A.pdf](https://en.nanotec.com/fileadmin/files/Datenblaetter/Schrittmotoren/ST4118/X/ST4118X0404-A.pdf)

CTetford, 2021-03-11:
I'm having this same issue on the project I'm working on.

As a hacked workaround, in TMCStepper.h, setting max_retries to 20 rather than 2 fixes it for now. 

Anything lower was still resulting in CRC errors. Even with 10 retries it was still getting the occasional error.

I'm using an MKS v2.1 board with a BigTreeTech TMC2209. It must have really poor data channel quality for the transmissions to be that unreliable.

For anyone following this, it changes the amount of retry attempts when reading UART data in TMC2208Stepper::read.

teemuatlut, 2021-03-11:
Try reducing the baudrate
https://github.com/teemuatlut/TMCStepper/blob/ddc0c121af9117c14c173427f1dae165d8409ea5/src/source/TMC2208Stepper.cpp#L50-L57

CTetford, 2021-03-11:
That's a good point. I switched it to beginSerial(19200) and it seems to be working now without the extra retries.

hefl99, 2025-02-19:
Hi,

i have the same problem with my setup: I am using an Arduino Uno R4 Minima to control a Bigtreetech TMC2209 V1.3.

The wiring is as follows:

Arduino Uno R4 Minima        ---->    Bigtreetech TMC2209 V1.3

                TX                                                        RX
                RX                                                        TX
                GND                                                     EN
                5V                                                         VIO
                GND                                                     GND

From the Bigtreetech TMC2209 V1.3 manual i found that i have to solder to pads on the bottom of the TMC together to use both RX and TX of the TMC.

I used the following code to try if i can control the TMC via UART from the arduino. But when i set the current to 800 mA for example and then ask the TMC for the set rms current it gives 1436. 

`#include <TMCStepper.h>

// Define UART connection
#define SERIAL_PORT Serial1  // On Arduino Uno R4 Minima, use Serial1 for TX/RX
#define DRIVER_ADDRESS 0x00  // Default address for standalone mode

// Define motor current (in mA) and microstep resolution
#define R_SENSE 0.11f  // Sense resistor value for TMC2209

// Create a TMC2209 driver instance
TMC2209Stepper driver(&SERIAL_PORT, R_SENSE, DRIVER_ADDRESS);

void setup() {
    Serial.begin(115200);
    SERIAL_PORT.begin(115200);
    delay(1000);
    Serial.println("Initializing TMC2209...");
    
    driver.begin();
    driver.pdn_disable(true);
    driver.I_scale_analog(false);
    driver.toff(5);
    driver.ihold(16);
    driver.rms_current(600);
    delay(50);
    driver.microsteps(16);
    driver.intpol(false);
    delay(50);
    driver.pwm_autoscale(true);
    driver.push();
    
    if (driver.test_connection()) {
        Serial.println("TMC2209 connection successful!");
    } else {
        Serial.println("TMC2209 connection failed!");
    }

}

void loop() {
    if (Serial.available()) {
        char command = Serial.read();
        switch (command) {
            
            case 'c':
                int usernumber;

                Serial.println("Enter current:");  // Use double quotes for strings

                // Wait until user enters something
                while (Serial.available() == 0) {
                    // Do nothing, just wait for input
                }

                usernumber = Serial.parseInt();  // Read user input
                
                driver.rms_current(usernumber);
                
                Serial.print("Current given: ");
                Serial.println(usernumber);
                
                delay(1000);
                
                Serial.print("Current setting: ");
                Serial.println(driver.rms_current());

                break;

            
            default:
                Serial.println("Unknown command");
        }
    }
}
`

SerialMonitor:
`Enter current:
Current given: 100
Current setting: 165
Enter current:
Current given: 800
Current setting: 1436
Enter current:
Current given: 0
Current setting: 55
Enter current:
Current given: 800
Current setting: 1436`

I tried changing the baudrate  as suggested by @teemuatlut but the behavior is still the same.

Thank you all a lot for you help in advance.