teemuatlut/TMCStepper #127: Question regarding Vref pot setting vs software current setting in SPI mode

bill-orange, 2020-04-22:
I don't understand the relationship between the current setting in Marlin and the Vref pot adjustment.

The Vref pot setting sets RMS average current. The factory setting of 1.23v results in a current of 870mA average. 

That being said, does the Marlin current adjustment  `#define X_CURRENT       900` override the pot adjustment?  Does it scale it?  Are they two different things perhaps, one idle,one running. 

I can not find documentation on this nuance anywhere.  

What is the relationship between the software and hardware setting?

teemuatlut, 2020-04-22:
Marlin disables the vref input and it has no effect on the current value.

The relation is that the vref input can scale down the "digital current" by vref/2.5V
So if you've set the rms current to 800mA in the registers and you've enabled I_scale_analog and your vref is at 1.25V, then your rms current is actually at 400mA.

You can find more on chapter 9 in the TMC2130 datasheet.

bill-orange, 2020-04-23:
Makes perfect sense.  Thanks.  

I do have a follow-on question.  In Marlin, the analog scale appears to go from 0 - 3000 rather than the more logical 1000.  So do you scale with 3000 as the denominator or are setting larger than 1000 equivalent to 1000?  So in your example, would the scaling fraction be 400/1000 or 400/3000?

teemuatlut, 2020-04-23:
I presume you mean the max value possible in the LCD menu? That is just a reasonable max value for any of the supported drivers and most can't even reach that.
Marlin does not in any way use the vref potentiometer for setting the current. There's no scaling or denominators or even anything analog involved.

bill-orange, 2020-04-23:
You presume correctly. Got it.  Do you know the actual allowable maximum for the 2130?

teemuatlut, 2020-04-23:
The theoretical max A_rms would be around 1.74A with a sense resistor of 0.11Ohms, but you'll hit the thermal limits much sooner than that.

bill-orange, 2020-05-17:
I am now running very successfully with Vref set to the factory default of 1.25 volts.  Stealthchop is OFF. My stepper motors are room temperature. Dividing my software current settings by two, my actual current settings are :

X:   400 mA
Y:   450 mA
E:   450 mA
Z:   400 mA
Z1: 400 mA

I already have a lot of fan noise. With the board vertical In free air would you expect this installation to require active cooling?

teemuatlut, 2020-05-17:
With currents as low as 400mA, I wouldn't expect you to need active cooling, or likely cooling at all. I'd still consider adding small heatsinks as it's easy to do and doesn't cost much.

closed 2020-08-07 by teemuatlut