teemuatlut/TMCStepper #129: 2209 STALL GUARD

Giannino-dot, 2020-05-01:
hELLO ,  
the fuction driver.SGTHRS(STALL_VALUE); DONT work. All other fuction work. Seem dont become blue SGTHRS. Can you help me ?

edwardocano, 2020-05-08:
The same for me, i am using ESP32.

KrabasChuliganas, 2020-05-12:
Same here. Arduino nano.

metalinspired, 2020-07-04:
I've made a working example [here](https://gist.github.com/metalinspired/dcfe07ed0b9f42870eb54dcf8e29c126).

johnsmakar, 2020-07-12:
@metalinspired I'm having trouble getting your example to work. 

My Serial monitor shows this "Status: 358 0 214"
Any idea on what I may be doing wrong?

edwardocano, 2020-07-12:
That number looks good. I think you do not know what is StallGuard. Read the data sheet

johnsmakar, 2020-07-12:
@edwardocano I am trying to get the motor to stop spinning when there is a high load on the motor, is that not StallGuard?

edwardocano, 2020-07-12:
No, StallGuard detecta the load that the motor has

edwardocano, 2020-07-12:
You have to program the level where the motor must stop. The number that you get with the sketch is in the repository is the load so when the load is high you in your code must stop it. It does not stop by itself

johnsmakar, 2020-07-12:
Okay, Gotcha. So I have to create a routine myself that takes the load value and stops the motor

thanks for the clarification!

wuram806, 2020-07-15:
> I've made a working example [here](https://gist.github.com/metalinspired/dcfe07ed0b9f42870eb54dcf8e29c126).

i am testing your example code.
but The error of the message below appears.
---------------------------------
Testing connection...failed!
Likely cause: no power
Fix the problem and reset board.
----------------------------------
I think the cable connection is wrong. Can you check?
I use Arduino Uno and tmc2209 directly connected.
See below for the cable connection between Arduino Uno and tmc2209.
I am using BIGtreetech TMC2209 Driver.

1)uno 5v : tmc vdd
2)uno gnd : tmc gnd
3)dc 24v power + : tmc vm
4)dc 24v power - : tmc gnd
5)uno rx : tmc pdn uart (nearest EN PIN)
6)uno tx : tmc pdn uart (nearest EN PIN, use 1k ohm)
7)uno gpio pins : tmc step / en / dir
8)tmc a1/2,b1/2 : step motor

metalinspired, 2020-07-15:
@wuram806 I used exact same TMC driver.
You actually do not need to connect anything to TMC VDD, but do leave the ground.
Also STEP and DIR pin are not heeded as speed and direction are controlled by VACTUAL using uart.
If I'm reading this correctly you're trying to communicate with TMC using UNO hardware serial port (pins 0 and 1) ?
That will not work because UNO hardware serial is also used for communication with PC.
Just use the pins already present in the example and you should have no issues.

wuram806, 2020-07-15:
what  I wrote about rx/tx pin is my mistake. I connected uno pin 5/6 to tmc
rx/tx.

thank you very very very much for your reply!!!
I had a hard time with the stallguard issue for a week.
connecting pis as your reply, Everything is perfect!!!
thank you!!!!!

wuram806, 2020-07-16:
Hello?
i have one question.
when using driver.VACTUAL(), how determine the direction of motor rotation?
Even if driver.VACTUAL(~~)  value is minus, but motor direction is same..

2020년 7월 15일 (수) 오후 10:49, 꾸람이 <wuram806@gmail.com>님이 작성:

metalinspired, 2020-07-16:
VACTUAL must change direction in respect to sign of the value given.
Given that VACTUAL(0) will stop the motor, or better said hold it in place.

johnsmakar, 2020-07-16:
[image]

anyone know why I keep getting these spikes?

metalinspired, 2020-07-16:
@johnsmakar I use a really cheap stepper for testing purposes and I've noticed stall guard values can be erratic on it with no load whatsoever. Could be the case for you too. IIRC increasing the motor current (rms_current) helped to an extent. Other than that I have no clue :D

teemuatlut, 2020-07-16:
Could be a failed communication which by default returns 0.

johnsmakar, 2020-07-16:
@metalinspired no luck with raising rms_current

@teemuatlut is there a way to prevent it?
I am trying to make use of the sensorless homing and the spikes make it think that it has hit something.

`    if(driver.SG_RESULT() < 50) {
      driver.toff(0);
    }`

This is the logic I'm using^ maybe there is a way around it?

wuram806, 2020-07-17:
> @johnsmakar I use a really cheap stepper for testing purposes and I've noticed stall guard values can be erratic on it with no load whatsoever. Could be the case for you too. IIRC increasing the motor current (rms_current) helped to an extent. Other than that I have no clue :D

I have an additional question.
When using tmc2209 as a hardware serial (VACTUAL), is there a way to count the number of steps?

metalinspired, 2020-07-17:
@wuram806  Step pulses can be monitored via INDEX pin

pranavb104, 2020-12-07:
@johnsmakar I'm also currently testing the StallGuard feature, but I'm getting a lot of spikes. Did you find a way around it?

johnsmakar, 2020-12-07:
I did not. I transitioned into using the 5161 as of now and plan to circle back and troubleshoot at a later date.

pardovot, 2021-01-08:
@metalinspired Thanks a lot for your example! I finally got stall guard to work, but now I'm missing some logic with how to follow that up.

I can't really find much documentation about it, but once stall guard activates, how do I 'restart' the motor and get it back to work again? I tried running the driver functions that first get it to start running, but it's not working.

Thanks!

chriscairns, 2021-01-10:
@metalinspired Just wanted to say thanks for your example. I needed to get stall guard working on the TMC2209 waterott board with a teensy 3.2 and your example meant that I didn't need to rewrite the timer code.

Thanks, of course, to @teemuatlut for writing such an excellent library.