teemuatlut/TMCStepper #130: ESP32 compatibility

edwardocano, 2020-05-08:
I am trying to enable the drivers TMC2209 with the ESP32. I modified the example StallGuard_TMC2209 to enable ESP32 timers, but the output is:

0 0 55

I am using the HardwareSerial to enable UART2, and have tried with the three UARTS in the ESP32. 
Do you have any idea how to enable the stallGuard reading in ESP32?

This is the code.

`#include <Arduino.h> 
#include <TMCStepper.h>
#include <HardwareSerial.h>
//Cambiar hardwareserial.cpp pin definition
#define EN_PIN           14
#define DIR_PIN          26 // Direction para el ESP
#define STEP_PIN         27 //Step para el ESP
//#define SERIAL_PORT      Serial2
#define DRIVER_ADDRESS 0b00
#define R_SENSE        0.11f 
#define STALL_VALUE       5

hw_timer_t * timer1 = NULL; //se declara un apuntador de tipo para la configuracion del timer 
HardwareSerial motor1(1);    //Si le pongo HardwareSerial s_1(1); me marca error ya que s_1 ya existe. 
HardwareSerial motor2(2);
TMC2209Stepper driver(&motor2, R_SENSE, DRIVER_ADDRESS);

//void giro_normal();

void IRAM_ATTR onTimer() {
  //digitalWrite(STEP_PIN, !digitalRead(STEP_PIN));
}
 
void setup() {
  
 
  Serial.begin(115200);
  while(!Serial); 

  motor2.begin(115200);
  //driver.beginS erial(115200); //use in case software serial.

  pinMode(STEP_PIN, OUTPUT);
  pinMode(DIR_PIN, OUTPUT);
  pinMode(EN_PIN , OUTPUT);

  digitalWrite(EN_PIN,LOW);
  
  driver.begin();
  driver.toff(4);
  driver.blank_time(24);
  driver.rms_current(400); // mA
  driver.microsteps(16);
  driver.TCOOLTHRS(0xFFFFF); // 20bit max
  driver.semin(5);
  driver.semax(2);
  driver.sedn(0b01);
  driver.SGTHRS(STALL_VALUE);
giro_normal();
}
 
void loop() {
  
  static uint32_t last_time=0;
  uint32_t ms = millis();

  if((ms-last_time) > 100) { //run every 0.1s
    
    last_time = ms;

    Serial.print("0 ");
    Serial.print(driver.SG_RESULT(), DEC);
    Serial.print(" ");
    Serial.println(driver.cs2rms(driver.cs_actual()), DEC);
  }

}

void giro_normal(){
  cli();
  timer1 = timerBegin(3, 8,true); // Se configura el timer, en este caso uso el timer 4 de los 4 disponibles en el ESP(0,1,2,3)
                                 // el prescaler es 8, y true es una bandera que indica si la interrupcion se realiza en borde o en nivel
  timerAttachInterrupt(timer1, &onTimer, true); //Se vincula el timer con la funcion AttachInterrup 
                                               //la cual se ejecuta cuando se genera la interrupcion
  timerAlarmWrite(timer1, 10000, true); //En esta funcion se define el valor del contador en el cual se genera la interrupción del timer
  timerAlarmEnable(timer1); //Función para habilitar el temporizador.
  sei();
}`

agkirk, 2020-05-21:
The schematic you are looking for :  I have more data   1agkirk2@gmail.com
[Esp_wroom_32_datasheet_en.pdf](https://github.com/teemuatlut/TMCStepper/files/4664709/Esp_wroom_32_datasheet_en.pdf)

agkirk, 2020-05-21:
You might find this link useful: 
http://neilkolban.com/tech/esp8266/

edwardocano, 2020-05-22:
Thanks!

http://neilkolban.com/tech/esp8266/

johnsmakar, 2020-07-09:
@edwardocano Did you ever get it to work? I am having the same issue.

edwardocano, 2020-07-09:
Hello! Yes, probably you are connecting to 5 volts the driver. The uart communication for esp32 works with 3.3v

closed 2020-07-09 by edwardocano

awilkie1, 2020-08-06:
Has anyone had success in controlling TMC2208/2209 through UART using esp32?

Trying to use the UART so I can switch between spreadcycle and stealthchop for more torque on fast movements and silent stepping on slow movements.

See there is several issues floating regarding serial compatability.