teemuatlut/TMCStepper #131: Why are 'using' statements making inherited public methods protected?

beldenfox, 2020-06-12:
I'm writing a sketch that supports both the 2130 and 5160 drivers. I'm running into compilation problems caused by `using` statements that make certain calls protected for no apparent reason. For example, `TMC5160Stepper` inherits from `TMC5130Stepper` but then contains a `protected: using TMC5130Stepper::rndtf;` line which makes the call inaccessible.

There seems to be a bit of a tangle related to the 2130/2160 integration and the classes that derive from them. For example, `pwm_ampl` goes from public in the 2130 class to protected in the 2160 class and back to public in the 5130 class. And it skips a level on that last one, the 5130 class inherits from 2160 but makes public the 2130 version of `pwm_ampl`.

Any reason I shouldn't go in and do some cleanup here?

teemuatlut, 2020-06-12:
If you take for example the TMC2160Stepper class and the declaration `using TMC2130Stepper::pwm_ampl`, it's because there is no `pwm_ampl` in the TMC2160. However there are a hundred other methods that you want to to inherit but some that do not apply. There is no way to "uninherit" something in C++, but you can prohibit access to a method.

So when the inheritance chain is 2130 -> 2160 -> 5130 -> 5160, there are some methods that are in 2130 and in 5130 but not in 2160 and 5160.
For the `rndtf` method, you can find that in TMC2130 chopconf register but the 5160 does not have it.

If you find some item that should be accessible but isn't, or vice-versa, of course feel free to do submit a pull request. In this case you may also be interested in the Release_v1 branch as that changes the architecture quite drastically.

beldenfox, 2020-06-12:
My bad, I got lost in the code and the spec sheets and the particular coding idiom your using. Should have done my homework.

closed 2020-06-12 by beldenfox