teemuatlut/TMCStepper #132: TMC2209 with LDO 0.9 VERY noisy/resonant/vibratory

freighttt, 2020-06-13:
Ive got a strange issue with a new printer im settting up. i have a set of tmc2209's, on a 24v corexy. 

the original 1.8* motors are louder than they should be, but i didnt think much of it. this week i installed a set of LDO 0.9* steppers, and they vibrate so hard that the x gantry sings. they arent missing steps, the printer operates fine, theres just insane resonance, even at low speeds. i spent the last 3 days working on the frame, replacing belts, went through marlin systematically and turned on and off every setting i could think of(LA, square wave, etc). i played with microstepping numbers, voltages, etc, to no avail. i had reached the end of my sanity when i decided to stick the steppers on another machine i have, running 2208s. suprise...smooth as butter. 

another thing to note is that turning stealthchop on and off results in less vibration, but not lower noise. stealthchop is actually where most of the noise and vibration comes from, with it off there is less vibration(but still tons) and the 'r2d2' sounds are pretty pronounced. i do not use hybrid threshold, but have tested it on and off when i went through the various options systematically testing them(and combinations of them).

im positive now that this is something wrong with my build, not the motors themselves. in doing some online research, i see some people mentioning issues with PWM? Beyond config/config_adv im pretty lost with coding. i thought id ask here before i go randomly messing with stuff. the video this gent posted sounds pretty similar to what im experiencing.

https://github.com/teemuatlut/TMCStepper/issues/82?fbclid=IwAR2gj2J4OsEmA-4OPWzcz25UkDJoPzrX7P2s0psY3jTxFugMeGdfXp9YvgU

im sure you would like me to post some of my build, so i have zipped it and attached. let me know what else you would like me to post.

[Marlin-2.0.x.zip](https://github.com/teemuatlut/TMCStepper/files/4773918/Marlin-2.0.x.zip)

Andychud, 2020-06-19:
I'm having very similar issues with 0.9 steppers on coreXY using BTT SKRV1.3. My previous 1.8 motors were quieter.

Pilson66, 2020-06-22:
Having the same issue on CoreXY with 0.9 deg. motors, Marlin 2.0.5.3 and tmcstepper 0.6.2

Portzal, 2020-06-23:
@freighttt  I am now on Bugfix 2.0.x uploaded and built today, 23-06-20. 
Swapped in new TMC2209 drivers for X and Y. 
 I have had a play with printing in SpreadCycle and had a overall improvement in noise, vibration etc. by selecting CHOPPER_TIMING CHOPPER_PRUSAMK3_24V. 
My X Stepper seems to vibrate more than my Y in StealthChop and SpreadCycle. It has the same features as the Y stepper but is 8mm shorter in the body 
E3D 0.9 Compact but powerful, X
E3D 0.9 High torque motor, Y

h3po, 2020-06-30:
I posted a video with waveform measurements in the marlin ticket, if it helps: https://github.com/MarlinFirmware/Marlin/issues/18286#issuecomment-652028094

phazei, 2020-08-27:
Yes!  I have the same issue!
https://imgur.com/a/NZsZzT9

The first movements are at G1 F9000, the next are at G1 F10000

This also happens on the diagonal, at much lower speeds of F6000/F7000

Additionally, same noises are happening if I disconnect the belts from the steppers and run them unloaded.

jackhistoria, 2020-08-29:
I have the TMC 2209 von fysetc with the following code driven by an arduino mini and it runs smooth. Like really smooth down to the lowest speed. _(Be aware that the pwm pins might have different bit resolutions. That might lead to less half steps.)_
The  TMC 2209 has a different pinout than the 2208 that might lead to a connection to a port with a different resolution _(just a guess)._

12V, 1.7A motor ~ 2.4V Vref.

```
#include <TMCStepper.h>

#define MAX_SPEED        40 // In timer value
#define MIN_SPEED      6000

#define STALL_VALUE     100//100 // [0..255]

//6 PWM ports:D3, D5, D6, D9, D10, D11
#define EN_PIN           7 //38 // Enable
#define DIR_PIN          8 //55 // Direction
#define STEP_PIN         9 //54 // Step
#define SW_RX            5 //63 // TMC2208/TMC2224 SoftwareSerial receive pin
#define SW_TX            6 //40 // TMC2208/TMC2224 SoftwareSerial transmit pin
#define SERIAL_PORT  Serial // TMC2208/TMC2224 HardwareSerial port
#define DRIVER_ADDRESS 0b00 // TMC2209 Driver address according to MS1 and MS2

#define R_SENSE 0.11f // Match to your driver
                      // SilentStepStick series use 0.11
                      // UltiMachine Einsy and Archim2 boards use 0.2
                      // Panucatt BSD2660 uses 0.1
                      // Watterott TMC5160 uses 0.075

// Select your stepper driver type
//TMC2209Stepper driver(&SERIAL_PORT, R_SENSE, DRIVER_ADDRESS);
TMC2209Stepper driver(SW_RX, SW_TX, R_SENSE, DRIVER_ADDRESS);

//.....

```
using namespace TMC2208_n;

SFRSKZ, 2020-11-20:
there must be a problem with 2209 or about something related. drivers are just like a4988 in stealthchop. with correct chopper timing settings even spreadcycle sounds quieter.  really weird.

SFRSKZ, 2020-11-20:
> Ive got a strange issue with a new printer im settting up. i have a set of tmc2209's, on a 24v corexy.
>
> the original 1.8* motors are louder than they should be, but i didnt think much of it. this [...]

can you change your chopper timings to (3, 0, 1) it helped me. (ldo 0.9° 4238)

Portzal, 2020-11-20:
I have just installed E3D high torque 0.9's on my CoreXY and they are a lot smoother in StealthChop that the previous non branded units I was running. Board is SKR V1.4 Turbo with TMC2226's. 
Base speed in video is 100mm/s with travel moves 150mm/s. Even at 200% speed they are surprisingly well behaved. https://youtu.be/gTiPXbDZtqo

jason7594, 2021-07-14:
Hi Freinds, what LDO 0.9 motor you are using? part no? We plan to do some reseaching to see if we  can get some solutions.

freighttt, 2021-07-14:
> Hi Freinds, what LDO 0.9 motor you are using? part no? We plan to do some reseaching to see if we can get some solutions.

LDO 42STH47-1684MAC

If there’s any way I can help, and you’d like to speak outside of this thread, my email is soldbuchdatabase@gmail.com

teemuatlut, 2021-07-30:
I can take a look with an oscope if you have a link where I could order the same exact model. Preferably within EU.

freighttt, 2021-08-14:
> I can take a look with an oscope if you have a link where I could order the same exact model. Preferably within EU.

http://reprapmania.ro/shop/componente-imprimante-cnc/componente-electronica/motoare-pas-cu-pas-drivere/nema17-strong-power-ldo-42sth48-1684mac-0-9deg/

https://caribou3d.com/en/stepper-motors/244-ldo-040nm-09-degree-stepper-motor-ldo-42sth48-mac.html

magnificu, 2021-11-02:
I have the same issue with the noise/vibrations running Marlin 2.0.9.2 on SKR2.0 board with TMC5160 drivers and Netma17 46Ncm 09deg 2A steppers for X & Y axis in a CoreXY configuration.

The vibrations (of the stepper motors) are higher on the diagonal moves.

freighttt, 2021-12-28:
any new developments?

teemuatlut, 2022-01-02:
Unfortunately I was not able to find good values for this specific motor and it seemed to exhibit more vibrations than others I've used.

You can use the following code and PIO monitor (Arduino IDE doesn't really work for this) to test for yourself.
Use WASD keys to edit values.
```cpp
#include <TMCStepper.h>

constexpr uint8_t StepPin = 60;
constexpr uint8_t EnablePin = 56;
constexpr uint8_t RxPin = 64;
constexpr uint8_t TxPin = 64;

TMC2209Stepper driver(RxPin, TxPin, 0.11, 0);

uint16_t target_rpm = 60;
uint16_t rms_current = 600;

ISR(TIMER1_COMPA_vect){
  PORTF ^= 1 << 6;
}

uint32_t rpm2us(uint16_t rpm) {
  const float rps = rpm/60.0;
  const float fullsteps_per_s = rps * 200;
  const float msteps_per_s = fullsteps_per_s * 16;
  const float us_between_steps = 1000000ul / msteps_per_s;
  return us_between_steps;
}

TMC2208_n::CHOPCONF_t chopconf{};
TMC2208_n::GCONF_t gconf{};
TMC2208_n::PWMCONF_t pwmconf{};

const char header[] = "spread toff  hend hstrt tbl    rms    rpm";
char buffer[128] = {0};
uint8_t item = 1;

void print_values();

void setup() {
  Serial.begin(115200);
  pinMode(EnablePin, OUTPUT);
  pinMode(StepPin, OUTPUT);

  // Set stepper interrupt
  {
    cli();//stop interrupts
    TCCR1A = 0;// set entire TCCR1A register to 0
    TCCR1B = 0;// same for TCCR1B
    TCNT1  = 0;//initialize counter value to 0
    OCR1A = rpm2us(60) - 1;
    // turn on CTC mode
    TCCR1B |= (1 << WGM12);
    // Set CS11 bits for 8 prescaler
    TCCR1B |= (1 << CS11);// | (1 << CS10);
    // enable timer compare interrupt
    TIMSK1 |= (1 << OCIE1A);
    sei();//allow interrupts
  }

  driver.begin();
  driver.rms_current(rms_current);
  driver.I_scale_analog(false);
  driver.TCOOLTHRS(0);

  auto vsense = driver.vsense();

  {
    chopconf.vsense = vsense;
    chopconf.dedge = true;
    chopconf.mres = 4;
    chopconf.tbl = 1;
    chopconf.intpol = true;
    chopconf.hend = 3;
    chopconf.hstrt = 0;
    chopconf.toff = 4;

    driver.CHOPCONF(chopconf.sr);
  }

  {
    gconf.i_scale_analog = true;
    gconf.internal_rsense = false;
    gconf.pdn_disable = true;
    gconf.mstep_reg_select = true;
    gconf.shaft = true;
    gconf.en_spreadcycle = false;
    gconf.multistep_filt = false;
    driver.GCONF(gconf.sr);
  }

  {
    pwmconf.pwm_autograd = true;
    pwmconf.pwm_autoscale = true;
    pwmconf.pwm_freq = 0b10;
    pwmconf.pwm_reg = 8;
    pwmconf.pwm_ofs = 36;
    pwmconf.pwm_grad = 128;
  }
  driver.PWMCONF(pwmconf.sr);

  digitalWrite(EnablePin, LOW);
  digitalWrite(StepPin, HIGH);
  delay(500);

  Serial.println("Ramping up...");

  for (uint16_t rpm = 20; rpm < target_rpm; rpm++) {
    OCR1A = rpm2us(rpm) - 1;
    delay(100);
  }

  Serial.println(header);
  print_values();
}

void edit_item(char c);

void loop() {

  if (Serial.available()) {
    auto c = Serial.read();

    switch (c) {
      case 'w': case 's':
        edit_item(c);
        break;
      case 'a':
        if (item > 1) item--;
        break;
      case 'd':
        if (item < 7) item++;
        break;
      default:
        break;
    }

    print_values();
  }
}

void edit_item(char c) {

  auto edit = [&](int val, int min, int max) {
    if (c=='w' && val < max) val++;
    if (c=='s' && val > min) val--;
    return val;
  };

  switch (item) {
    case 1:
      gconf.en_spreadcycle = edit(gconf.en_spreadcycle, 0, 1);
      break;
    case 2:
      chopconf.toff = edit(chopconf.toff, 0, 15);
      break;
    case 3:
      chopconf.hend = edit(chopconf.hend, 0, 15);
      break;
    case 4:
      chopconf.hstrt = edit(chopconf.hstrt, 0, 7);
      break;
    case 5:
      chopconf.tbl = edit(chopconf.tbl, 0, 3);
      break;
    case 6:
      if (c=='w' && rms_current < 1200) rms_current += 50;
      if (c=='s' && rms_current >  200) rms_current -= 50;
      driver.rms_current(rms_current);
      return;
    case 7:
      if (c=='w' && target_rpm < 200) target_rpm += 5;
      if (c=='s' && target_rpm >  10) target_rpm -= 5;
      OCR1A = rpm2us(target_rpm) - 1;
      return;
    
    default:
      return;
  }

  driver.GCONF(gconf.sr);
  driver.CHOPCONF(chopconf.sr);
}

void print_values() {
    auto n = snprintf(buffer, sizeof(buffer),
      "%c "
      "%c %1u %c"
      "%c %2u %c"
      "%c %2u %c"
      "%c %1u %c"
      "%c %1u %c"
      "%c %4u %c"
      "%c %3u %c",
      0xD,
      item == 1 ? '[' : ' ',
      gconf.en_spreadcycle,
      item == 1 ? ']' : ' ',
      item == 2 ? '[' : ' ',
      chopconf.toff,
      item == 2 ? ']' : ' ',
      item == 3 ? '[' : ' ',
      chopconf.hend,
      item == 3 ? ']' : ' ',
      item == 4 ? '[' : ' ',
      chopconf.hstrt,
      item == 4 ? ']' : ' ',
      item == 5 ? '[' : ' ',
      chopconf.tbl,
      item == 5 ? ']' : ' ',
      item == 6 ? '[' : ' ',
      rms_current,
      item == 6 ? ']' : ' ',
      item == 7 ? '[' : ' ',
      target_rpm,
      item == 7 ? ']' : ' '
    );

    Serial.write(buffer, n);
}

```