teemuatlut/TMCStepper #133: Please release the latest version (v0.6.3)

odaki, 2020-06-21:
Hello teemuatlut.

Could you please officially release the latest version (v0.6.3)?
We are waiting for the following fixes to be officially released.
https://github.com/teemuatlut/TMCStepper/commit/7ce0d8c981d52ea17c343ba62cdb0d789e30c4cb

We are developing an ESP32 compliant version of Grbl.
https://github.com/bdring/Grbl_Esp32/tree/Devt

Our target board performs SPI CS operations via a shift register IC that is driven by I2S.
https://github.com/bdring/6-Pack_CNC_Controller

There is a delay between the SPI CS operation and the actual reflection of the SPI CS signal,
so we need to delay the CS operation by TMCStepper a bit.

As a developer, we can change the source of TMCStepper, but it is too difficult for users.
So we want to replace switchCSpin() with our own implementation code that we will provide in Grbl_ESP32.
For that, the improvements you've made to the weak symbol would be very useful to us.

If you release the latest version, the Arduino IDE's library manager will allow ordinary users
to install the latest TMCStepper library needed for Grbl_ESP32.

Please consider releasing the latest version.

Thank you.

odaki, 2020-06-21:
@bdring @MitchBradley
If you have anything to add, please leave a comment.

teemuatlut, 2020-06-21:
I'd be happy to make the release. Have you tested if the weak override works as expected? I just pushed an update to move the attributes to the definitions instead.

I'm also curious about your project as I'm currently designing a CNC for myself and haven't yet settled on a control board. What's your current status with the hardware? Is there anywhere where I could purchase a board just yet?

bdring, 2020-06-21:
There are several boards [available on Tindie](https://www.tindie.com/stores/33366583/). [This one](https://www.tindie.com/products/33366583/4-axis-trinamic-cnc-controller-for-grbl_esp32/) might be of interest to you. The [firmware](https://github.com/bdring/Grbl_Esp32) has been stable for well over a year and is very actively developed.

I would be happy to send any one them to you no charge. We rely heavily on your library for Trinamic support. 

The changes requested allowed us to build [this controller](https://github.com/bdring/6-Pack_CNC_Controller). We have a few changes to the prototype, so it will not be available for many weeks.

[image]

teemuatlut, 2020-06-21:
[Version 0.7.0](https://github.com/teemuatlut/TMCStepper/releases/tag/v0.7.0) is now released.

I'd very much appreciate a board since you're offering!

bdring, 2020-06-21:
@teemuatlut Here is an [invite to our Slack](https://join.slack.com/t/buildlog/shared_invite/zt-7twbithi-bRGScOaCgJoBpTbkykfxoA)

If you go there and private message me, I can get a controller to you.

odaki, 2020-06-21:
Good morning. I woke up this morning and found a wonderful gift on my pillow!

> I'd be happy to make the release. Have you tested if the weak override works as expected? I just pushed an update to move the attributes to the definitions instead.

Yes, the weak attribute you specified is working as intended.
It allows us to override methods without having to use C++ inheritance.
https://github.com/bdring/Grbl_Esp32/blob/Devt/Grbl_Esp32/Motors/MotorClass.cpp#L443

I wish you the best of luck with the CNC you are designing.

Thanks for the very quick response, and happy birthday v0.7.0!

closed 2020-06-21 by odaki